meta-flutter/meta-flutter #4: Honister Runtime=Debug flutter-engine build break

jwinarske, 2021-11-04:
```
| FAILED: obj/third_party/dart/runtime/vm/libdart_vm_jit.malloc_hooks_unsupported.o
| /__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/src/buildtools/linux-x64/clang/bin/clang++ -MD -MF obj/third_party/dart/runtime/vm/libdart_vm_jit.malloc_hooks_unsupported.o.d --target=aarch64-unknown-linux-gnu --sysroot /__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/recipe-sysroot -DUSE_OPENSSL=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_LIBCPP_DISABLE_AVAILABILITY=1 -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DTARGET_ARCH_ARM64 -DDART_TARGET_OS_LINUX -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_ENABLE_TRACING=1 -DU_ENABLE_RESOURCE_TRACING=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -I../../third_party/dart/runtime -I../.. -Igen -I../../third_party/libcxx/include -I../../third_party/libcxxabi/include -I../../third_party/dart/runtime/include -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -fno-strict-aliasing -fstack-protector --param=ssp-buffer-size=4 -fPIC -pipe -pthread --target=aarch64-linux-gnu -DBORINGSSL_CLANG_SUPPORTS_DOT_ARCH -fcolor-diagnostics -Wall -Wextra -Wendif-labels -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-implicit-int-float-conversion -Wno-c99-designator -Wno-deprecated-copy -Wno-psabi -Wno-non-c-typedef-for-linkage -Wno-range-loop-construct -fvisibility=hidden --sysroot=/__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/recipe-sysroot -Wstring-conversion -Wnewline-eof -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -Werror -Wall -Wextra -Wno-unused-parameter -Wno-unused-private-field -Wnon-virtual-dtor -Wvla -Woverloaded-virtual -Wno-comments -g3 -ggdb3 -fno-rtti -fno-exceptions -Wimplicit-fallthrough -O3 -fvisibility-inlines-hidden -std=c++17 -fno-rtti -nostdinc++ -nostdinc++ -fvisibility=hidden -fno-exceptions -c ../../third_party/dart/runtime/vm/malloc_hooks_unsupported.cc -o obj/third_party/dart/runtime/vm/libdart_vm_jit.malloc_hooks_unsupported.o
| ../../third_party/dart/runtime/vm/malloc_hooks_unsupported.cc:73:26: error: 'mallinfo' is deprecated [-Werror,-Wdeprecated-declarations]
|   struct mallinfo info = mallinfo();
|                          ^
| /__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/recipe-sysroot/usr/include/malloc.h:113:48: note: 'mallinfo' has been explicitly marked deprecated here
| extern struct mallinfo mallinfo (void) __THROW __MALLOC_DEPRECATED;
| /__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/recipe-sysroot/usr/include/malloc.h:31:30: note: expanded from macro '__MALLOC_DEPRECATED'
| # define __MALLOC_DEPRECATED __attribute_deprecated__
|                              ^
| /__w/meta-flutter/rpi4-drm-honister-latest/raspberrypi4-64/tmp/work/cortexa72-poky-linux/flutter-engine/git-r0/recipe-sysroot/usr/include/sys/cdefs.h:281:51: note: expanded from macro '__attribute_deprecated__'
| # define __attribute_deprecated__ __attribute__ ((__deprecated__))
|                                                   ^
```

jwinarske, 2021-11-04:
This combo builds fine:

```
echo 'FLUTTER_CHANNEL = "dev"' >> ./conf/local.conf
echo 'PACKAGECONFIG:pn-flutter-engine = "embedder-for-target disable-desktop-embeddings fontconfig full-dart-sdk mode-release"' >> ./conf/local.conf
echo 'TARGET_GCC_VERSION = "11.2.0"' >> ./conf/local.conf
```

jwinarske, 2021-11-09:
To clarify this is a bug that originates from Dart VM source tree.

jwinarske, 2021-12-09:
https://dart-review.googlesource.com/c/sdk/+/218161 has landed on the Dart side, waiting for DEP update in master for fix.

snke, 2022-01-10:
With #39 the mallinfo2 issue is fixed. But building with 34be923d2750c05cfc15738bcfeb9609cb18416c from `jw/honister_debug_enable` and a non-X11 (`DISTRO_FEATURES:remove = " x11"`), but Wayland-only image I get the following error building flutter-engine-debug on honister:

```
| [1016/4529] /build/tmp/work/cortexa72-poky-linux/flutter-engine-debug/git-r0/src/buildtools/linux-x64/clang/bin/clang -MD -MF obj/third_party/glfw/src/glfw.input.o.d --target=aarch64-unknown-linux-gnu --sysroot /build/tmp/work/cortexa7
2-poky-linux/flutter-engine-debug/git-r0/recipe-sysroot -D_GLFW_X11 -D_GLFW_HAS_XF86VM -DUSE_OPENSSL=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_LIBCPP_DISABLE_AVAILABILITY=1 -D_LIBCPP_DISABLE_VISIBILITY_ANNOTAT
IONS -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/glfw/src -I../.. -Igen -I../../third_party/libcxx/include -I../../third_party/libcxxabi/include -I../../third_party/
glfw/include -I../../third_party/glfw/include/GLFW -fno-strict-aliasing -fstack-protector --param=ssp-buffer-size=4 -fPIC -pipe -pthread --target=aarch64-linux-gnu -DBORINGSSL_CLANG_SUPPORTS_DOT_ARCH -fcolor-diagnostics -fvisibility=hidd
en --sysroot=/build/tmp/work/cortexa72-poky-linux/flutter-engine-debug/git-r0/recipe-sysroot -Wstring-conversion -Wnewline-eof -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -Wendif-labels -Werror -Wno-missing-field-initializers 
-Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-implicit-int-float-conversion -Wno-c99-designator -Wno-deprecated-copy -Wno-psabi -Wno-non-c-typedef-for-linkage -Wno-range-loop-construct  -c ../../th
ird_party/glfw/src/input.c -o obj/third_party/glfw/src/glfw.input.o                                                                                                                                                                          
| FAILED: obj/third_party/glfw/src/glfw.input.o                                                                                                                                                                                              
| /build/tmp/work/cortexa72-poky-linux/flutter-engine-debug/git-r0/src/buildtools/linux-x64/clang/bin/clang -MD -MF obj/third_party/glfw/src/glfw.input.o.d --target=aarch64-unknown-linux-gnu --sysroot /build/tmp/work/cortexa72-poky-linux
/flutter-engine-debug/git-r0/recipe-sysroot -D_GLFW_X11 -D_GLFW_HAS_XF86VM -DUSE_OPENSSL=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_LIBCPP_DISABLE_AVAILABILITY=1 -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBC
PP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/glfw/src -I../.. -Igen -I../../third_party/libcxx/include -I../../third_party/libcxxabi/include -I../../third_party/glfw/include
 -I../../third_party/glfw/include/GLFW -fno-strict-aliasing -fstack-protector --param=ssp-buffer-size=4 -fPIC -pipe -pthread --target=aarch64-linux-gnu -DBORINGSSL_CLANG_SUPPORTS_DOT_ARCH -fcolor-diagnostics -fvisibility=hidden --sysroot
=/build/tmp/work/cortexa72-poky-linux/flutter-engine-debug/git-r0/recipe-sysroot -Wstring-conversion -Wnewline-eof -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -Wendif-labels -Werror -Wno-missing-field-initializers -Wno-unused-
parameter -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-implicit-int-float-conversion -Wno-c99-designator -Wno-deprecated-copy -Wno-psabi -Wno-non-c-typedef-for-linkage -Wno-range-loop-construct  -c ../../third_party/gl
fw/src/input.c -o obj/third_party/glfw/src/glfw.input.o                                                                                                                                                                                      
| In file included from ../../third_party/glfw/src/input.c:28:                                                                                                                                                                               
| In file included from ../../third_party/glfw/src/internal.h:187:                                                                                                                                                                           
| ../../third_party/glfw/src/x11_platform.h:33:10: fatal error: 'X11/Xlib.h' file not found                                                                                                                                                  
| #include <X11/Xlib.h>                                                                                                                                                                                                                      
|          ^~~~~~~~~~~~                                                                                                                                                                                                                      
| 1 error generated.
```

Do you know what the problem is there?

jwinarske, 2022-01-10:
That requires passing in `--no-build-glfw-shell`.  I have an issue open [issue ](https://github.com/flutter/flutter/issues/95711#event-5842456016) on flutter/flutter about this.  Chinmay decided to default the GLFW example shell, and that is being built for X11.  I'm requesting it not be enabled by default.

snke, 2022-01-10:
Thank you for the explanation and for taking care of this!

snke, 2022-01-25:
Everything works for me again with `FLUTTER_CHANNEL = "master"`

jwinarske, 2022-01-25:
@snke The hot fixes finally made it into beta.  The fixes started in master.

closed 2022-02-04 by jwinarske