meta-flutter/meta-flutter #7: problems adding flutter-sdk-native to SDK

pberginkonsult, 2021-11-09:
I plan to add flutter sdk to the SDK generated by Yocto but fail to do so.

`TOOLCHAIN_HOST_TASK:append = " nativesdk-flutter-sdk-native"`

results in:

```
$ bitbake <image>
...
ERROR: Task do_populate_sdk in <image>.bb rdepends upon non-existent task do_package_write_rpm in virtual:nativesdk:/workdir/sources/meta-flutter/recipes-graphics/flutter-apps/flutter-sdk-native_git.bb
```

I would like to check if this is a bug or if there is another way to add flutter sdk to Yocto SDK? With flutter in the Yocto SDK I would like to use it to build flutter app ready to be run on target. Comment if this is a preferred way of working.

jwinarske, 2021-11-09:
Hi Peter,

That hasn't come up as the standard dev workflow so far is to entirely use desktop tooling for flutter dev.  Regardless SDK support should be added.  The only downside to this is certs would need to be periodically updated.  In addition the desktop tools can be locked against an VM/Engine commit, so no real need for SDK scenario.  The SDK install with engine provides parts to generate AOT on host, if target has Runtime=Release.

The dev work flow we use:
* Flutter Devs do all the development using the standard host tools in a dedicate repo.
* Embedder -  Ubuntu 20 ivi-homescreen desktop native build, or QEMU/Yocto emulator that includes ivi-homescreen.
* Yocto Flutter App recipe runs as part of CI for release builds (based off of Flutter Gallery recipe).
* CI jobs that generate Ubuntu 20 (desktop) builds of everything, including engine builds Runtime=Debug, Runtime=Release, Runtime=Profile.
* Runtime=Profile is used for CI in conjunction with headless config.
* SDK scenario is only for native code work

I will add support for SDK for Flutter SDK.  The only downside is certificates will expire, so this scenario is not ideal for LTS.

pberginkonsult, 2021-11-09:
Hi Joel,
thanks for feedback. I did some POC here https://github.com/pberginkonsult/meta-flutter/commit/24997e28e91ff57c2ea777c3b85b20665d4913a1. With that commit I have been able to build a SDK containing the Flutter SDK. There is a script in SDK that is run on initialization and sets up some variables.

With the help of https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps#2-3-cross-build-for-arm64-targets-on-x64-hosts-manually I have then been able to build something with this flow:

1. Setup SDK
`source <sdk-path>/environment-setup-cortexa53-crypto-poky-linux`

2. Create Flutter project
```
flutter create hello
cd hello
```

3. Build Flutter asset
```
export FLUTTER_RESULT_DIR=build/linux-embedded-arm64
export FLUTTER_BUILD_MODE=release
export APP_PACKAGE_NAME="hello"
flutter build bundle --asset-dir=${FLUTTER_RESULT_DIR}/${FLUTTER_BUILD_MODE}/bundle/data/flutter_assets
cp ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-x64/icudtl.dat ${FLUTTER_RESULT_DIR}/${FLUTTER_BUILD_MODE}/bundle/data/
```

4. Build kernel snapshot
```
mkdir -p .dart_tool/flutter_build/flutter-embedded-linux
${FLUTTER_SDK}/bin/cache/dart-sdk/bin/dart \
  --verbose \
  --disable-dart-dev ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-x64/frontend_server.dart.snapshot \
  --sdk-root ${FLUTTER_SDK}/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ \
  --target=flutter \
  --no-print-incremental-dependencies \
  -Ddart.vm.profile=false \
  -Ddart.vm.product=true \
  --aot \
  --tfa \
  --packages .dart_tool/package_config.json \
  --output-dill .dart_tool/flutter_build/flutter-embedded-linux/app.dill \
  --depfile .dart_tool/flutter_build/flutter-embedded-linux/kernel_snapshot.d \
  package:${APP_PACKAGE_NAME}/main.dart
  ```

5. Build AOT image
```
${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64-release/clang_x64/gen_snapshot \
  --deterministic \
  --snapshot_kind=app-aot-elf \
  --elf=.dart_tool/flutter_build/flutter-embedded-linux/libapp.so \
  --strip \
  .dart_tool/flutter_build/flutter-embedded-linux/app.dill
mkdir -p ${FLUTTER_RESULT_DIR}/${FLUTTER_BUILD_MODE}/bundle/lib/
cp .dart_tool/flutter_build/flutter-embedded-linux/libapp.so ${FLUTTER_RESULT_DIR}/${FLUTTER_BUILD_MODE}/bundle/lib/
```

I'm newbie in Flutter area and feedback is very welcome! I can open a PR on this if you think this can be a way to go? In that case I would also like to document the workflow somewhere, README or elsewhere?

jwinarske, 2021-11-10:
fix is in the honister branch.  

The only thing required in SDK after setting up environment is to update the PATH with location of `flutter`.

closed 2021-11-10 by jwinarske

jwinarske, 2021-11-10:
I'm working on factoring out using Engine SDK in favor of using Flutter SDK.  It will enable 32-bit builds again.

jwinarske, 2021-11-11:
sdk refactor on honister is complete.

pberginkonsult, 2021-11-11:
Hi Joel,
thanks for the info! I must admit that I can not fully follow the change you've done and the purpose of it. Can you elaborate a bit around "Use Flutter SDK instead of Engine SDK", what is the real difference?

Then a wish from my side is to try to get better commit messages in this layer. Right now it is quite hard to follow what changes that have been done and the purpose of them, by reading git log.

jwinarske, 2021-11-11:
Hi Peter,

The SDK difference is:
Don't pass flutter sdk package config by default to flutter-engine.  Before
I was building flutter-engine by default with flutter sdk, and using the
front end snapshot to build AOT.  This is not needed as flutter-sdk-native
has the same bits.  So now I just use the flutter sdk bits to generate the
AOT image.  There was recent change in flutter-engine that re-enabled
32-bit ARM builds.  This change enables that, as you can see there are now
RaspberryPi 32 bit builds in CI.

I'll try to stick with PRs for changes.  I think that will help with
following changes.  I am prepping things for AGL integration, so there will
be a number of changes then things should slow down.  My near term goal is
to get CI builds for all the common targets running.  ST, NXP, and NVidia
are upcoming.  If you have anything specific let me know.

You're more than welcome to raise issues, ideas, or concerns.

Cheers,
Jeol