meta-flutter/meta-flutter #21: stm3215mp CI image fails

jwinarske, 2021-11-18:
```
ERROR: libxkbcommon-0.10.0-r0 do_packagedata: Can NOT get PRAUTO, exception No module named '_sysconfigdata'
ERROR: Logfile of failure stored in: /__w/meta-flutter/stm32mp15-dunfell/build-openstlinuxeglfs-stm32mp1-disco/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/libxkbcommon/0.10.0-r0/temp/log.do_packagedata.82728
NOTE: recipe libxkbcommon-0.10.0-r0: task do_packagedata: Failed
```

Have question posted to community forum with no response as of yet:
https://community.st.com/s/question/0D53W00001BwP61SAF/openstlinux-uboot-container-packaging-issue?t=1636750626238

tomstokes, 2021-11-25:
I had a similar issue in a different recipe on an STM32MP1 project. I never traced it all the way up but I worked around it by unexporting `_PYTHON_SYSCONFIGDATA_NAME` in the affected pacakge:

`_PYTHON_SYSCONFIGDATA_NAME[unexport] = "1"`

More discussion on a similar issue here: https://patchwork.openembedded.org/patch/175735/

jwinarske, 2021-11-27:
@tomstokes thanks Tom.  It's something related to running in docker container using a non-standard uid and Docker.  Someone reported on AGL mailing list having resolved similar issue using Podman instead of Docker.  Podman isn't supported by GitHub yet.  Another thing i played with was adding 'INHERIT_remove = "uninative" '.  It moved past u-boot build break to a later recipe.  I'll play around with your suggestion later next week.  Thanks again.

tomstokes, 2021-11-30:
Now that you mention it, I was also using a Docker container with a different UID.

This is one of those bugs I worked around and hoped it would be fixed upstream by the time I got back to it. Apparently not. :)

jwinarske, 2021-12-02:
@tomstokes problem resolved in this build job:
https://github.com/meta-flutter/meta-flutter/blob/dunfell/.github/workflows/stm32mp15.yml

A bit of trial and error but it builds on Ubuntu 20.  Still need to smoke test the image.

closed 2021-12-02 by jwinarske