meta-flutter/meta-flutter #26: is it possible to run flutter on a device without a GPU (i.mx7 Dual)

dbeulen, 2021-11-30:
I am trying to run flutter in a custom Yocto build on a device with a i.mx7 (armv7 32 bit) core. As far as i can see this device does not have a GPU. 

I managed to compile the sony embedders as well as flutter-pi, but they all give runtime errors. This makes me wondering:

Is it even possible to run flutter on a device without GPU?

jwinarske, 2021-11-30:
Yes I am very familiar with the i.mx7.

It would require building with SwiftShader enabled.  I have not tried this recently, and it looks like SwiftShader has deprecated their OpenGL ES implementation in favor of Vulkan.  So once there is a Vulkan based embedder available, it will be attainable.

I'll put SwiftShader support on my to-do list.  I'm thinking it should be ready sometime in January.

dbeulen, 2021-11-30:
Hy jwinarske,

Thank you for your response, it is great news! I already gave up all hopes to get it working. Some additional questions that pop up now:

Do you mean it would be possible to use DRM (flutter-pi) when SwiftShader is used? Or is it just the x11/wayland embedders which might work.

On the SwiftShader GitHub they also point to ANGLE:
`NOTE: SwiftShader's OpenGL ES implementation is no longer supported, and will be removed. Read more about our recommendation to use ANGLE on top of SwiftShader Vulkan here (aka. "SwANGLE").`
Do you think that might be an option for me to try to get working?

I'm willing to give it a go and share my experiences on the go, but it would help if you could point me in the right direction (or at least tell me if my thoughts are the right way to go).

1. Crosscompile SwiftShader for our target.
2. Crosscompile SwANGLE for our target.
3. Set ANGLE_DEFAULT_PLATFORM=swiftshader (from their github):
`Set ANGLE_DEFAULT_PLATFORM=swiftshader environment variable. When running your GLES application, ANGLE will use libvk_swiftshader as its Vulkan driver.`
4. Make sure flutter-pi uses the swiftshader opengl implementation (I'm not sure how to do this yet).

jwinarske, 2021-11-30:
Currently there is not a Flutter embedder that does this today.  The solution that I am proposing involves a Vulkan based embedder (coming soon) and Flutter Engine built with Vulkan and SwiftShader enabled.  The SwiftShader Front End = Vulkan, the backend can be either a frame buffer, a window, or an offscreen buffer.  So in the case of i.mx7 SwiftShader would be configured to render to a frame buffer.

So this involves PR not yet merged to master, with additional changes to configure SwiftShader backend.

I suppose it is possible to run one of the existing embedders via SwANGLE/SwiftShader, but I can't image that would be worthwhile if Vulkan is an option.  In the case of i.mx7 you would definitely want to avoid SwANGLE and simply use Vulkan embedder and Flutter Engine built with SwiftShader (BE=framebuffer).

Joel

jwinarske, 2021-11-30:
This thread indicates SwiftShader is being used in Android Things for devices  that don't have an (adequate) GPU (i.mx7):  https://groups.google.com/g/swiftshader/c/9Rq6hbgskTM?pli=1

So I would expect performance to be along the lines of Android Things running on i.mx7, and possibly better.

jwinarske, 2021-11-30:
Take a look at https://github.com/jwinarske/meta-vulkan.  If you add this to your Yocto build, and add `swiftshader` and `sascha-samples` to your image, you should be able to run the examples under `/usr/bin/vulkan-samples`.  This would be minimum requirement to run Vulkan based embedder with SoC that has no GPU.

jwinarske, 2022-02-01:
@dbeulen If you have a hardware setup to donate it would happen faster.

dbeulen, 2022-05-03:
@jwinarske Sorry, at this moment im not able to provide any hardware to donate. 

I had a very busy time the last few months, so i was not really able to dive into this. But this week i had some spare time to get things running.

Actually i managed to get the sony embedders and the toyota-homescreen embedders running with wayland display manager on the Imx7 device. Unfortunately the performance was not very good. Therefore i tried to get swiftshader running as you stated it would be the best luck into an usable implementation with acceptable performance. 

I saw that the flutter engine had merged some PR to support Vulkan. Do you know if anybody is allready working on an embedder with Vulkan support?

Thank you for al your great work!

jwinarske, 2022-05-03:
@dbeulen I have Vulkan BE support in the works.  I think the optimal scenario for your case would be DirectToDisplay Vulkan.  I'm currently working on Wayland/Vulkan.  At some point after this gets released I plan to add DirectToDisplay Vulkan support, which would be best case for Swiftshader.

jwinarske, 2022-09-08:
Adding a note.  There is a software rendering interface in the engine header file.  This is doable to implement but the tuning for CPU archa would take a fair amount of work.  It requires someone who is motivated to help the cause, or motivated by a full time job.  The BOM cost + NRE to develop support might be the same amortized BOM cost of a better GPU SoC (no NRE).

jwinarske, 2023-03-04:
Closing this as ivi-homescreen/flutter-auto runs on swiftshader. There would need to be some  perf tuning work in swiftshader as it does not leverage NEON instructions.  But it works, and that would unblock this request.

closed 2023-03-04 by jwinarske