meta-flutter/meta-flutter #63: no vc4.ko in honister-rpi3-64 images

t33x, 2022-02-12:
In github actions, your successful honister-rpi3-64 images do not contains the vc4 kernel module (there is no /lib/modules/..../drivers/gpu/vc4/vc4.ko). Is there a known solution to this problem?

jwinarske, 2022-02-12:
@t33x I just tested the honister-rpi3-64 DRM/EGL image.  It works fine:
```
flutter-pi --release /usr/share/gallery
```
Not sure what is a problem here.

t33x, 2022-02-12:
My mistake, I apologize. I trying the DPI interface to come to life (not working) and I thought this was the problem, but it's not. The kernel includes the vc4 driver (modules.builtin contains that). Now I think that the vc4-kms-dpi-generic overlay will be the solution to my problem... You can close this issue and again I apologize.

closed 2022-02-12 by jwinarske