meta-flutter/meta-flutter #84: Set default tag

rzr, 2022-03-12:
This layer wont work out of the box without setting the sdk tag currently:

FLUTTER_SDK_TAG = "2.10.2"

Would it be possible to just set AUTOREV in CI recipes and default to supported ones ?

Relate-to: https://gitlab.eclipse.org/eclipse/oniro-core/oniro/-/pipelines/2330

https://gitlab.eclipse.org/eclipse/oniro-core/oniro/-/merge_requests/4

jwinarske, 2022-03-17:
@rzr The LTS support I just merged into Dunfell has default set.  This will go into Honister branch soon.

jwinarske, 2022-03-19:
Default FLUTTER_SDK_TAG is set to 2.10.3 in latest honister commit.

closed 2022-03-19 by jwinarske

rzr, 2022-03-21:
Thanks would it be also possible create a branch for next LTS kirkstone (expected this coming april https://wiki.yoctoproject.org/wiki/Releases# )