meta-flutter/meta-flutter #105: Unable to build Flutter application with Yocto SDK

tprrt, 2022-04-12:
Hello,

I built a Yocto SDK including `nativesdk-flutter-sdk`, but it isn't possible to build Flutter application with the SDK because the command `flutter build bundle` is unable to read the version of Flutter SDK (provided by the Yocto SDK):
```
flutter build bundle
      The current Flutter SDK version is 0.0.0-unknown.
      Because gallery requires Flutter SDK version >=2.2.0, version solving failed.
      Running "flutter pub get" in gallery...
      pub get failed (1; Because gallery requires Flutter SDK version >=2.2.0, version solving failed.)

      flutter doctor -v
      Failed to find the latest git commit date: VersionCheckError: Command exited with code 128: git -c log.showSignature=false log -n 1 --pretty=format:%ad --date=iso
      Standard out:
      Standard error: error: object directory build/downloads/git2/github.com.flutter.flutter.git/objects does not exist; check .git/objects/info/alternates
      fatal: bad object HEAD
      Returning 1970-01-01 01:00:00.000 instead.
      [✓] Flutter (Channel unknown, 0.0.0-unknown, on Fedora Linux 35 (Workstation Edition) 5.16.18-200.fc35.x86_64, locale en_US.UTF-8)
          • Flutter version 0.0.0-unknown at sdk/nano/foo/sysroots/x86_64-nodistrosdk-linux/usr/share/flutter/sdk
          • Upstream repository unknown
          • Framework revision  (), 1970-01-01 01:00:00.000
          • Engine revision a83ed0e5e3
          • Dart version 2.16.1
          • DevTools version 2.9.2

```

It is easier to reproduce this issue having to build the Flutter gallery application:
```
git clone [git@github.com](mailto:git@github.com):flutter/gallery.git
cd gallery
git checkout 9eb785cb997ff56c46e933c1c591f0a6f31454f6

export FLUTTER_SDK="<folder>/sysroots/x86_64-nodistrosdk-linux/usr/share/flutter/sdk"
export PATH=${FLUTTER_SDK}/bin:$PATH
export ENGINE_SDK="./engine_sdk/sdk"

flutter build bundle
```

From the trace above, we can see the command try to access to a patch witch is only available from the Yocto build environment:
`      Standard error: error: object directory build/downloads/git2/github.com.flutter.flutter.git/objects does not exist; check .git/objects/info/alternates`

jwinarske, 2022-04-12:
@tprrt Thanks for reporting the issue.  I'll take a look.

jwinarske, 2022-04-20:
@tprrt See if this works for you:
```
$ export SDK_ROOT=<folder>/sysroots/x86_64-nodistrosdk-linux/usr/share/flutter/sdk
$ git config --global --add safe.directory $SDK_ROOT
$ sudo chmod a+rw $SDK_ROOT -R
$ sudo rm -rf $SDK_ROOT/bin/cache/pkg/sky_engine/
```

tprrt, 2022-04-25:
Hello @jwinarske,

I tried to use the commands above, without success, the same error occurs.

jwinarske, 2022-04-27:
@tprrt What branch of meta-flutter are you using?

tprrt, 2022-04-28:
Hello @jwinarske,

We are using the Honister branch (08b9a6c668b904c1abc2d769a2ace5697bea984b).

Kind regards,
Thomas

jwinarske, 2022-04-29:
@tprrt would you mind trying a build with the dunfell branch?  The steps I shared work on dunfell.
https://github.com/meta-flutter/meta-flutter/blob/dunfell/conf/layer.conf#L11

jwinarske, 2022-04-29:
Honister is not LTS so changing to bug

tprrt, 2022-04-29:
Hello @jwinarske,

The Honister branch uses the new syntax override; is there a trick to use the Dunfell branch of your layer with Honister?

```
ERROR: /src/layers/meta-flutter/recipes-graphics/flutter-engine/flutter-engine-release_git.bb: Variable do_configure_prepend contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.

```

Honister isn't a LTS but it is supported 

Do you have a plan to support Kirkstone?

We started our projet with Honister to be able to switch more quickly on the Kirkstone branch.

Kind regards,
Thomas

jwinarske, 2022-04-29:
@tprrt Currently Toyota is on Dunfell, and AGL has just moved to Kirkstone, and the default branch has been changed to Kirkstone as of yesterday.  I haven't had any requests or issues around Honister so it is lagging behind Dunfell.  I suspect in your case if I bring Honister to parity with Dunfell your issue will be resolved with the suggested fix.  

Normally for flutter development one directly uses the Flutter SDK off GitHub, checking out the desired SDK version.  One just needs to know what Flutter SDK Tag is  used in target image.  The standard approach flutter devs use for building (eFlutter) apps directly tracks the mobile/desktop model.  One just uses/create Yocto recipe to generate AOT/profile release images.  "flutter build bundle" works as-is on the target. Which is why I'm treating this as a lower priority bug.

I have not confirmed SDK scenario with Kirkstone yet, but I believe it's closer to parity with Dunfell than Honister.  In addition Kirkstone issues will be treated with a higher priority due to LTS customers (AGL, etc).

Hope this helps to clarify things a bit more.

Cheers,
Joel

tprrt, 2022-05-25:
Hello @jwinarske,

We tested the commands you proposed above, it works on Kirkstone.
Thank you for your support.

Kind regards,
Thomas

rzr, 2022-06-20:
May this relate to this issue:

VM snapshot invalid and could not be inferred from settings.

https://github.com/jwinarske/meta-flutter/issues/36

jwinarske, 2022-08-28:
Adding workaround to tools/setup_workspace_flutter.py.
Also honister is at parity with kirkstone.

jwinarske, 2023-03-17:
Closing this out as I have documented the safe.directory step in the README.md.
https://github.com/meta-flutter/meta-flutter/commit/cac4ad5f6e14fae8f9c272ae3a43f5662fb1d488

The other steps were added to flutter-sdk-native some time back.
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-sdk/flutter-sdk_git.bb#L60

closed 2023-03-17 by jwinarske