meta-flutter/meta-flutter #110: Unable to run apps on armv7 in release mode

timbz, 2022-05-05:
Hello,

I am trying to run flutter apps using the toyota embedder  on an i.MX6 (armv7). Debug (JIR) mode works but when I try to run the app in release mode (AOT) the app fails to start:

```
# homescreen --f --a=/usr/share/gallery
[INFO:display.cc(82)] No agl_shell extension present
[INFO:display.cc(86)] WL_SHM_FORMAT_XRGB32 not available
[INFO:display.cc(315)] Keyboard Present
[INFO:engine.cc(553)] Loading AOT: /usr/share/gallery/libapp.so
Illegal instruction (core dumped)
```

I guess this is probably related to [cross compilation](https://github.com/meta-flutter/meta-flutter/blob/kirkstone/classes/flutter-app.bbclass#L161). Any idea what could be the issue?

jwinarske, 2022-05-05:
@timbz 

Could be related to Wayland versions.  If would be very helpful if you build ivi-homescreen with:
```
EXTRA_OECMAKE += "-D CMAKE_SYSROOT=${STAGING_DIR_TARGET}/usr -D CMAKE_BUILD_TYPE=Debug"
```
Then share the log output.  Without this I can't be of much help.

Also what Yocto release, and what meta-flutter branch.  As well as what peripherals you have present (mouse, keyboard, touch).

Cheers

timbz, 2022-05-06:
```
# gdb --args homescreen --f --a=/usr/share/gallery
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-fslc-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from homescreen...Reading symbols from /usr/bin/.debug/homescreen...done.
done.
(gdb) run
Starting program: /usr/bin/homescreen --f --h=1080 --a=/usr/share/gallery
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
[INFO:main.cc(56)] Override Assets Path: /usr/share/gallery
[INFO:main.cc(80)] Fullscreen
[INFO:main.cc(143)] Screen Width: 1920
[INFO:main.cc(144)] Screen Height: 1080
[INFO:gl_resolver.cc(30)] dlopen: libEGL.so.1
[INFO:gl_resolver.cc(30)] dlopen: libGLESv2.so.2
[INFO:display.cc(47)] + Display()
[INFO:display.cc(127)] Wayland: wl_compositor
[INFO:display.cc(127)] Wayland: wl_subcompositor
[INFO:display.cc(127)] Wayland: wp_viewporter
[INFO:display.cc(127)] Wayland: wp_presentation
[INFO:display.cc(127)] Wayland: zwp_relative_pointer_manager_v1
[INFO:display.cc(127)] Wayland: zwp_pointer_constraints_v1
[INFO:display.cc(127)] Wayland: zwp_input_timestamps_manager_v1
[INFO:display.cc(127)] Wayland: wl_data_device_manager
[INFO:display.cc(127)] Wayland: wl_shm
[INFO:display.cc(127)] Wayland: wl_viv
[INFO:display.cc(127)] Wayland: zwp_linux_dmabuf_v1
[INFO:display.cc(127)] Wayland: wl_seat
[INFO:display.cc(127)] Wayland: wl_output
[INFO:display.cc(127)] Wayland: zwp_input_panel_v1
[INFO:display.cc(127)] Wayland: zwp_input_method_v1
[INFO:display.cc(127)] Wayland: zwp_text_input_manager_v1
[INFO:display.cc(127)] Wayland: zxdg_shell_v6
[INFO:display.cc(127)] Wayland: wl_shell
[INFO:display.cc(127)] Wayland: weston_desktop_shell
[INFO:display.cc(127)] Wayland: weston_screenshooter
[INFO:display.cc(82)] No agl_shell extension present
[INFO:display.cc(86)] WL_SHM_FORMAT_XRGB32 not available
[INFO:display.cc(89)] - Display()
[INFO:egl.cc(35)] + Egl()
[New LWP 2631]
[New LWP 2632]
[INFO:egl.cc(42)] EGL major: 1, minor 5
[INFO:egl.cc(52)] EGL has 60 configs
[INFO:egl.cc(69)] Buffer size for config 0 is 32
[INFO:egl.cc(85)] m_context[0] = 0xb49b0
[INFO:egl.cc(90)] m_resource_context[0] = 0x12dfb0
[INFO:egl.cc(96)] m_texture_context = 0x1a4258
[INFO:egl.cc(100)] - Egl()
[INFO:egl_window.cc(50)] + EglWindow()
[INFO:egl_window.cc(54)] EglWindow::m_surface = 0x21a3e8
[INFO:egl_window.cc(65)] EglWindow::m_shell_surface = 0x21a5a8
[INFO:display.cc(315)] Keyboard Present
[INFO:display.cc(217)] x: 0
[INFO:display.cc(218)] y: 0
[INFO:display.cc(219)] physical_width: -1
[INFO:display.cc(220)] physical_height: -1
[INFO:display.cc(221)] size: 1
[INFO:display.cc(222)] subpixel: 0
[INFO:display.cc(223)] make: unknown
[INFO:display.cc(224)] model: DISP4 BG
[INFO:display.cc(225)] transform: 0
[INFO:display.cc(246)] width: 1920
[INFO:display.cc(247)] height: 1080
[INFO:display.cc(248)] dpi: 1.33781e+07
[INFO:display.cc(282)] configuration is done
[INFO:egl_window.cc(79)] create egl_window: 1920x1080
[INFO:egl_window.cc(114)] - EglWindow()
[INFO:app.cc(60)] +App::App
[INFO:engine.cc(424)] PersistentCachePath: /home/root/.homescreen
[INFO:engine.cc(139)] (0) +Engine::Engine
[INFO:engine.cc(182)] (0) assets_path: /usr/share/gallery
[INFO:engine.cc(189)] (0) icu_data_path: /usr/share/flutter/icudtl.dat
[INFO:engine.cc(553)] Loading AOT: /usr/share/gallery/libapp.so
[INFO:engine.cc(238)] (0) -Engine::Engine
[INFO:engine.cc(288)] (0) +Engine::Run
[New LWP 2633]
[New LWP 2634]
[New LWP 2635]
[New LWP 2636]
[New LWP 2637]
[New LWP 2638]
[New LWP 2639]
[New LWP 2640]
[INFO:engine.cc(304)] (0) Engine::m_running = 1
[INFO:engine.cc(323)] (0) -Engine::Run
[INFO:engine.cc(345)] (0) SetWindowSize: width=1920, height=1080
[INFO:app.cc(88)] (0) Engine running...
[INFO:app.cc(147)] -App::App
[INFO:egl_window.cc(361)] ** handle_shell_configure_callback **

Thread 4 "io.flutter.ui" received signal SIGILL, Illegal instruction.
[Switching to LWP 2633]
0x6a574350 in ?? ()
(gdb) display/i $pc
1: x/i $pc
=> 0x6a574350:			; <UNDEFINED> instruction: 0xe7100c13
```
I am using the dunfell branch shoehorned into sumo. I also tried to remove the [arm_tune](https://github.com/meta-flutter/meta-flutter/blob/dunfell/recipes-graphics/flutter-engine/flutter-engine.inc#L127) options

jwinarske, 2022-05-06:
@timbz Stack trace? `bt`
I merged in xdg_shell support 10 days ago.  Please re-base and report back as that is what I'm currently working with; it also adds version info to Wayland output that is useful.
https://github.com/toyota-connected/ivi-homescreen

timbz, 2022-05-11:
The stack trace doesn't help as it is crashing in `libapp.so`. It seems `libapp.so` isn't properly compiled for arm 32bit. I read that you need to cross compile `gen_snapshot` using x86 32bit clang to work on armv7 but I didn't verify that. I ended up using `--runtime-mode jit_release`.

Thanks for the great work @jwinarske

closed 2022-05-11 by timbz

jwinarske, 2022-05-11:
@timbz The engine SDK generated in yocto build is built with clang.  You can optionally install to image.  Are you not using engine SDK to generate AOT?

jwinarske, 2022-05-11:
@timbz The STM32MP15 (armv7) Flutter Gallery app release build runs fine on target; meta-flutter CI artifact.  Should be no different for i.MX6.  I would narrow in on this, as running anything other than release is less than optimal.

dbeulen, 2022-07-12:
I am having the same problem. I am running homescreen on an i.mx6. As soon as the libapp.so starts the application logs a Illegal instruction.
```
[INFO:main.cc(97)] Bundle Path: /usr/share/flutter/gallery
[INFO:main.cc(106)] Backend Debug
[INFO:configuration.cc(184)] View Count: 1
[INFO:configuration.cc(232)] **********
[INFO:configuration.cc(233)] * Global *
[INFO:configuration.cc(234)] **********
[INFO:configuration.cc(235)] Application Id: .......... homescreen
[INFO:configuration.cc(239)] Cursor Theme: ............ 
[INFO:configuration.cc(240)] Disable Cursor: .......... false
[INFO:configuration.cc(242)] Debug Backend: ........... true
[INFO:configuration.cc(244)] ********
[INFO:configuration.cc(245)] * View *
[INFO:configuration.cc(246)] ********
[INFO:configuration.cc(253)] Bundle Path: .............. /usr/share/flutter/gallery
[INFO:configuration.cc(254)] Window Type: .............. NORMAL
[INFO:configuration.cc(255)] Size: ..................... 1920 x 720
[INFO:configuration.cc(257)] Fullscreen: ............... false
[INFO:configuration.cc(259)] Accessibility Features: ... 0
[INFO:display.cc(35)] + Display()
[INFO:display.cc(104)] Wayland: wl_compositor version 4
[INFO:display.cc(111)] 	Buffer Scale Enabled
[INFO:display.cc(104)] Wayland: wl_subcompositor version 1
[INFO:display.cc(104)] Wayland: wp_viewporter version 1
[INFO:display.cc(104)] Wayland: zxdg_output_manager_v1 version 2
[INFO:display.cc(104)] Wayland: wp_presentation version 1
[INFO:display.cc(104)] Wayland: zwp_alpha_compositing_v1 version 1
[INFO:display.cc(104)] Wayland: zwp_relative_pointer_manager_v1 version 1
[INFO:display.cc(104)] Wayland: zwp_pointer_constraints_v1 version 1
[INFO:display.cc(104)] Wayland: zwp_input_timestamps_manager_v1 version 1
[INFO:display.cc(104)] Wayland: wl_data_device_manager version 3
[INFO:display.cc(104)] Wayland: wl_shm version 1
[INFO:display.cc(104)] Wayland: weston_touch_calibration version 1
[INFO:display.cc(104)] Wayland: zwp_linux_dmabuf_v1 version 3
[INFO:display.cc(104)] Wayland: zwp_linux_explicit_synchronization_v1 version 2
[INFO:display.cc(104)] Wayland: wl_seat version 7
[INFO:display.cc(104)] Wayland: wl_viv version 1
[INFO:display.cc(104)] Wayland: wl_drm version 2
[INFO:display.cc(104)] Wayland: wl_output version 3
[INFO:display.cc(104)] Wayland: xdg_wm_base version 1
[INFO:display.cc(104)] Wayland: zxdg_shell_v6 version 1
[INFO:display.cc(104)] Wayland: wl_shell version 1
[INFO:display.cc(49)] agl_shell extension not present
[INFO:display.cc(52)] - Display()
[INFO:egl.cc(204)] dlopen: libEGL.so.1
[INFO:egl.cc(204)] dlopen: libGLESv2.so.2
[INFO:egl.cc(46)] EGL major: 1, minor 5
[INFO:egl.cc(56)] EGL has 40 configs
[INFO:egl.cc(539)] OpenGL ES Attributes:
[INFO:egl.cc(540)] 	EGL_VENDOR: "Vivante Corporation"
[INFO:egl.cc(542)] 	EGL_VERSION: "1.5"
[INFO:egl.cc(544)] 	EGL_CLIENT_APIS: "OpenGL_ES"
[INFO:egl.cc(546)] 	EGL_EXTENSIONS:
	    EGL_KHR_fence_sync, EGL_KHR_reusable_sync, EGL_KHR_wait_sync, 
	    EGL_KHR_image, EGL_KHR_image_base, EGL_KHR_image_pixmap, 
	    EGL_KHR_gl_texture_2D_image, EGL_KHR_gl_texture_cubemap_image, 
	    EGL_KHR_gl_renderbuffer_image, EGL_EXT_image_dma_buf_import, 
	    EGL_EXT_image_dma_buf_import_modifiers, EGL_KHR_lock_surface, 
	    EGL_KHR_create_context, EGL_KHR_no_config_context, 
	    EGL_KHR_surfaceless_context, EGL_KHR_get_all_proc_addresses, 
	    EGL_EXT_buffer_age, EGL_ANDROID_native_fence_sync, 
	    EGL_WL_bind_wayland_display, EGL_WL_create_wayland_buffer_from_image, 
	    EGL_KHR_partial_update, EGL_EXT_swap_buffers_with_damage, 
	    EGL_KHR_swap_buffers_with_damage, EGL_EXT_pixel_format_float
[INFO:egl.cc(559)] EGL framebuffer configurations:
[INFO:egl.cc(561)] 	Configuration #0:

[INFO:egl.cc(566)] 		EGL_CONFIG_ID: 1
[INFO:egl.cc(573)] 		EGL_CONFIG_CAVEAT: EGL_NONE
[... 1723 lines cut ...]
[INFO:display.cc(253)] Pointer Present
[INFO:display.cc(262)] Keyboard Present
[INFO:display.cc(271)] Touch Present
[INFO:display.cc(286)] Seat: default
[INFO:display.cc(197)] Physical width: -1 mm x -1 mm
[INFO:display.cc(226)] Display Scale Factor: 1
[INFO:display.cc(214)] Video mode: 1024 x 600 @ 68.655 Hz
[INFO:display.cc(540)] [keyboard repeat info] rate: 40, delay: 400
[INFO:window.cc(112)] - WaylandWindow()
[INFO:app.cc(69)] +App::App
[INFO:engine.cc(449)] PersistentCachePath: /home/root/.homescreen
[INFO:engine.cc(97)] (0) +Engine::Engine
[INFO:engine.cc(145)] (0) flutter_assets: /usr/share/flutter/gallery/data/flutter_assets
[INFO:engine.cc(169)] (0) icudtl.dat: /usr/share/flutter/icudtl.dat
[INFO:engine.cc(583)] Loading AOT: /usr/share/flutter/gallery/lib/libapp.so
[INFO:engine.cc(222)] (0) -Engine::Engine
[INFO:engine.cc(272)] (0) +Engine::Run
[INFO:engine.cc(203)] (0) Running Task
[INFO:engine.cc(288)] (0) Engine::m_running = 1
[INFO:engine.cc(312)] (0) -Engine::Run
[INFO:engine.cc(337)] (0) SetWindowSize: width=1024, height=600
[INFO:app.cc(90)] (0) Engine running...
[INFO:app.cc(149)] -App::App
Illegal instruction
```

jwinarske, 2022-07-12:
@dbeulen. I need more context
1. Are you building/running with transparency enabled or disabled?  Low end SoCs I disable transparency to start; test enabled later.
2. What Flutter SDK version are you using?
3. What commit of ivi-homescreen are you using?
4. Output of `readelf -h libapp.so`?

dbeulen, 2022-07-19:
@jwinarske Thank you for your help! Really appreciate it. 

1. I guess i have transparency enabled. (i do have a png with transparent background in my app which appears to be transparent). How could i disable the transparency?
2. I am using flutter sdk 3.0.4 within yocto to compile the application.
3. latest ivi homescreen commit: e0ede14ab7fbb013f37a2e62b655c8bedacb0c63
4.
```
  ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          52 (bytes into file)
  Start of section headers:          23494808 (bytes into file)
  Flags:                             0x5000400, Version5 EABI, hard-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         6
  Size of section headers:           40 (bytes)
  Number of section headers:         11
  Section header string table index: 10```

jwinarske, 2022-07-20:
@dbeulen 

Create a ivi-homescreen-<runtime variant>_git.bbappend with the minimum contents:
```
PACKAGECONFIG:remove = "transparency"
```

dbeulen, 2022-07-25:
Thank you, unfortunately building it without transparency did not help. Any idea what i could try next?

jwinarske, 2022-07-25:
@dbeulen What type of engine build are you running?  Do you have the unoptimized flag set?  If not try running this build variant.  It adds debug asserts which may be useful.