meta-flutter/meta-flutter #112: flutter-sdk_git.bb is not available because HTTPS_PROXY is not set.

WAKazuyoshiAkiyama, 2022-05-11:
If flutter-sdk_git.bb is used in the site with the specific proxy setting, since flutter-sdk cannot be fetched, the build will be failed.

So, to my mind, the setting for https proxy like `export https_proxy=${https_proxy}` is required in common_compile() of flutter-sdk_git.bb like the followings.

```
common_compile() {

    export CURL_CA_BUNDLE=${STAGING_DIR_NATIVE}/etc/ssl/certs/ca-certificates.crt
    export PATH=${S}/bin:$PATH
    export PUB_CACHE=${S}/.pub-cache

    export http_proxy=${http_proxy}
    export https_proxy=${https_proxy}

    bbnote "Flutter SDK: ${FLUTTER_SDK_TAG}"

    flutter config --no-enable-android
    flutter config --no-enable-ios
    flutter config --no-enable-web
    flutter config --no-enable-linux-desktop
    flutter config --enable-custom-devices

    bbnote `flutter doctor -v`
}
```

Can I have your opinions? If ok, I will create the PR.

jwinarske, 2022-05-12:
@WAKazuyoshiAkiyama I don't have an environment that requires http/https proxy, or one I can test with.  So please create PR.

WAKazuyoshiAkiyama, 2022-05-13:
@jwinarske I created the PR. https://github.com/meta-flutter/meta-flutter/pull/113
In the environment requiring the proxy, the same change is added in bbappend, and it is tested.

WAKazuyoshiAkiyama, 2022-05-13:
@jwinarske Thank you for merging.

closed 2022-05-13 by WAKazuyoshiAkiyama