meta-flutter/meta-flutter #114: Question - Accessing Flutter over Web Interface For i.MX6ULL

79rpm, 2022-05-17:
Hello,

I am looking at using Flutter on i.MX6ULL. I am building the BSP with Yocto. It is a headless system with Ethernet access. Where can I find some documentation on how to build Flutter in that scenario? The backend will most likely be node.js.

Thank you,

jwinarske, 2022-05-17:
@79rpm You would need to build the Flutter UI as a web interface, then implement a HTTP server on your device to host the pages.  See general docs for web support: https://flutter.dev/multi-platform/web

No idea what your skill level is, but I have used this server on a few projects with great success:
https://h2o.examp1e.net/

Closing issue as this query has nothing to do with meta-flutter.

79rpm, 2022-05-18:
@jwinarske thank you for the details and confirmation. I can go for it!