meta-flutter/meta-flutter #117: bitbake core-image-base fails with nothing provides /usr/bin/env

bojidartonchev, 2022-06-08:
Hello,

First I want to say that this is my first step in Yocto, so probably there is something missed on my side, but I am stuck on that and can't find a solution. So I am trying to build an Yocto image with meta-flutter for raspberry pi 4. For all of my layers I am using kirkstone branches. Before adding meta-flutter I successfully built an image and ran it on my Pi.

Any help would be appreciated.

Best regards.

```
ERROR: core-image-base-1.0-r0 do_rootfs: Could not invoke dnf. Command '/mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/yum.repos.d --installroot=/mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs --setopt=logdir=/mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/temp --repofrompath=oe-repo,/mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo --nogpgcheck install flutter-engine-release flutter-gallery-release flutter-pi-release psplash-raspberrypi run-postinsts locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.11.1
cachedir: /mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo
User-Agent: falling back to 'libdnf': could not detect OS or basearch
repo: using cache for: oe-repo
oe-repo: using metadata from Wed 08 Jun 2022 03:13:09 PM UTC.
Last metadata expiration check: 0:00:01 ago on Wed 08 Jun 2022 03:13:09 PM UTC.
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: package flutter-pi-release-git-r0.cortexa7t2hf_neon_vfpv4 requires libinput.so.10, but none of the providers can be installed
  - package flutter-pi-release-git-r0.cortexa7t2hf_neon_vfpv4 requires libinput.so.10(LIBINPUT_0.12.0), but none of the providers can be installed
  - package flutter-pi-release-git-r0.cortexa7t2hf_neon_vfpv4 requires libinput.so.10(LIBINPUT_0.21.0), but none of the providers can be installed
  - package flutter-pi-release-git-r0.cortexa7t2hf_neon_vfpv4 requires libinput.so.10(LIBINPUT_1.11), but none of the providers can be installed
  - package flutter-pi-release-git-r0.cortexa7t2hf_neon_vfpv4 requires libinput10 >= 1.19.4, but none of the providers can be installed
  - conflicting requests
  - nothing provides /usr/bin/env needed by libinput10-1.19.4-r0.cortexa7t2hf_neon_vfpv4
(try to add '--skip-broken' to skip uninstallable packages)
 
ERROR: Logfile of failure stored in: /mnt/data/Projects/kribo/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_rootfs.3443074
ERROR: Task (/mnt/data/Projects/kribo/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'
```

jwinarske, 2022-06-08:
I haven't put together a manifest for RPI4 on Kirkstone yet.  The focus has been the lowest common denominator Zero2.

For Dunfell see:  https://github.com/meta-flutter/meta-flutter/blob/dunfell/.github/workflows/dunfell-rpi4-64.yml
You can also download an image to test.

jwinarske, 2022-06-09:
This build has Flutter SDK 3.0.1: https://github.com/meta-flutter/meta-flutter/actions/runs/2464971404

closed 2022-06-09 by jwinarske