meta-flutter/meta-flutter #121: Is Zeus (3.0) supported?

LeoBound, 2022-06-13:
Hi,

I have been trying to build an image for Zeus, with a fresh Poky install, meta-clang on the zeus branch and meta-flutter on dunfell.

When trying to execute bitbake core-image-base I get:
```
ERROR: ParseError at /home/lbound/zeus/meta-flutter/recipes-graphics/toyota/ivi-homescreen.inc:31: Could not inherit file classes/features_check.bbclass
```

It looks like this is because features_check was first added in dunfell, replacing distro_features_check?
Am I doing something incorrectly or is Zeus not supported (it is present in layer.conf)?

Thanks

jwinarske, 2022-06-15:
@LeoBound Hi there.  Yeah I haven't flushed out Zeus at this time.  If you are aware of the differences I can work with you on it.

jwinarske, 2022-07-14:
@LeoBound I would stick to Dunfell or Kirkstone which are both LTS; better support for security updates.

LeoBound, 2022-07-15:
@jwinarske I'd like to but unfortunately the hardware I have only has a board support package for Warrior

jwinarske, 2022-07-15:
@LeoBound can you share what the SoC is?   If not then at least machine arch.  I may have a reference dev board I could test on.  Also what embedder & gfx stack?

LeoBound, 2022-07-15:
@jwinarske Sure, it's got an Freescale I.MX6 Dual (armv7), has Weston/Wayland and uses EGL for graphics. 

Originally I was going to use flutter-pi, but the default yocto package doesn't use systemd which is required; I thought I might as well modify it and, if I was going to do that, add meta-flutter at the same time.

LeoBound, 2022-08-12:
I've managed to get a successful build on Warrior now with gallery-debug, flutter-engine debug and flutter-wayland-client (with the debug flags set in the local.conf).

For anyone else interested in doing this the main changes I had to make were replacing features_check -> distro_features_check in any recipes I wanted to build, make sure you have gcc-multilib, and the first build I did would fail due to some missing includes in the LLVM source.   
This can be fixed by:
Open `{build_directory}/tmp/work-shared/llvm-project-source-8.0.1-r0/git/llvm/include/llvm/Demangle/MicrosoftDemangleNodes.h`

Below the other includes add:   
`#include <cstdint>`  
`#include <string>`

Then continue the build.

Unfortunately the ivi-homescreen recipe wouldn't build for me due to the CMake version being too old for one of the dependencies.

closed 2022-08-12 by LeoBound