meta-flutter/meta-flutter #122: _get_flutter_release_info in case of no network

witekio-kwe, 2022-06-22:
Looking at
```python
    ret = ""
    flutter_sdk_tag = d.getVar("FLUTTER_SDK_TAG")

    try:
        response = urllib.request.urlopen( \
            'https://storage.googleapis.com/flutter_infra_release/releases/releases_linux.json' \
            )
    except urllib.error.URLError as e:
        bb.fatal('HTTP error: %s' % (e))
    except (ValueError, TypeError) as e:
        bb.fatal('JSON error: %s' % (e))

    buf = response.read()
    releases_linux_json = json.loads(buf.decode('utf-8'))

    if flutter_sdk_tag == "AUTOINC":
        flutter_sdk_tag = releases_linux_json["current_release"]["stable"]
 ```
 
I don't see a way to get a fixed revision for the e.g. SDK when the build host has no network connectivity at all.
Wouldn't it better to have a way to bail out early *before* making any network call, if the user can specify the exact resource they want to pull?

As the SRC_URI in the sdk recipe for instance expands already while parsing, there's not even a proper way to override the SRC_URI by a bbappend - only way I see is to override the whole include file by a layer that is placed into BBLAYERS after meta-flutter

An alternative fix would be to have always use ??= operator on all variables that make use of include file's code, so I could override the decisions from distro or local.conf

jwinarske, 2022-06-22:
FLUTTER_SDK_TAG origin, see: https://github.com/meta-flutter/meta-flutter/blob/kirkstone/conf/include/flutter-version.inc

The Flutter SDK is cached in DL_DIR.

witekio-kwe, 2022-06-23:
But still of these functions do a network call before the settings of the variables are used right?
what I had now a couple of times in our CI is that the network call timed out or was responded with an temporary error, leading to a bb.fatal and therefore a failing build, even though we have the needed sources cached in our DL_DIR

witekio-kwe, 2022-06-24:
Thank you - the network access is sadly needed - but what would like to have addressed is that we could get rid of this network call when for instance a tag is defined (or even better a sha hash)

Something like

```python
    flutter_sdk_tag = d.getVar("FLUTTER_SDK_TAG")
    if flutter_sdk_tag != 'FROZEN':
        with urllib.request.urlopen('https://storage.googleapis.com/flutter_infra_release/releases/releases_linux.json') as f:
            releases_linux_json = json.load(f)
            if flutter_sdk_tag == "AUTOINC":
                flutter_sdk_tag = releases_linux_json["current_release"]["stable"]

            releases = releases_linux_json["releases"]
            for r in releases:
                if r["version"] == flutter_sdk_tag or r["hash"] == flutter_sdk_tag:
                    ret = r[key]
                    break
    else:
        ret = d.getVar("FLUTTER_SDK_FROZEN_HASH")
    if ret == "":
        raise ValueError("Could not get flutter sdk archive url")

    return ret
```

I could prepare a PR if that would be something accepted by the project

jwinarske, 2022-06-24:
I think it's a bit more complicated than this.  It needs to know the correlation between Flutter SDK and engine commit.

That needs to be stored and accessed appropriately.

PRs are welcome.  Running through the CI matrix for multiple branches with a change like this is not trivial and needs to be carefully thought out.

jwinarske, 2022-06-24:
The engine commit can be looked up in the flutter SDK, using path bin/internal/engine.version.

jwinarske, 2022-06-24:
I think the right thing would be to cache the JSON results to a file in DL_DIR.  If no network then fallback to resolving versions from that.  As if it were results from a live lookup.  Looking up engine from SDK introduces dependencies best avoided.

jwinarske, 2022-07-06:
@witekio-kwe Are you going to submit a PR?

WAKazuyoshiAkiyama, 2022-07-06:
From my understanding, there are two ways to get the flutter sdk.
The first is to use https://github.com/flutter/flutter, but, from my experience, the revision hash of this repo sometimes changes (is overwritten).
So, this is an unstable way.

The second is to use the tar archive of `https://storage.googleapis.com/flutter_infra_release/releases/${@get_flutter_archive(d)}`.
At least, this url is not changed from my experience, so more stable than the first.

The previous flutter-sdk_git.bb uses the first way, but, due to the above reason, I changed flutter-sdk_git.bb to use the second way.

This is the background of flutter-version.inc.

Basically, I agree [the Joel's opinion](https://github.com/meta-flutter/meta-flutter/issues/122#issuecomment-1165466828).
But, I think `${TMPDIR}/work-shared/flutter/` is better than `${DL_DIR}` because `${DL_DIR}` is the directory for do_fetch i.e. bb.fetcher classes.
But, to use flutter-sdk_git.bb on no network env, DL_DIR is more suitable, so this is not a strong opinion.

witekio-kwe, 2022-07-06:
> @witekio-kwe Are you going to submit a PR?

I'm not sure - as of now we helped ourselves with fixing the revisions using bbappends.
We will likely start to play around with the discussed options, but that will take some time.

It will (as you rightfully mentioned) not solve all of the issues (network access while compiling), so it still feels like a workaround to me.

Anyway I will get back with a PR at some point later this year.
For now I'm fine with set this issue here to resolved

witekio-kwe, 2022-12-14:
Been a while, but we worked on enabling such a thing and ended up with making `releases_linux.json` a static asset of layer and creating a mapping file that for the engine revision SHA.
These files can be created by running a script.

Would that be something of interest to the project?

jwinarske, 2022-12-14:
@witekio-kwe I'm interested in taking a look at a rough PR.   I may use it as a baseline.  Some other things I'm thinking about are archiver.bbclass, and how to best capture patches for a specific flutter version, making a generic gun fetcher to build any gun project

jwinarske, 2022-12-14:
Closed it by accident

witekio-kwe, 2022-12-14:
> @witekio-kwe I'm interested in taking a look at a rough PR. I may use it as a baseline. Some other things I'm thinking about are archiver.bbclass, and how to best capture patches for a specific flutter version, making a generic gn fetcher to build any gn project. The other is a fair amount of flutter projects tend to fetch assets during compile, which is terrible for LTS.

Will try to compile a draft this week

witekio-kwe, 2022-12-14:
@jwinarske https://github.com/meta-flutter/meta-flutter/pull/229 - please let me know what you think - we have this code now in our setup for almost a 1.5 months and it’s behaving as expected, but there might be edge cases I do not see, as we mainly operate on release versions that do not change that often

WAKazuyoshiAkiyama, 2022-12-15:
While installing eSDK, I faced on the error related to this issue.

```
ERROR: ExpansionError during parsing /workdir/eSDK/layers/meta-flutter/recipes-graphics/flutter-apps/flutter-test-plugins_git.bb
Traceback (most recent call last):
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 525, in OpenerDirector.open(fullurl='https://storage.googleapis.com/flutter_infra_release/releases/releases_linux.json', data=None, timeout=<object object at 0x7f7cabe04f00>):
             sys.audit('urllib.Request', req.full_url, req.data, req.headers, req.get_method())
    >        response = self._open(req, data)

  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 542, in OpenerDirector._open(req=<urllib.request.Request object at 0x7f7ca9313f40>, data=None):
             protocol = req.type
    >        result = self._call_chain(self.handle_open, protocol, protocol +
                                       '_open', req)
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 502, in OpenerDirector._call_chain(chain={'unknown': [<urllib.request.UnknownHandler object at 0x7f7ca9313fd0>], 'http': [<urllib.request.HTTPHandler object at 0x7f7ca930ddc0>], 'ftp': [<urllib.request.FTPHandler object at 0x7f7ca930df70>], 'file': [<urllib.request.FileHandler object at 0x7f7ca930d4c0>], 'data': [<urllib.request.DataHandler object at 0x7f7ca930d790>], 'https': [<urllib.request.HTTPSHandler object at 0x7f7ca930d130>]}, kind='https', meth_name='https_open', *args=(<urllib.request.Request object at 0x7f7ca9313f40>,)):
                 func = getattr(handler, meth_name)
    >            result = func(*args)
                 if result is not None:
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 1397, in HTTPSHandler.https_open(req=<urllib.request.Request object at 0x7f7ca9313f40>):
             def https_open(self, req):
    >            return self.do_open(http.client.HTTPSConnection, req,
                     context=self._context, check_hostname=self._check_hostname)
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 1357, in HTTPSHandler.do_open(http_class=<class 'http.client.HTTPSConnection'>, req=<urllib.request.Request object at 0x7f7ca9313f40>, **http_conn_args={'context': None, 'check_hostname': None}):
                 except OSError as err: # timeout error
    >                raise URLError(err)
                 r = h.getresponse()
bb.data_smart.ExpansionError: Failure expanding variable do_install, expression was
    install -d /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/data/flutter_assets
    cp -r /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/git/plugins/build/flutter_assets/* /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/data/flutter_assets/

    if ${@bb.utils.contains('FLUTTER_RUNTIME', 'release', 'true', 'false', d)} || \
        ${@bb.utils.contains('FLUTTER_RUNTIME', 'profile', 'true', 'false', d)}; then

        bbnote "Flutter Application: Installing debug"

        install -d /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/lib
        cp /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/git/plugins/libapp.so /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/lib/
    fi

    if [[ "bundle" =~ .*"linux".* ]]; then

        bbnote "Flutter Application: Installing GTK bundle"

        if [ -n "${FLUTTER_REMOVE_LINUX_BUILD_ARTIFACTS}" ]; then
            rm /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/linux_plugins | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/lib/*.so | true
        else
            # expecting default "release" build
            mv /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/linux_plugins /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/bin/ | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/bin/linux_plugins | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/lib/*.so | true
        fi
    fi
    install -D -m0644 /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/lsb-release \
        /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/etc/lsb-release
    sed -i "s|@FLUTTER_SDK_TAG@|${@get_flutter_sdk_version(d)}|g" /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/etc/lsb-release
 which triggered exception URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1131)>
```

Since the network access happens while parsing recipes but somewhy HTTPS access cannot be executed in eSDK installation, the above error is caused.

I think https://github.com/meta-flutter/meta-flutter/pull/229 may solve the above error also.

At least, we need to add something like the following before `urllib.request.urlopen` in order to support Yocto eSDK if using `urllib.request.urlopen`.

```
    import ssl
    ssl._create_default_https_context = ssl._create_unverified_context
```

If https://github.com/meta-flutter/meta-flutter/pull/229  is not merged, I will need to create the PR to add the above.

WAKazuyoshiAkiyama, 2023-01-20:
@jwinarske 
I'm not sure that your plan can solve the issue of eSDK.

```
    import ssl
    ssl._create_default_https_context = ssl._create_unverified_context
```

The above configuration or a similar one will be added in your plan?

While install eSDK, DL_DIR may be empty, so, only setting SSL context unverified or using https://github.com/meta-flutter/meta-flutter/pull/229 can solve the issue of eSDK.

jwinarske, 2023-01-20:
@WAKazuyoshiAkiyama 
What recipes need to be installed in SDK to replicate your issue?

WAKazuyoshiAkiyama, 2023-01-25:
@jwinarske 
flutter-engine.
The issue of eSDK will be caused if there is a recipe accessing to internet when parsing it.
And, such a recipe does not need to be installed.
If such a recipe is a target of bitbake parsing, the issue will be caused.

jwinarske, 2023-03-17:
@WAKazuyoshiAkiyama when you have a chance, the change is merged into kirkstone.  Anything else unresolved here?

WAKazuyoshiAkiyama, 2023-03-17:
@jwinarske 
For the network accessing while parsing recipes, I know only the issue on Yocto eSDK (ext-sdk). So, I have no other issues.

jwinarske, 2023-03-17:
@WAKazuyoshiAkiyama let's open a new issue specific to eSDK issue.  I will close this, as this is issue as titled is resolved.

jwinarske, 2023-03-17:
closing since the initial issue has been merged/resolved.  I moved eSDK issue here:
https://github.com/meta-flutter/meta-flutter/issues/268

closed 2023-03-17 by jwinarske