meta-flutter/meta-flutter #124: Avoid installing stuff in $HOME dir

pberginkonsult, 2022-07-04:
When building inside a container I get his error:

```
ERROR: demo-gui-1.01-r0 do_archive_pub_cache: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_archive_pub_cache(d)
     0003:
File: '/workdir/sources/meta-flutter/classes/flutter-app.bbclass', lineno: 83, function: do_archive_pub_cache
     0079:        '%s/bin/flutter pub get --offline' % \
     0080:        (pub_cache, flutter_sdk, flutter_sdk)
     0081:
     0082:    bb.note("Running %s in %s" % (pub_cache_cmd, app_root))
 *** 0083:    runfetchcmd('%s' % (pub_cache_cmd), d, quiet=False, workdir=app_root)
     0084:
     0085:    cp_cmd = \
     0086:        'mkdir -p %s/.project | true; ' \
     0087:        'cp -r .dart_tool %s/.project/ | true; ' \
File: '/workdir/sources/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 929, function: runfetchcmd
     0925:                bb.utils.remove(f, True)
     0926:            except OSError:
     0927:                pass
     0928:
 *** 0929:        raise FetchError(error_message)
     0930:
     0931:    return output
     0932:
     0933:def check_network_access(d, info, url):
Exception: bb.fetch2.FetchError: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/workdir/build/imx8mp-lpddr4-evk/tmp/sysroots-uninative/x86_64-linux/usr/bin:/workdir/sources/poky/scripts:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot/usr/bin/crossscripts:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/usr/sbin:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/usr/bin:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/sbin:/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/bin:/workdir/sources/poky/bitbake/bin:/workdir/build/imx8mp-lpddr4-evk/tmp/hosttools"; export HOME="/home/pokyuser"; export PUB_CACHE=/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/pub_cache; /workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/flutter pub get;/workdir/build/imx8mp-lpddr4-evk/tmp/work/cortexa53-crypto-poky-linux/demo-gui/1.01-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/flutter pub get --offline failed with exit code 255, output:
Unhandled exception:
Exception: Flutter failed to create a directory at "/home/pokyuser/.config/flutter".
Please ensure that the SDK and/or project is installed in a location that has read/write permissions for the current user.

#0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
#1      _throwFileSystemException (package:flutter_tools/src/base/error_handling_io.dart:820:3)
#2      _handlePosixException (package:flutter_tools/src/base/error_handling_io.dart:761:3)
#3      _runSync (package:flutter_tools/src/base/error_handling_io.dart:607:7)
#4      ErrorHandlingDirectory.createSync (package:flutter_tools/src/base/error_handling_io.dart:432:12)
#5      new Config._common (package:flutter_tools/src/base/config.dart:76:17)
#6      new Config (package:flutter_tools/src/base/config.dart:37:19)
#7      new _DefaultPersistentToolState (package:flutter_tools/src/persistent_tool_state.dart:65:18)
#8      new PersistentToolState (package:flutter_tools/src/persistent_tool_state.dart:21:9)
#9      runInContext.<anonymous closure> (package:flutter_tools/src/context_runner.dart:300:34)
#10     AppContext._generateIfNecessary.<anonymous closure> (package:flutter_tools/src/base/context.dart:104:42)
#11     _LinkedHashMapMixin.putIfAbsent (dart:collection-patch/compact_hash.dart:524:23)
#12     AppContext._generateIfNecessary (package:flutter_tools/src/base/context.dart:92:20)
#13     AppContext.get (package:flutter_tools/src/base/context.dart:121:32)
#14     PersistentToolState.instance (package:flutter_tools/src/persistent_tool_state.dart:35:55)
#15     persistentToolState (package:flutter_tools/src/globals.dart:77:69)
#16     _defaultBotDetector (package:flutter_tools/src/globals.dart:83:24)
#17     _defaultBotDetector (package:flutter_tools/src/globals.dart)
#18     botDetector (package:flutter_tools/src/globals.dart:79:62)
#19     isRunningOnBot (package:flutter_tools/src/globals.dart:89:36)
#20     runInContext.runnerWrapper (package:flutter_tools/src/context_runner.dart:81:34)
#21     runInContext.runnerWrapper (package:flutter_tools/src/context_runner.dart:80:28)
#22     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:29)
#23     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:7)
#24     _rootRun (dart:async/zone.dart:1426:13)
#25     _CustomZone.run (dart:async/zone.dart:1328:19)
#26     _runZoned (dart:async/zone.dart:1861:10)
#27     runZoned (dart:async/zone.dart:1783:10)
#28     AppContext.run (package:flutter_tools/src/base/context.dart:149:12)
#29     runInContext (package:flutter_tools/src/context_runner.dart:85:18)
#30     run (package:flutter_tools/runner.dart:45:10)
#31     main (package:flutter_tools/executable.dart:94:16)
#32     main (file:///b/s/w/ir/x/w/archive/flutter/packages/flutter_tools/bin/flutter_tools.dart:10:14)
#33     _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:295:32)
#34     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:192:12)
[... 40 lines cut ...]
Flutter is putting its configuration files in the path of `XDG_CONFIG_HOME` (https://github.com/flutter/flutter/pull/66645). A possible solution to this problem is to set `XDG_CONFIG_HOME` to a location inside the working dir.

I'm working on kirkstone branch and currently using commit 7d905a8a280cd2cf21ea9252b1ce4d070fe9dc96. This is a solution to my problem:

```
$ git diff
diff --git a/classes/flutter-app.bbclass b/classes/flutter-app.bbclass
index d638c9f..f68f62b 100644
--- a/classes/flutter-app.bbclass
+++ b/classes/flutter-app.bbclass
@@ -75,9 +75,10 @@ python do_archive_pub_cache() {
 
     pub_cache_cmd = \
         'export PUB_CACHE=%s; ' \
+        'export XDG_CONFIG_HOME=%s;' \
         '%s/bin/flutter pub get;' \
         '%s/bin/flutter pub get --offline' % \
-        (pub_cache, flutter_sdk, flutter_sdk)
+        (pub_cache, workdir, flutter_sdk, flutter_sdk)
 
     bb.note("Running %s in %s" % (pub_cache_cmd, app_root))
     runfetchcmd('%s' % (pub_cache_cmd), d, quiet=False, workdir=app_root)
```

Is this something we can add to the layer? Or can we find other solutions to the problem and avoid cluttering users home dir?

jwinarske, 2022-07-06:
@pberginkonsult 

Hi Peter.  The Kirkstone CI jobs are running with this Docker container:
https://github.com/meta-flutter/docker-containers/blob/main/ubuntu-20-dev/Dockerfile

What OS/version are you using for your container?

pberginkonsult, 2022-07-06:
> @pberginkonsult
> 
> Hi Peter. The Kirkstone CI jobs are running with this Docker container: https://github.com/meta-flutter/docker-containers/blob/main/ubuntu-20-dev/Dockerfile
> 
> What OS/version are you using for your container?

Im using crops/poky with Ubuntu 22.04. https://hub.docker.com/layers/poky/crops/poky/ubuntu-22.04/images/sha256-3275936ec9b026b7c598066f195975c3da35b9cdc4451b771fca521addd61b78?context=explore

jwinarske, 2022-07-06:
@pberginkonsult i haven't looked at the flutter code.  This applies to both `flutter build bundle` and `flutter build linux`?

If the SDK and App have two config locations that leads to redundancy, and slower builds.  As SDK and each app would recreate this folder.

The benefit of using the container is it's as common location for all the recipes.  One could map the config folder to container to preserve it.  Another is to archive it like the pub cache or SDK.

So the change would degrade build perf.

My preference would be to archive/restore it for each recipe, only these jobs run in parallel.

What's the concern about writing to $HOME, as this happens with any flutter development.

pberginkonsult, 2022-08-08:
Hi @jwinarske,
sorry for delayed response.

> @pberginkonsult i haven't looked at the flutter code. This applies to both `flutter build bundle` and `flutter build linux`?
>
> If the SDK and App have two config locations that leads to redundancy, [...]
If this change introduce degraded performance it is not good and need to be looked at deeper.
> My preference would be to archive/restore it for each recipe, only these jobs run in parallel.
> 
I haven't that deep knowledge in the flutter build process yet. But if I understand your response correct I think the `.config/flutter` dir should be put in sysroot. Do you think this is possible?
> What's the concern about writing to $HOME, as this happens with any flutter development?
My main concern is that it brakes the whole idea of a sandboxed development environment if some parts is placed in the users HOME dir.

jwinarske, 2022-08-09:
I'm not against the idea of sandboxing .config.  The approach would need to archive the .config in SDK compile append.  Then in apps pre-compile unarchives it to ${WORKDIR} and set XDG_CONFIG_HOME to extracted folder.

jwinarske, 2023-03-05:
@pberginkonsult 
https://github.com/meta-flutter/meta-flutter/pull/256

Thanks for pointing this out!

closed 2023-03-05 by jwinarske