meta-flutter/meta-flutter #130: How to set master branch

dbeulen, 2022-07-12:
I would like to use custom devices to debug our application (https://github.com/flutter/flutter/wiki/Using-custom-embedders-with-the-Flutter-CLI). However, custom devices is only available on the master branch. 

When i build my app using the master branch it will not run on the device because the device is on the latest stable release. See error below. How can i configure meta-flutter so it uses the master branch?

Error when running debug version from custom-devices:
```
[ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
```

jwinarske, 2022-07-13:
@dbeulen The Flutter SDK Version you build your bundle with needs to match the engine build used with the embedder.  If they do not match, you will see this error.

jwinarske, 2022-07-13:
Also it's fairly trivial to patch the flutter SDK to enable custom-devices.

jwinarske, 2022-07-13:
Closing as this is unrelated to meta-flutter.

closed 2022-07-13 by jwinarske