meta-flutter/meta-flutter #136: flutter-engine.inc

bernardcraddock, 2022-07-23:
Do we need to add the branch in SRC_URI ?

SRC_URI = "gn://github.com/flutter/engine.git;branch=main;name=src/flutter \

SRC_URI = "gn://github.com/flutter/engine.git;name=src/flutter \
           file://0001-clang-toolchain.patch \
           file://0002-x64-sysroot-assert.patch \
           file://0001-allow-deprecated-calls.patch \
           file://0001-remove-x11-dependency.patch \
           file://0001-prevent-redefinition-of-glib_autoptr_clear_AtkObject.patch \
           "

jwinarske, 2022-07-23:
Nope.  Notice the `gn:` prefix.  It uses a custom fetcher included in the layer.

bernardcraddock, 2022-07-24:
yep i did notice it, maybe i'm doing something else incorrect ?
Building on AGL/master as follows it failed with Unable to fetch URL, after poking around i amended branch=main. retstarted bitbake and it completed ok and built the image

  $ source meta-agl/scripts/aglsetup.sh -f -m qemux86-64 -b qemux86-64 agl-flutter agl-devel
    $ nohup time bitbake agl-image-flutter & 

________ deleting 'src/ios_tools' in '/home/bcraddock/AGL/master/qemux86-64/downloads/gn/https__github.com_flutter_engine.git'

Error: Command 'git -c core.quotePath=false diff --name-status v1.1.91' returned non-zero exit status 128 in /home/bcraddock/AGL/master/qemux86-64/downloads/gn/https__github.com_flutter_engine.git/src/third_party/vulkan

ERROR: flutter-engine-release-git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'gn://github.com/flutter/engine.git;name=src/flutter')
ERROR: Logfile of failure stored in: /home/bcraddock/AGL/master/qemux86-64/tmp/work/corei7-64-agl-linux/flutter-engine-release/git-r0/temp/log.do_fetch.800486
NOTE: recipe flutter-engine-release-git-r0: task do_fetch: Failed
ERROR: Task (/home/bcraddock/AGL/master/external/meta-flutter/recipes-graphics/flutter-engine/flutter-engine-release_git.bb:do_fetch) failed with exit code '1'

bernardcraddock, 2022-07-24:
booting the newly built image using  runqemu it's launching gallery

closed 2022-07-24 by jwinarske