meta-flutter/meta-flutter #141: User Service vs System Service?

jwinarske, 2022-07-27:
Looking to consolidate running Homescreen as user session under compositor.  Looking for any feedback from users on what they prefer.  User Service, or System service?

AGL is running as user session under the agl-compositor.  I'm considering making this standard for base images, which will require setting up a user etc.  I was avoiding this until now.

@rzr Are you using a system service to auto-start Flutter?  Is the image setup to run compositor in user session, or system session?

rzr, 2022-07-28:
in oniro currently It's a system service but set as weston user:

https://gitlab.eclipse.org/pcoval/meta-oniro-blueprint-flutter/-/blob/kirkstone/conf/layer.conf#L51

I have no objection to change it but I can share some opinions:

In past life, I remember Tizen:Common project used user service for starting compositor, It worked enough (even if systemd --user was not recommended by systemd project). 

May I suggest to propose to change weston's recipe to create user service and listen to feedback from  upstream (Yocto/OE).

jwinarske, 2022-07-29:
I see.  I have a pending merge which removes the system service definition in favor of using a "platform" specific bbappend.  For AGL it's being dictated that I run as a user session.

https://github.com/meta-flutter/meta-flutter/pull/140/files#diff-a8eb8e01c6cdb4eadba150e359641113c52796378f319c44d424e4412d635650

Corresponding AGL bbappend:
https://github.com/meta-flutter/meta-agl-devel/blob/main/meta-agl-flutter/recipes-graphics/toyota/ivi-homescreen_aglflutter.inc

Once this is settled in the next couple of days, it should remain that way indefinitely.

closed 2022-08-19 by jwinarske