meta-flutter/meta-flutter #144: flutter-engine-runtimerelease do_install fails

bojidartonchev, 2022-08-05:
Hello,

After updating to latest kirkstone, flutter-engine do_install fails with the following error:

```
NOTE: Executing Tasks
ERROR: flutter-engine-runtimerelease-git-r0 do_install: ExpansionError('do_install', "\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/so.unstripped/libflutter_engine.so \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib/libflutter_engine.so\n\n    if ${@bb.utils.contains('PACKAGECONFIG', 'desktop-embeddings', 'true', 'false', d)}; then\n        install -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/so.unstripped/libflutter_linux_gtk.so /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib\n    fi\n\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/flutter_embedder.h \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/include/flutter_embedder.h\n\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/icudtl.dat \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/icudtl.dat\n\n    # create SDK\n    install -D -m0755 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/clang_x64/gen_snapshot \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/clang_x64/gen_snapshot\n\n    cd /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/flutter\n    echo `git rev-parse HEAD` > /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version\n    echo ${FLUTTER_ENGINE_REPO_URL} >> /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version\n    echo ${@get_flutter_sdk_version(d)} >> /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/flutter_sdk.version\n\n    cd /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image//usr/share/flutter\n    zip -r engine_sdk.zip sdk\n    rm -rf sdk\n", URLError(gaierror(-3, 'Temporary failure in name resolution')))
ERROR: Logfile of failure stored in: /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/temp/log.do_install.17355
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-flutter/recipes-devtools/depot-tools/depot-tools-native_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-graphics/fontconfig/fontconfig_2.13.1.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/python/python3_3.10.4.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-extended/zip/zip_3.0.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-support/ca-certificates/ca-certificates_20211016.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-support/curl/curl_7.82.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['clang-cross-aarch64', 'compiler-rt', 'libcxx', 'depot-tools-native', 'glibc', 'quilt-native', 'fontconfig', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 'pbzip2-native', 'zip-native', 'ca-certificates-native', 'curl-native', 'libtool-native', 'linux-libc-headers', 'libgcc', 'gcc-runtime', 'readline-native', 'bzip2-native', 'libffi-native', 'sqlite3-native', 'libtirpc-native', 'gdbm-native', 'zlib-native', 'libnsl2-native', 'xz-native', 'util-linux-libuuid-native', 'ncurses-native', 'openssl-native', 'clang-native', 'binutils-cross-aarch64', 'attr-native', 'expat', 'zlib', 'util-linux', 'freetype', 'texinfo-dummy-native', 'gettext-minimal-native', 'perl-native', 'binutils-native', 'cmake-native', 'libedit-native', 'swig-native', 'libxml2-native', 'flex-native', 'gnu-config-native', 'ncurses', 'util-linux-libuuid', 'libxcrypt', 'libcap-ng', 'opkg-utils', 'bash-completion', 'libpng', 'make-native', 'zstd-native', 'libpcre-native', 'm4-native']
| DEBUG: Python function extend_recipe_sysroot finished
ERROR: Task (virtual:runtimerelease:/mnt/data/Projects/trevor/build/../meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3754 tasks of which 3753 didn't need to be rerun and 1 failed.
```

Reverting back to [1f6516e257713c3db1d241240323bb590580973c](https://github.com/meta-flutter/meta-flutter/commit/1f6516e257713c3db1d241240323bb590580973c) fixed the issue. Am I missing some breaking change, different from `release` becomes `runtimerelease`

Best regards,
Bozhidar

jwinarske, 2022-08-05:
@bojidartonchev My CI jobs needed clean build with AGL refactor PR.  Have you tried a clean build?

jwinarske, 2022-08-05:
If the clean doesn't do it, it's most likely related to the fact I'm missing a push to https://github.com/meta-flutter/meta-flutter-rpi.  Which uses new bbclass extension.  That's unfortunate.  It'll have to wait until the 17th when I'm back in the country.

jwinarske, 2022-08-05:
Also if you're not using metta-flutter-pi, then a clean should fix it, as the kirkstone-linux-dummy jobs are all green:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-linux-dummy.yml#L148

jwinarske, 2022-08-09:
Adding `flutter-pi-runtimerelease` is what you want.  It adds the runtime dependency, so libflutter_engine.so will be installed in image automatically.

Same naming for your app recipe.  e.g. `myapp-runtimerelease`

bojidartonchev, 2022-08-10:
I already had that since I am using succesfully rev. 3c9c95194761a969d359e71337d2dc7820658da5 wich has this change. I will with full clean build after I return to my country next week

gencay3535, 2022-08-18:
Hello, were you able to push the newer version with .bbclassextension?

gencay3535, 2022-08-18:
and another question is I am really new to this area so, I should write a recipe for my flutter project and then built it with my image am I correct?

jwinarske, 2022-08-18:
@gencay3535 You will want to follow this pattern:
https://github.com/meta-flutter/meta-flutter/issues/145#issuecomment-1219460731

Also please open new issue if your questions are unrelated to an open issue.  Something like `trouble building app into image`.

gencay3535, 2022-08-19:
Any updates about this problem?

MEd-Douiri, 2022-08-19:
> Hello,
> 
> After updating to latest kirkstone([0655f2e](https://github.com/meta-flutter/meta-flutter/commit/0655f2eb220d2a929157643f542f8f980986442c)), flutter-engine do_install fails with the following error:
> 
> ```
> NOTE: Executing Tasks
> ERROR: flutter-engine-runtimerelease-git-r0 do_install: ExpansionError('do_install', "\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/so.unstripped/libflutter_engine.so \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib/libflutter_engine.so\n\n    if ${@bb.utils.contains('PACKAGECONFIG', 'desktop-embeddings', 'true', 'false', d)}; then\n        install -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/so.unstripped/libflutter_linux_gtk.so /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib\n    fi\n\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/flutter_embedder.h \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/include/flutter_embedder.h\n\n    install -D -m0644 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/icudtl.dat \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/icudtl.dat\n\n    # create SDK\n    install -D -m0755 /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_arm64/clang_x64/gen_snapshot \\\n        /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/clang_x64/gen_snapshot\n\n    cd /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/src/flutter\n    echo `git rev-parse HEAD` > /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version\n    echo ${FLUTTER_ENGINE_REPO_URL} >> /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version\n    echo ${@get_flutter_sdk_version(d)} >> /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/flutter_sdk.version\n\n    cd /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/image//usr/share/flutter\n    zip -r engine_sdk.zip sdk\n    rm -rf sdk\n", URLError(gaierror(-3, 'Temporary failure in name resolution')))
> ERROR: Logfile of failure stored in: /mnt/data/Projects/trevor/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/flutter-engine-runtimerelease/git-r0/temp/log.do_install.17355
> Log data follows:
> | DEBUG: Executing python function extend_recipe_sysroot
> | NOTE: Direct dependencies are ['/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta-flutter/recipes-devtools/depot-tools/depot-tools-native_git.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/mnt/data/Projects/trevor/build/../meta/recipes-graphics/fontconfig/fontconfig_2.13.1.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-devtools/python/python3_3.10.4.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-extended/zip/zip_3.0.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-support/ca-certificates/ca-certificates_20211016.bb:do_populate_sysroot', 'virtual:native:/mnt/data/Projects/trevor/build/../meta/recipes-support/curl/curl_7.82.0.bb:do_populate_sysroot']
> | NOTE: Installed into sysroot: []
> | NOTE: Skipping as already exists in sysroot: ['clang-cross-aarch64', 'compiler-rt', 'libcxx', 'depot-tools-native', 'glibc', 'quilt-native', 'fontconfig', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'py [... cut ...]
> ```
> 
> Reverting back to [3c9c951](https://github.com/meta-flutter/meta-flutter/commit/3c9c95194761a969d359e71337d2dc7820658da5) fixed the issue. Am I missing some breaking change here?
> 
> Best regards, Bozhidar

Hello, do_install task it use network in this line (echo ${@get_flutter_sdk_version(d)} >> ${D}${datadir}/flutter/sdk/flutter_sdk.version). So  try to add this line in flutter-engine_git.bb file:

   do_install[depends] += "zip-native:do_populate_sysroot"
+ do_install[network] = "1"

jwinarske, 2022-08-19:
@MEd-Douiri If you take a look at the run.do_install script that bitbake creates you will see that is not the case.  The GN call gets made when run.do_install is created, adding `++do_install[network] = "1"` is not a fix.

```
# line: 148, file: /__w/meta-flutter/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb
do_install() {

    install -D -m0644 /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_x64/so.unstripped/libflutter_engine.so \
        /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib/libflutter_engine.so

    if false; then
        install -m0644 /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_x64/so.unstripped/libflutter_linux_gtk.so /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/lib
    fi

    install -D -m0644 /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_x64/flutter_embedder.h \
        /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/include/flutter_embedder.h

    install -D -m0644 /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_x64/icudtl.dat \
        /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/icudtl.dat

    # create SDK
    install -D -m0755 /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/out/linux_release_x64/clang_x64/gen_snapshot \
        /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/clang_x64/gen_snapshot

    cd /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/src/flutter
    echo `git rev-parse HEAD` > /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version
    echo ${FLUTTER_ENGINE_REPO_URL} >> /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/engine.version
    echo 3.0.5 >> /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image/usr/share/flutter/sdk/flutter_sdk.version

    cd /__w/meta-flutter/yocto-kirkstone/build/tmp/work/core2-64-poky-linux/flutter-engine-runtimerelease/git-r0/image//usr/share/flutter
    zip -r engine_sdk.zip sdk
    rm -rf sdk
}
```

jwinarske, 2022-08-19:
@bojidartonchev Since I'm not seeing this in daily CI runs I'm closing this.  Feel free to re-open if you have a sequence I can replicate to repo the issue.

bojidartonchev, 2022-08-23:
Unfortunately even after `bitbake -c do_cleansstate flutter-engine-runtimerelease` do_install fails with the same error. However after adding the `do_install[network] = "1"` to flutter-engine_git.bb it worked. I think it is needed for get_flutter_sdk_version.

jwinarske, 2022-09-07:
@bojidartonchev I don't get a repro of this in my build matrix.  I need more information.

Did you try commenting out these two lines (not adding `do_install[network] = "1"`):
```
    echo `git rev-parse HEAD` > ${D}${datadir}/flutter/sdk/engine.version
    echo ${FLUTTER_ENGINE_REPO_URL} >> ${D}${datadir}/flutter/sdk/engine.version
```

jwinarske, 2022-09-07:
Until there is a root cause:  https://github.com/meta-flutter/meta-flutter/pull/177

closed 2022-09-07 by jwinarske