meta-flutter/meta-flutter #145: libflutter_engine.so not found

mortenboye, 2022-08-08:
I am trying to run a yocto image with meta-flutter.

My recipe and image was working on honister but after updating to kirkstone I'm seeing the below error when trying to run flutter-pi.

The directory /usr/share/midium does not contain the libflutter_engine.so file, but only a data folder with flutter_assets within.
I do have /usr/lib/libflutter_engine.so as part of my image though.

```
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /usr/share/midium/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.debug...
[flutter-pi] Warning: Could not load libflutter_engine.so.debug. dlopen: libflutter_engine.so.debug: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so...
embedder.cc (1177): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[flutter-pi] Could not initialize the flutter engine. FlutterEngineInitialize: Invalid arguments.
```

I'm probably doing something wrong, but not sure what. Any clues?

jwinarske, 2022-08-09:
@bojidartonchev can you take a look at this since your patch was the last change?

jwinarske, 2022-08-09:
@mortenboye are you pointing to the flutter_asset folder, and you have matching Flutter SDK and flutter engine?

mortenboye, 2022-08-09:
> @mortenboye are you pointing to the flutter_asset folder, and you have matching Flutter SDK and flutter engine?

Yes, running flutter-pi like this
`flutter-pi /usr/share/midium`

And have not touched SDK and engine settings since honister (which worked).

I was a bit confused about the new runtime type defines, but did manage to get poky to build eventually.

my-app.inc
```
SUMMARY = "My Application"
DESCRIPTION = "My Application"

LICENSE = "CLOSED"

VERSION = "0.13"

SRC_URI =   "git://bitbucket.org/xxx/my_app.git;protocol=http;branch=master;user=xxxxxxx"
SRCREV = "${AUTOREV}"
PV = "${VERSION}+git${SRCPV}"

S = "${WORKDIR}/git"

PUBSPEC_APPNAME = "myapp"

inherit flutter-app
```

my_app_debug_git.bb
```
include my-app.inc

FLUTTER_RUNTIME = "debug"

DEPENDS += "\
    flutter-engine-runtimedebug \
    "
```

bojidartonchev, 2022-08-09:
@jwinarske I can look it, but I am on vacation for the next two weeks, so it would be after that.

jwinarske, 2022-08-18:
@mortenboye Just getting back from vacation myself.  

The new pattern should look like this:

my-app_git.bb

```
SUMMARY = "My app summary"
DESCRIPTION = "My boo description"
AUTHOR = "Me"
HOMEPAGE = "..."
BUGTRACKER = ".../issues"
SECTION = "graphics"

LICENSE = "CLOSED"

SRCREV = "${AUTOREV}"
SRC_URI = "..."

S = "${WORKDIR}/git"

PUBSPEC_APPNAME = "<actual name from pubspec>"
FLUTTER_APPLICATION_INSTALL_PREFIX = "/flutter"

inherit flutter-app
```

local.conf

New override specifier
```
IMAGE_INSTALL:append = "my-app-runtimerelease"
```

Old override specifier
```
IMAGE_INSTALL_append = "my-app-runtimerelease"
```

Runtime deps are implicit installs.  So no need to select the engine to be installed.

jwinarske, 2022-08-19:
@mortenboye 
The other scenario that can be confusing.  If you build image that include flutter-pi-runtimerelease, then you have to supply valid AOT image in the proper place.

I patch flutter-pi so that it matches the bundle folder layout.  This originates from the Linux Desktop build support.  The format looks like this:

```
<bundle folder>
    data
        flutter_assets
    lib
        libapp.so <only if libflutter_engine.so is runtime=release>
```

if debug:
`flutter-pi <bundle folder>`

if release:
`flutter-pi <bundle folder> --release`

By default flutter engine is installed as a system library, and this file should be present:
`/usr/lib/libflutter_engine.so`

Are you following the above patterns?

mortenboye, 2022-08-22:
> Are you following the above patterns?

I believe I am. 
I can run my app in --release mode (although with missing fonts), but in debug I still get the
`'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[flutter-pi] Could not initialize the flutter engine. FlutterEngineInitialize: Invalid arguments.`
error.

I will double check everything and return if I need more assistance. Thank you

jwinarske, 2022-08-22:
Someone else reported flutter-pi issue, which looks like the bundle path is out of whack.  Looking upstream it looks like this area was being played with.

Try copying contents of <app>/data/flutter_assets/* to <app> and let me know what you get.

bojidartonchev, 2022-08-23:
So I checked your setup, and indeed using `IMAGE_INSTALL:append = "my-app-runtimerelease"` the fonts are missing. However copying the content of `<app>/data/flutter_assets/*` to `<app>` fixes that.

`cp -r /usr/share/flutter/<app>/data/flutter_assets/* /usr/share/flutter/<app>`

@mortenboye For debug, could you try adding the right engine binaries, using runtimedebug?
`IMAGE_INSTALL:append = "my-app-runtimedebug"`

jwinarske, 2022-08-23:
@bojidartonchev There is no need to explicitly add the engine flavor to the image.  It's explicitly included as it's a runtime dependency.  If you try to mix variants in the image, the image build will fail.

@mortenboye Can you clarify the problem better?

mortenboye, 2022-08-23:
@jwinarske and @bojidartonchev thank you both for the guidance.

It seems that copying `flutter_assets/*` to `/<app>` did the trick for me.

My image recipes are now

```
IMAGE_INSTALL += ' \
  flutter-pi-runtimedebug \
  my-app-runtimedebug \
'
```

and 

```
IMAGE_INSTALL += ' \
  flutter-pi-runtimerelease \
  my-app-runtimerelease \
'
```

respectively.

Can I fix the bundle path issue myself or should I wait for a fix?

jwinarske, 2022-08-23:
@mortenboye I should have a fix into kirkstone and honister by tomorrow AM PST.

jwinarske, 2022-08-24:
Closing in favor of https://github.com/meta-flutter/meta-flutter/issues/152

closed 2022-08-24 by jwinarske

jwinarske, 2022-08-24:
I won't have time to smoke test till later.  If you do that would be great.
https://github.com/meta-flutter/meta-flutter/pull/160

jwinarske, 2022-08-24:
Smoke test passed, and fix merged to kirkstone