meta-flutter/meta-flutter #146: Error building head of honister branch

ajlennon, 2022-08-15:
Has anybody got any idaes?

```
Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mmevk"
DISTRO               = "fsl-imx-xwayland"
DISTRO_VERSION       = "5.10-hardknott"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:269265c00091fa65f93de6cad32bf24f1e7f72a3"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:f44e1a2b575826e88b8cb2725e54a7c5d29cf94a"
meta-freescale       = "HEAD:469d6c958c76ea235b3d3c1527e797ce3a7392e3"
meta-freescale-3rdparty = "HEAD:f8150f3b37cb83cba1f9e2378e57bb63e02d4610"
meta-freescale-distro = "HEAD:e6daa26ba1f748326546063d63a085ae671827d9"
meta-bsp             
meta-sdk             
meta-ml              
meta-v2x             = "HEAD:a38040cc159f0ef9fc177358932c2c75257c59e9"
meta-nxp-demo-experience = "HEAD:9dcc11ea9f525cffedbb28895e0abb443e56c3e0"
meta-chromium        = "HEAD:8be1d3a0ba0cf32e61144900597207af5698c10d"
meta-clang           = "HEAD:b0d805060791006d651efd3d7ae3dd5add8f70fe"
meta-gnome           
meta-networking      
meta-filesystems     = "HEAD:f44e1a2b575826e88b8cb2725e54a7c5d29cf94a"
meta-qt5             = "HEAD:43f8f539d40070a70fe89136db89bf5bb1dfe7ed"
meta-python2         = "HEAD:8db9e4f6ceae33d7a4f55453d31e69f9858af4eb"
meta-virtualization  = "HEAD:7f719ef40896b6c78893add8485fda995b00d51d"
meta-flutter         = "honister:050c679e6a3d64c8b43efb2fefbea8dc22139d00"
meta-phasora         = "main:63c57d08e76f3e005b741c8d9c800823b4205901"
```
```
ERROR: flutter-engine-release-git-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/bin/python3-native:/opt/yocto/sources/poky/scripts:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot/usr/bin/crossscripts:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/sbin:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/bin:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/sbin:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/bin:/opt/yocto/sources/poky/bitbake/bin:/opt/yocto/build/tmp/hosttools"; export HOME="/home/ajlennon"; export PATH="/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/share/depot_tools:/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/usr/share/depot_tools/bootstrap-2@3.8.9.chromium.14_bin/python/bin:${PATH}"; export DEPOT_TOOLS_UPDATE=0; export GCLIENT_PY3=0;             export CURL_CA_BUNDLE=/opt/yocto/build/tmp/work/cortexa53-crypto-poky-linux/flutter-engine-release/git-r0/recipe-sysroot-native/etc/ssl/certs/ca-certificates.crt;             gclient.py config --spec 'solutions = [
    {
        "managed": False,
        "name": "src/flutter",
        "url": "https://github.com/flutter/engine.git",
        "custom_vars": {    "download_android_deps": False,     "download_windows_deps": False,     "download_linux_deps": False,   }
    }
]' &&             gclient.py sync --shallow --no-history -R -D --revision 57d3bac3dd5cb5b0e464ab70e7bc8a0d8cf083ab -j 12 -v failed with exit code 1, output:
Looking for .gclient starting from /opt/yocto/downloads/gn/https__github.com_flutter_engine.git

Loaded .gclient config in /opt/yocto/downloads/gn/https__github.com_flutter_engine.git:
solutions = [
    {
        "managed": False,
        "name": "src/flutter",
        "url": "https://github.com/flutter/engine.git",
        "custom_vars": {    "download_android_deps": False,     "download_windows_deps": False,     "download_linux_deps": False,   }
    }
]

src/flutter (Elapsed: 0:00:04)
----------------------------------------
[0:00:00] Started.
_____ src/flutter at 57d3bac3dd5cb5b0e464ab70e7bc8a0d8cf083ab
[0:00:00] Finished running: git config remote.origin.url
[0:00:00] Finished running: git rev-list -n 1 HEAD
[0:00:00] Finished running: git rev-parse --abbrev-ref=strict HEAD

________ running 'git reset --hard HEAD' in '/opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src/flutter'

[0:00:02] HEAD is now at 57d3bac3dd Adds a gn script flag to select mallinfo2 when building the Dart VM (#32187) (#32221)
[0:00:02] Finished running: git rev-list -n 1 HEAD
[0:00:02] Up-to-date; skipping checkout.
[0:00:02] Finished running: git rev-parse HEAD
Checked out revision 57d3bac3dd5cb5b0e464ab70e7bc8a0d8cf083ab
[0:00:02] Finished running: git rev-parse --verify HEAD
[0:00:04] Finished.
----------------------------------------

src (Elapsed: 0:00:00)
----------------------------------------
[0:00:04] Started.
_____ src at 4660f4a166979c64ff56f6b856b44ccc1dba5c63
[0:00:04] Finished running: git config remote.origin.url
[0:00:04] Finished running: git rev-list -n 1 HEAD
[0:00:04] Finished running: git rev-parse --abbrev-ref=strict HEAD

________ running 'git reset --hard HEAD' in '/opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src'

[0:00:04] HEAD is now at 4660f4a Add path to public headers for GLFW. (#539)
[0:00:04] Finished running: git rev-list -n 1 HEAD
[0:00:04] Up-to-date; skipping checkout.
[0:00:04] Finished running: git rev-parse HEAD
Checked out revision 4660f4a166979c64ff56f6b856b44ccc1dba5c63
[0:00:04] _____ removing unversioned directory _gclient_gittmp_ios_toolsELDnAV/
[0:00:04] _____ removing unversioned directory _gclient_ios_tools_ks36JG/
[0:00:04] Finished running: git rev-parse --verify HEAD
[0:00:04] Finished.
----------------------------------------

src/buildtools/linux-x64/clang:fuchsia/third_party/clang/linux-amd64 (Elapsed: 0:00:00)
----------------------------------------
[0:00:04] Started.
[0:00:04] Finished.
----------------------------------------

src/flutter/third_party/gn:gn/gn/${platform} (Elapsed: 0:00:00)
----------------------------------------
[0:00:04] Started.
[0:00:04] Finished.
----------------------------------------
```

Gmatarrubia, 2022-08-17:
I get the same error

ajlennon, 2022-08-17:
Thanks for confirming :pray:

jwinarske, 2022-08-17:
@ajlennon what branch are you using?

jwinarske, 2022-08-17:
@ajlennon you want to use dunfell branch.  Notice the layer config file supports honester.  Dunfell represents the older override specifier, kirkstone represents the new.

ajlennon, 2022-08-17:
ok thanks - will try that here

Gmatarrubia, 2022-08-17:
Hi @ajlennon, I got a successful build again. I created a PR in which I pick latest changes from kirkstone branch to the honister one. In the meantime, you can use my changes from my fork. Let me know, if it could work for you. [Gmatarrubia/meta-flutter honister branch](https://github.com/Gmatarrubia/meta-flutter/tree/honister)

ajlennon, 2022-08-17:
Same problem with `dunfell`

```
Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mmevk"
DISTRO               = "fsl-imx-xwayland"
DISTRO_VERSION       = "5.10-hardknott"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:269265c00091fa65f93de6cad32bf24f1e7f72a3"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:f44e1a2b575826e88b8cb2725e54a7c5d29cf94a"
meta-freescale       = "HEAD:469d6c958c76ea235b3d3c1527e797ce3a7392e3"
meta-freescale-3rdparty = "HEAD:f8150f3b37cb83cba1f9e2378e57bb63e02d4610"
meta-freescale-distro = "HEAD:e6daa26ba1f748326546063d63a085ae671827d9"
meta-bsp             
meta-sdk             
meta-ml              
meta-v2x             = "HEAD:a38040cc159f0ef9fc177358932c2c75257c59e9"
meta-nxp-demo-experience = "HEAD:9dcc11ea9f525cffedbb28895e0abb443e56c3e0"
meta-chromium        = "HEAD:8be1d3a0ba0cf32e61144900597207af5698c10d"
meta-clang           = "HEAD:b0d805060791006d651efd3d7ae3dd5add8f70fe"
meta-gnome           
meta-networking      
meta-filesystems     = "HEAD:f44e1a2b575826e88b8cb2725e54a7c5d29cf94a"
meta-qt5             = "HEAD:43f8f539d40070a70fe89136db89bf5bb1dfe7ed"
meta-python2         = "HEAD:8db9e4f6ceae33d7a4f55453d31e69f9858af4eb"
meta-virtualization  = "HEAD:7f719ef40896b6c78893add8485fda995b00d51d"
meta-flutter         = "dunfell:dbebc62f13c09f02eb4adc644b77d086b44695e2"
meta-phasora         = "main:63c57d08e76f3e005b741c8d9c800823b4205901"
```

ajlennon, 2022-08-17:
@Gmatarrubia there doesn't seem to be a `flutter-engine-release` any more?

jwinarske, 2022-08-17:
@ajlennon personally I would suggest sticking with LTS releases (kirkstone + Dunfell).  These are the two i have been maintaining regularly.  AGL is on kirkstone.  So a branch like honister is a much lower priority.  If the PR from @Gmatarrubia passes all checks I'll merge it to honister.  If your target BSP doesn't have a kirkstone branch yet, chances are it will soon.  What SoC are you targeting?

ajlennon, 2022-08-17:
I tried `dunfell` as you suggested @jwinarske - it fails to build in the same way

jwinarske, 2022-08-17:
@ajlennon Dunfell has a CI job for imx8m-mini
https://github.com/meta-flutter/meta-flutter/blob/dunfell/.github/workflows/dunfell-dart-mx8m-mini.yml

jwinarske, 2022-08-17:
I'll be updating Dunfell to match recent kirkstone releases this week.  I was out on vacation the last two weeks

ajlennon, 2022-08-17:
I'm building for the NXP reference target here rather than the variscite board. I don't think I'm doing anything particularly custom though. I just added in the lines from the README.

ajlennon, 2022-08-17:
This is my `local.conf`

```
MACHINE ??= 'imx8mmevk'
DISTRO ?= 'fsl-imx-xwayland'
PACKAGE_CLASSES ?= 'package_rpm'
EXTRA_IMAGE_FEATURES ?= "debug-tweaks tools-debug"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
CONF_VERSION = "1"

DL_DIR ?= "${BSPDIR}/downloads/"
ACCEPT_FSL_EULA = "1"

# Switch to Debian packaging and include package-management in the image
PACKAGE_CLASSES = "package_deb"
EXTRA_IMAGE_FEATURES += "package-management"

INHERIT += " rm_work "

# Support flutter
FLUTTER_SDK_TAG = "2.10.4"
IMAGE_INSTALL:append = " flutter-engine"
#IMAGE_INSTALL:append = " ivi-homescreen"
#IMAGE_INSTALL:append = " flutter-gallery"
LICENSE_FLAGS_WHITELIST = " commercial "

IMAGE_INSTALL:append = " kernel-modules"

OE_TERMINAL = "screen"
```

jwinarske, 2022-08-17:
I can add a CI job.  Where's the BSP and what's the machine name?

jwinarske, 2022-08-17:
Found it:
https://www.nxp.com/design/software/embedded-software/i-mx-software/embedded-linux-for-i-mx-applications-processors:IMXLINUX?

ajlennon, 2022-08-17:
Yeah I am building out of sources from manifest `imx-5.10.72-2.2.0.xml`

jwinarske, 2022-08-17:
I'll add a CI job to kirkstone branch.  LTS is the way to go as security updates are more frequent.

ajlennon, 2022-08-17:
Yes - I've been working on kernel patches to support a new MIPI display so don't really want to rejig everything. Just want to get Flutter on there so a client can take a look and see if they fancy using it for the UX work.

ajlennon, 2022-08-17:
Yes I want to be on Kirskstone LTS for the forseable future too :smile:

ajlennon, 2022-08-17:
Your fork failed to build here for me too @Gmatarrubia. Maybe I'll try syncing to the kirkstone repo then....

ajlennon, 2022-08-18:
Building `kirkstone` now and getting the same errors. I wonder what is going on :(

ajlennon, 2022-08-18:
Guess I need to work out what the difference is between here and the CI 

```
[0:28:12] fatal: Couldn't find remote ref HEAD
[0:28:13] 
________ running 'git -c core.deltaBaseCacheLimit=2g fetch origin --verbose --no-tags' in '/opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src/third_party/dart/third_party/pkg/linter' attempt 3 / 4

[0:28:13] fatal: Couldn't find remote ref HEAD
[0:28:16] 
________ running 'git -c core.deltaBaseCacheLimit=2g fetch origin --verbose --no-tags' in '/opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src/third_party/dart/third_party/pkg/linter' attempt 4 / 4

[0:28:16] fatal: Couldn't find remote ref HEAD
----------------------------------------
Error: Command 'git -c core.deltaBaseCacheLimit=2g fetch origin --verbose --no-tags' returned non-zero exit status 128 in /opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src/third_party/dart/third_party/pkg/linter

Warnings:
Conflicting directory /opt/yocto/downloads/gn/https__github.com_flutter_engine.git/src/third_party/dart/third_party/pkg/linter moved to /opt/yocto/downloads/gn/https__github.com_flutter_engine.git/_bad_scm/src/third_party/dart/third_party/pkg/linteroVdy8t.

ERROR: flutter-engine-git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'gn://github.com/flutter/engine.git;name=src/flutter')
ERROR: Logfile of failure stored in: /opt/yocto/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/temp/log.do_fetch.1057476
ERROR: Task (/opt/yocto/sources/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 11067 tasks of which 10915 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  virtual:runtimerelease:/opt/yocto/sources/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_fetch
  /opt/yocto/sources/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_fetch
```

jwinarske, 2022-08-18:
@ajlennon when switching branches be sure to run a clean build - clean sstate for all related flutter recipes, and remove any cached download files.  The difference could also be in the build/host environment.  Using a container makes it easier to isolate issues like this.  Your best bet is to attempt to replicate the CI pattern.

ajlennon, 2022-08-18:
I removed the build folder and did a clean build. I am using a container forked from imx-docker

https://github.com/DynamicDevices/imx-docker

I might _possibly_ have cached download files somewhere...

ajlennon, 2022-08-18:
One other odd thing is the fetching seems to take an absolute age

```
0: flutter-engine-git-r0 do_fetch - 47m11s (pid 1159309)
```

ajlennon, 2022-08-18:
This seems to be the error

[image]

jwinarske, 2022-08-18:
The fetch will use cached download folder file if present.  If you use a stale download directory (switching branches) problems might arise.

Looks like maybe one of the deps repos is not present.

Have you tried running repo unit of the same version manually?  If one of the deps disappeared online, it would repo via manual engine build.

The other is to try not to fix the flutter SDK to this version, and try with tip of tree stable (default for kirkstone).

jwinarske, 2022-08-18:
Meaning remove `FLUTTER_SDK_TAG = "2.10.4"` from your local.conf, and run a clean build.

ajlennon, 2022-08-18:
I've removed the downloads folder and am trying the build again now

Next I'll remove the `FLUTTER_SDK_TAG = "2.10.4"`

I'll also have a look at the CI workflow and see if I can spot some difference

ajlennon, 2022-08-18:
Removing the downloads folder didn't help.

BUT with the `FLUTTER_SDK_TAG = "2.10.4"` removed it's now compiling !!!

ajlennon, 2022-08-18:
>Looks like maybe one of the deps repos is not present.

I agree - I suspect they've broken something upstream

jwinarske, 2022-08-18:
@ajlennon Cool, glad you're unblocked.  It can be frustrating!

closed 2022-08-18 by jwinarske

ajlennon, 2022-08-18:
That's the nature of all this stuff eh! I really appreciate you jumping in and responding thanks. Will confirm shortly that my image builds and runs up OK...

jwinarske, 2022-08-18:
@ajlennon One other thing for Kirkstone your local.conf would want this:

IMAGE_INSTALL:append = "\
    ivi-homescreen-runtimerelease \
    flutter-gallery-runtimerelease \
    flutter-pi-runtimerelease \
"

Also I would use fsl-imx-wayland distro, and remove x11 from DISTRO_FEATURES unless you have a hard requirement on X11.

I started a CI pipeline here (it's not done yet - later):
https://github.com/meta-flutter/meta-flutter/blob/jw/imx_ci/.github/workflows/kirkstone-imx8mmevk.yml

ajlennon, 2022-08-18:
Thanks will try this - I suspect I need `flutter-engine-runtimerelease` though?

jwinarske, 2022-08-18:
installation of runtime dependencies are implicit.  So no it's not needed.

ajlennon, 2022-08-18:
>installation of runtime dependencies are implicit. So no it's not needed.

OK - but I shouldn't need `flutter-pi-runtimerelease` should I?

jwinarske, 2022-08-18:
Only if you want DRM support.  I just had that as an example of the bbclass extension logic.

ajlennon, 2022-08-18:
>Only if you want DRM support. I just had that as an example of the bbclass extension logic.

What I mean is this seems to be for a Raspberry Pi and I am on an i.MX8M Mini so I won't want the Pi stuff will I?

jwinarske, 2022-08-18:
It runs on most targets that require DRM support.  The name is misleading.  At some point I will be adding DRM support into homescreen/flutter-auto.  Not all use cases need a compositor.

Just ignore my suggestion to add it :)

ajlennon, 2022-08-18:
AHA! OK :)

jwinarske, 2022-08-18:
Kirkstone CI is passing:  https://github.com/meta-flutter/meta-flutter/actions/runs/2884740773

Interestingly DISTRO_FEATURES includes vulkan by default, which means ivi-homescreen will default to the Vulkan backend.  If you want EGL backend, you would either remove vulkan from DISTRO_FEATURES, or override the package config of ivi-homescreen.

ajlennon, 2022-08-19:
>Interestingly DISTRO_FEATURES includes vulkan by default,

I'm not really sure about the different backends to be honest. Should I be removing vulkan in favour of EGL do you recommend?

ajlennon, 2022-08-19:
I've got it all built which is great. I see `flutter-pi` on there which is good. 

I've tried to build the debug bundle for the gallery from the README https://github.com/ardera/flutter-pi

```
git clone https://github.com/flutter/gallery.git flutter_gallery
cd flutter_gallery
git checkout 1dfcd617a260e4468f0883d7dfbfdfe3d021f7b3
flutter build bundle
rsync -a ./build/flutter_assets/ pi@raspberrypi:/home/pi/flutter_gallery/
```

But when I try to run I get

`[flutter-pi] Could not find "kernel.blob" file inside "/home/root/flutter_assets/data/flutter_assets/", which is required for debug mode.`

I wondered if you have any idea how I can run something simple up for demo?

jwinarske, 2022-08-19:
For flutter-pi I apply a path so it matches the rest of the embedders in meta-flutter:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-pi/files/0001-path-updates.patch
This follows the "bundle" pattern also seen with the Linux GTK embedder:

<bundle folder>
    data/flutter_assets
    lib/libapp.so <if runtimerelease used>

pass the bundle folder to flutter-pi

jwinarske, 2022-08-19:
Also heads up I have a script in the works that setups a Flutter Workspace and will do this setup auto-magically:
https://github.com/meta-flutter/meta-flutter/tree/jw/custom-devices/tools

ajlennon, 2022-08-19:
ok thanks it looks like that patch is what the Flutter "gallery" is having trouble with then?

I'm building this here and have new problems now! Maybe this is as I am now following upstream master

[image]

jwinarske, 2022-08-19:
The patch is automatically applied as part of the flutter-pi recipe.

You just need to format your folder on target to match the bundle format I laid out above.  

If you include `flutter-gallery-runtimerelease` in your image, you would just run `flutter-pi /usr/share/flutter/gallery`.  It may want --release, I haven't played with it in a while.

I would only demo AOT images (runtimerelease), as it's natively compiled opposed to running via interpreter (runtimedebug).

ajlennon, 2022-08-19:
OK so I think we have two different things happening

- when I build the gallery bundle on my host it seems to be in a different format from the data/lib format you are patching the flutter-pi to use. I've updated flutter on my host to latest so I'm obviously not understanding something about why there are different bundle formats

- coincidentally it looks as though upstream flutter may have just updated so this patch doesn't apply (as I have to remove the tag to get it to compile so not am not versioned and just following master. I probably need to work out how to use the version that worked instead of master)

ajlennon, 2022-08-19:
I think it's this commit they've just upt into flutter-pi

https://github.com/ardera/flutter-pi/commit/cf57b52a5848439dd4a53b657cbcb06706340ee9#diff-eca2e37422698db262830c966c5532d6b07452df497e85ccda83fd2315f78089

ajlennon, 2022-08-19:
OK this is what I get with `flutter-pi /usr/share/flutter/gallery`

```
root@imx8mmevk:~# flutter-pi --release /usr/share/flutter/gallery
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[keyboard] Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
==============Locale==============
Flutter locale:
  default: C
  locales: C
===================================
===================================
display mode:
  resolution: 480 x 640
  refresh rate: 44Hz
  physical size: 68mm x 121mm
  flutter device pixel ratio: 1.857585
===================================
EGL information:
  version: 1.5
  vendor: "Vivante Corporation"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_platform_wayland EGL_EXT_platform_wayland EGL_EXT_device_query EGL_EXT_device_drm EGL_EXT_device_drm_render_node
 EGL_KHR_platform_gbm"
  display extensions: "EGL_KHR_fence_sync EGL_KHR_reusable_sync EGL_KHR_wait_sync EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap
_image EGL_KHR_gl_renderbuffer_image EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_lock_surface EGL_KHR_create_context EGL_KHR_no_config_context EGL_KHR_surface
less_context EGL_KHR_get_all_proc_addresses EGL_EXT_buffer_age EGL_ANDROID_native_fence_sync EGL_WL_bind_wayland_display EGL_WL_create_wayland_buffer_from_image EGL_KHR_partial_update EGL_EX
T_swap_buffers_with_damage EGL_KHR_swap_buffers_with_damage EGL_EXT_pixel_format_float"
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 V6.4.3.p4.398061"
  shading language version: "OpenGL ES GLSL ES 1.0.0"
  vendor: "Vivante Corporation"
  renderer: "Vivante GC7000NanoUltra"
  extensions: "GL_OES_vertex_half_float GL_OES_element_index_uint GL_OES_mapbuffer GL_OES_vertex_array_object GL_OES_compressed_ETC1_RGB8_texture GL_OES_compressed_paletted_texture GL_OES_te
xture_npot GL_OES_rgb8_rgba8 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_depth32 GL_OES_packed_depth_stencil GL_OES_fbo_render_mipmap GL_OES_get_program_binary G
L_OES_fragment_precision_high GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_required_internalformat GL_OES_sur
faceless_context GL_OES_texture_border_clamp GL_EXT_texture_type_2_10_10_10_REV GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_EXT_texture_compression_s3tc GL_EXT_read_for
mat_bgra GL_EXT_multi_draw_arrays GL_EXT_frag_depth GL_EXT_discard_framebuffer GL_EXT_blend_minmax GL_EXT_multisampled_render_to_texture GL_EXT_robustness GL_EXT_texture_sRGB_decode GL_EXT_t
exture_border_clamp GL_EXT_texture_rg GL_EXT_unpack_subimage GL_VIV_direct_texture "
===================================
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /usr/share/flutter/gallery/libflutter_engine.so: cannot open shared object file: No such file or dire
ctory. Trying to open libflutter_engine.so.release...
[flutter-pi] Warning: Could not load libflutter_engine.so.release. dlopen: libflutter_engine.so.release: cannot open shared object file: No such file or directory. Trying to open libflutter_
engine.so...
flutter: Error loading AssetManifest.json, e: Unable to load asset: AssetManifest.json
flutter: Error: google_fonts was unable to load font Montserrat-Bold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Bold was not found in the application assets. Ensure Montserrat-Bold.otf exists in a folder that is included i
n your pubspec's assets.
flutter: Error: google_fonts was unable to load font Oswald-SemiBold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Oswald-SemiBold was not found in the application assets. Ensure Oswald-SemiBold.otf exists in a folder that is included i
n your pubspec's assets.
flutter: Error: google_fonts was unable to load font Oswald-Medium because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Oswald-Medium was not found in the application assets. Ensure Oswald-Medium.otf exists in a folder that is included in yo
ur pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-Medium because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Medium was not found in the application assets. Ensure Montserrat-Medium.otf exists in a folder that is includ
ed in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-Regular because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Regular was not found in the application assets. Ensure Montserrat-Regular.otf exists in a folder that is incl
uded in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-SemiBold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-SemiBold was not found in the application assets. Ensure Montserrat-SemiBold.otf exists in a folder that is in
cluded in your pubspec's assets.
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: Unable to load asset: packages/flutter_localized_locales/data/en.json
#0      PlatformAssetBundle.load (package:flutter/src/services/asset_bundle.dart:237)
<asynchronous suspension>
#1      AssetBundle.loadString (package:flutter/src/services/asset_bundle.dart:72)
<asynchronous suspension>
```

ajlennon, 2022-08-19:
I had to add in three symlinks so it finds the files

```
root@imx8mmevk:/usr/share/flutter/gallery# ls -l
total 8
lrwxrwxrwx 1 root root   38 Aug 19 17:14 AssetManifest.json -> data/flutter_assets/AssetManifest.json
drwxr-xr-x 3 root root 4096 Mar  9  2018 data
drwxr-xr-x 2 root root 4096 Mar  9  2018 lib
lrwxrwxrwx 1 root root   29 Aug 19 17:06 libflutter_engine.so -> /usr/lib/libflutter_engine.so
lrwxrwxrwx 1 root root   28 Aug 19 17:15 packages -> data/flutter_assets/packages
root@imx8mmevk:/usr/share/flutter/gallery# 
```

[image]

jwinarske, 2022-08-19:
I see.  It looks like flutter-pi currently expects the contents of flutter_assets in the root of the bundle folder.  I will fix that to match the bundle pattern.  Thanks for the input, very helpful.

ajlennon, 2022-08-19:
Glad to be of help :pray:

jwinarske, 2022-08-22:
`honister` is now at parity with `kirkstone`