meta-flutter/meta-flutter #149: flutter-engine-runtimerelease-git-r0 do_unpack: Bitbake Fetcher Error

gencay3535, 2022-08-18:
So I tried to install flutter gallery app, 
I added this line to my local.conf

`IMAGE_INSTALL:append= "flutter-gallery-runtimerelease"`

and when I try to build it throws this error:

`ERROR: flutter-engine-runtimerelease-git-r0 do_unpack: Bitbake Fetcher Error: UnpackError('Unpack command PATH="/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/sysroots-uninative/x86_64-linux/usr/bin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/usr/bin/python3-native:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/layers/openembedded-core/scripts:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot/usr/bin/crossscripts:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/usr/sbin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/usr/bin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/sbin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/recipe-sysroot-native/bin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/layers/openembedded-core/bitbake/bin:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/hosttools" pbzip2 -dc -p12 /home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/downloads/gn/https__github.com_flutter_engine.git-e85ea0e79c6d894c120cda4ee8ee10fe6745e187.tar.bz2 | tar x --no-same-owner -f - failed with return value 2', 'gn://github.com/flutter/engine.git;name=src/flutter')
ERROR: Logfile of failure stored in: /home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/flutter-engine-runtimerelease/git-r0/temp/log.do_unpack.2505159
ERROR: Task (virtual:runtimerelease:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7714 tasks of which 7692 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: 1 task failed:
  virtual:runtimerelease:/home/gencay/STM32MPU_workspace/openstlinux-5.15-yocto-kirkstone-mp1-v22.06.15/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_unpack
`
I am on kirkstone branch so do you have any idea?

jwinarske, 2022-08-18:
Do you have the tool installed on host that packs/unpacks the archive?

jwinarske, 2022-08-18:
What does this file look like running `ls -la`?  Is it a valid file, and can you open it from nautilus?
${DL_DIR}/gn/https__github.com_flutter_engine.git-e85ea0e79c6d894c120cda4ee8ee10fe6745e187.tar.bz2

gencay3535, 2022-08-18:
Yes I can open it /./src/flutter/ it has this kind of files.

gencay3535, 2022-08-19:
[image]
Here is the error when I try to extract myself.

jwinarske, 2022-08-19:
I have a PR in the works which adds kirkstone stm32mp15 CI job.  It's not my top priority but it may give you some clues until it's done.

https://github.com/meta-flutter/meta-flutter/pull/150

jwinarske, 2022-08-21:
Also if you cancel a build mid-build chances are you may need to clean the build environment.  This is a standard Yocto thing.

gencay3535, 2022-08-21:
> Also if you cancel a build mid-build chances are you may need to clean the build environment. This is a standard Yocto thing.

Yep that worked for this error but I still get the do_install error, I am trying with this CI job now #150 . Thank you  so much.

closed 2022-08-21 by gencay3535

gencay3535, 2022-08-22:
may I ask why are you using clang12 instead of master branch?

jwinarske, 2022-08-22:
I generally track clang version to what the flutter engine is currently at.  10 is too old, and I avoid master as it's not considered stable; and subject to change.  Clang kirkstone is 14 I believe, and kirkstone-12 is 12.  I haven't largely tested all the recipes against 14; little to no test mileage.  Why do you ask?

jwinarske, 2022-08-22:
Actually I just went through the combos.  I am using clang `kirkstone` in a number of scenarios already.
1. AGL builds use clang kirkstone
2. rpi-zero2w-64 CI job uses clang kirkstone
3. Dragon Board uses master (changing to kirkstone)
4. imx CI is using clang kirkstone.

Technically Dragon Board CI is providing some limited test mileage for clang master.

jwinarske, 2022-08-22:
I just made a commit to kirkstone for the linux-dummy CI job from `kirkstone-12` to `kirkstone`.  So all kirkstone CI is using clang `kirkstone`

now.https://github.com/meta-flutter/meta-flutter/commit/67b6816a4f963f85ad7f62ff7aebced2613f3b67

gencay3535, 2022-08-22:
Thank you for your answer, and when can you checkout the CI job for the stm32mp1 with the kirkstone branch?

jwinarske, 2022-08-25:
I'm looking at the kirkstone CI now.  It just passed the eglfs st-core-image and SDK builds.  Looks promising.