meta-flutter/meta-flutter #151: Error compiling for IMX7ULP

ajlennon, 2022-08-19:
I get the following trying to compile for MACHINE=imx7ulp

Something about `gbm` not being found

```
ERROR: Logfile of failure stored in: /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/temp/log.do_configure.524334
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/opt/yocto/sources/meta-imx/meta-bsp/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.20.0.imx.bb:do_populate_sysroot', '/opt/yocto/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/sources/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/opt/yocto/sources/meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/opt/yocto/sources/poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/opt/yocto/sources/meta-imx/meta-bsp/recipes-graphics/imx-gpu-viv/imx-gpu-viv_6.4.3.p4.2-aarch32.bb:do_populate_sysroot', '/opt/yocto/sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot', '/opt/yocto/sources/poky/meta/recipes-graphics/wayland/libinput_1.19.3.bb:do_populate_sysroot', '/opt/yocto/sources/meta-imx/meta-bsp/recipes-graphics/drm/libdrm_2.4.109.imx.bb:do_populate_sysroot', 'virtual:runtimerelease:/opt/yocto/sources/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_populate_sysroot', '/opt/yocto/sources/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/sources/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', '/opt/yocto/sources/poky/meta/recipes-graphics/xorg-lib/libxkbcommon_1.4.0.bb:do_populate_sysroot', '/opt/yocto/sources/meta-imx/meta-bsp/recipes-multimedia/gstreamer/gstreamer1.0_1.20.0.imx.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/opt/yocto/sources/poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gstreamer1.0-plugins-base', 'quilt-native', 'pkgconfig-native', 'pseudo-native', 'clang-cross-arm', 'glibc', 'imx-gpu-viv', 'compiler-rt', 'libinput', 'libdrm', 'flutter-engine-runtimerelease', 'libcxx', 'ninja-native', 'libxkbcommon', 'gstreamer1.0', 'patch-native', 'cmake-native', 'util-linux', 'gcc-runtime', 'pango', 'orc', 'iso-codes', 'wayland', 'libpng', 'libogg', 'gobject-introspection', 'wayland-protocols', 'libvorbis', 'linux-imx-headers', 'imx-gpu-g2d', 'zlib', 'alsa-lib', 'libtheora', 'libtool-native', 'binutils-cross-arm', 'clang-native', 'linux-libc-headers', 'libgcc', 'libcheck', 'libevdev', 'mtdev', 'systemd', 'libpthread-stubs', 'libpciaccess', 'fontconfig', 're2c-native', 'libxml2', 'glib-2.0', 'attr-native', 'ncurses-native', 'curl-native', 'bzip2-native', 'zstd-native', 'xz-native', 'zlib-native', 'opkg-utils', 'bash-completion', 'ncurses', 'util-linux-libuuid', 'libpam', 'libcap-ng', 'libxcrypt', 'freetype', 'cairo', 'harfbuzz', 'fribidi', 'libffi', 'expat', 'autoconf-archive', 'python3', 'texinfo-dummy-native', 'flex-native', 'gnu-config-native', 'libxml2-native', 'binutils-native', 'python3-native', 'swig-native', 'libffi-native', 'libedit-native', 'base-passwd', 'zstd', 'shadow-sysroot', 'libseccomp', 'shadow-native', 'base-files', 'kmod', 'libcap', 'shadow', 'acl', 'util-macros', 'libpcre', 'gettext-minimal-native', 'openssl-native', 'cracklib', 'flex', 'pixman', 'icu', 'libnsl2', 'xz', 'gdbm', 'openssl', 'sqlite3', 'bzip2', 'libtirpc', 'readline', 'm4-native', 'elfutils-native', 'libtirpc-native', 'util-linux-libuuid-native', 'gdbm-native', 'libnsl2-native', 'readline-native', 'sqlite3-native', 'libpcre-native', 'attr', 'perl-native', 'cryptodev-linux', 'libarchive-native', 'libmicrohttpd-native', 'lzo-native', 'libgcrypt-native', 'gnutls-native', 'libcap-native', 'libgpg-error-native', 'libunistring-native', 'p11-kit-native', 'libidn2-native', 'gmp-native', 'nettle-native', 'libtasn1-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- CMAKE_BUILD_TYPE not set, defaulting to Release.
| -- The C compiler identification is Clang 14.0.3
| -- The CXX compiler identification is Clang 14.0.3
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Generator .............. Ninja
| -- Build Type ............. Release
| -- Flutter Header ......... /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot/include/flutter_embedder.h
| -- Found PkgConfig: /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot-native/usr/bin/pkg-config (found version "0.29.2")
| -- Checking for module 'libdrm'
| --   Found libdrm, version 2.4.109
| -- Checking for module 'gbm'
| --   No package 'gbm' found
| -- Configuring incomplete, errors occurred!
| See also "/opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/build/CMakeFiles/CMakeOutput.log".
| CMake Error at /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
|   A required package was not found
| Call Stack (most recent call first):
|   /opt/yocto/build/tmp/work/cortexa7t2hf-neon-mx7ulp-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
|   CMakeLists.txt:103 (pkg_check_modules)
```

jwinarske, 2022-08-19:
@ajlennon does that part have a GPU?

ajlennon, 2022-08-19:
It does oddly enough - we designed a board with it I am trying to get going as it's one of the few of this series that have GPU acceleration so can run Android

jwinarske, 2022-08-19:
Interesting.  What are the DISTRO_FEATURES for your build?

```
bitbake -e ${{ env.IMAGE }} | grep "^DISTRO_FEATURES"
```

ajlennon, 2022-08-19:
`DISTRO_FEATURES="acl alsa argp bluetooth debuginfod ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc  vfat seccomp largefile opengl ptest multiarch    systemd wayland pam systemd pulseaudio gobject-introspection-data ldconfig"`

jwinarske, 2022-08-19:
That looks right.  You need to determine what recipe provides gdm, and add that via a bbappend (e.g. `flutter-pi-runtimerelease.bbappend`)

ivi-homescreen builds?

ajlennon, 2022-08-19:
>ivi-homescreen builds?

Yes that builds. Having the devil of a job working out where gbm is supposed to be coming from. Maybe mesa

jwinarske, 2022-08-19:
Yeah there's a couple other vendors that have similar scenario.  Some require additional libraries etc.  bbappend is your friend in these cases, also a good place to add the system/user service logic.

jwinarske, 2022-08-19:
It's also possible flutter-pi needs adjustment.

@ardera What about targets that don't have gdm package?

ardera, 2022-08-19:
Not sure about the iMX7, but on iMX6 you had to do some strange things to get a rendering surface from the graphics driver, it's completely different to the KMS/GBM way that flutter-pi normally uses. (Basically they have an `fbdev` driver with extra steps). On iMX8 KMS/GBM is supported. Could be iMX7 still only has the old interface, which would explain why there's no `libgbm`.

For example, qt has extra code just for supporting vivante GPUs as well: https://github.com/qt/qtbase/blob/1b96b645ec848ce911ecedbbb31d21b1b2c9d58b/src/plugins/platforms/eglfs/deviceintegration/eglfs_viv/qeglfsvivintegration.cpp

It's possible to add support for that in flutter-pi too, but I don't have a device for testing and never had a reason to add support for it, that's why it's not there

jwinarske, 2022-08-19:
@ardera Makes sense, thanks!

jwinarske, 2022-08-19:
@ajlennon So possibly not supported at this time.  What Hannes provided should be enough to create a patch.

ajlennon, 2022-08-19:
I had a go at running the `ivi homescreen` with the gallery bundle on the i.MX8M Mini. Seems to seg-fault.

[image]

ajlennon, 2022-08-19:
Hmm - it is building `ivi-homescreen` with `-DBUILD_BACKEND_WAYLAND_VULKAN=OFF` so I'm unsure why it seems to be looking for the vulkan shared libraries before it segfaults

[image]

jwinarske, 2022-08-19:
@ajlennon remove vulkan from DISTRO_FEATURES in local.conf and it will build EGL backend.

ajlennon, 2022-08-19:
There's no vulkan in DISTRO_FEATURES and I can see in the devshell it is building with `-DBUILD_BACKEND_WAYLAND_VULKAN=OFF`

`DISTRO_FEATURES="acl alsa argp bluetooth debuginfod ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc  vfat seccomp largefile opengl ptest multiarch    systemdgbm wayland pam systemd pulseaudio gobject-introspection-data ldconfig"`

jwinarske, 2022-08-19:
hmm.  That's not right.  I'll have to take a closer look on Sunday.  There has been a bit of churn lately for AGL release.  Things should be stable again by next Friday.  There are lots of moving parts :)

ajlennon, 2022-08-19:
Hmm this does't look like it should be in there

[image]

ajlennon, 2022-08-20:
My mistake! I confused myself between my imx8mmevk and imx7ulp builds. Vulkan was in DISTRO_FEATURES for imx8mmevk as you thought and with that removed I can now run up ivi_homescreen! So will try and build this for the imx7ulp

[image]

jwinarske, 2022-08-21:
I am closing this as flutter-pi issue.  Feel free to re-open as needed.

closed 2022-08-21 by jwinarske

ajlennon, 2022-08-22:
Having done some experimenting with homescreen I feel I'd really like to have a more basic image (based on core-image-minimal) and use flutter-pi. Seeing the comments above about the surface and libgbm I'm wondering if there might be a place I can pull libgbm from to try to get this going simply?

jwinarske, 2022-08-22:
I would start by asking the NXP support forum about KMS/GBM support for your part in Yocto.  It might be in a side branch, it might not exist, or it might be supported by adding variable to local.conf.  

While waiting for an answer perhaps dig around in the kernel pulled in by the specific yocto build.  I am suspecting Hannes's point about it possibly using prior Vivante support is most likely the case; but rule it out by digging into the kernel source.

ajlennon, 2022-08-22:
okie ta. will have to leave it for a bit. got some integration work to do first and the imx7ulp support is more a fun project than a need for the client atm

ardera, 2022-08-22:
@ajlennon Forgot to mention this, you can try using `etnaviv` instead of the proprietary (`galcore`) Vivante driver. That should work with flutter-pi. Could be you can choose that somewhere in the yocto setup. In buildroot at least you could just choose `etnaviv` in the options and it worked.

Etnaviv is [not perfect though](https://mesamatrix.net/), it's the mesa driver with the least support (except for some WIP ones). Though the proprietary galcore driver is not perfect either, for example on iMX8 the whole system would crash (without a kernel panic or any kind of error message) if you tried to get the last vsync timestamp before the first frame was presented.