meta-flutter/meta-flutter #153: depot-tool-native , bitbake error

Pengli2208, 2022-08-21:
hi
I am using dunfell branch to build basic image with flutter-pi support in raspberrypi4-64;

use bblayer, below
  /home/yocto/poky/meta \
  /home/yocto/poky/meta-poky \
  /home/yocto/poky/meta-yocto-bsp \
  /home/yocto/poky/meta-raspberrypi \
  /home/yocto/poky/meta-openembedded/meta-oe \
  /home/yocto/poky/meta-openembedded/meta-python \
  /home/yocto/poky/meta-openembedded/meta-multimedia \
  /home/yocto/poky/meta-openembedded/meta-networking \
  /home/yocto/poky/meta-clang \
  /home/yocto/poky/meta-flutter \

add flutter in local.conf

FLUTTER_SDK_TAG = "2.10.3"
PACKAGECONFIG:pn-flutter-engine-release = "disable-desktop-embeddings embedder-for-target fontconfig release"
PACKAGECONFIG:pn-flutter-engine-debug = "disable-desktop-embeddings embedder-for-target fontconfig debug"
PACKAGECONFIG:pn-flutter-engine-profile = "disable-desktop-embeddings embedder-for-target fontconfig profile"
FLUTTER_RUNTIME:pn-flutter-drm-gbm-backend = "debug"
FLUTTER_RUNTIME:pn-flutter-wayland-client = "debug"
FLUTTER_CHANNEL = "dev"
IMAGE_INSTALL_append = " flutter-wayland-client"

bitbake core-image-minimal

after building , I get the following error:
ERROR: depot-tools-native-git-r0 do_compile: Execution of '/home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/temp/run.do_compile.352595' failed with exit code 127
ERROR: Logfile of failure stored in: /home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/temp/log.do_compile.352595
Log data follows:
| DEBUG: Executing shell function do_compile
| /home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/git/bootstrap_python3: line 32: bootstrap-2@3.8.9.chromium.14_bin/python3/bin/python3: No such file or directory
| cat: /home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/git/python3_bin_reldir.txt: No such file or directory
| /home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/git/vpython3: line 52: /home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/git/.cipd_bin/vpython3: No such file or directory
| WARNING: exit code 127 from a shell command.
| ERROR: Execution of '/home/yocto/poky/rpi-build/tmp/work/x86_64-linux/depot-tools-native/git-r0/temp/run.do_compile.352595' failed with exit code 127
ERROR: Task (/home/yocto/poky/meta-flutter/recipes-devtools/depot-tools/depot-tools-native_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3619 tasks of which 3605 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/yocto/poky/meta-flutter/recipes-devtools/depot-tools/depot-tools-native_git.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

jwinarske, 2022-08-21:
@Pengli2208 

I'll spin the `dunfell` CI jobs and see if I get a repro.

I'm using the same config for `depot-tools-native` in `kirkstone`, and those are running clear short of a new flutter-pi patch issue.

jwinarske, 2022-08-21:
Looks fine:
https://github.com/meta-flutter/meta-flutter/actions/runs/2899276257

Something about your build environment.

closed 2022-08-21 by jwinarske