meta-flutter/meta-flutter #158: [honister] rpi3 (32-bit) - flutter-pi

jwinarske, 2022-08-22:
Build is failing with:

```
| /__w/meta-flutter/rpi3-32-drm-honister-latest/raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/flutter-pi-runtimerelease/git-r0/git/src/flutter-pi.c:1990:14: error: field designator 'thread_priority_setter' does not refer to any field in type 'FlutterCustomTaskRunners'
|             .thread_priority_setter = NULL
|              ^
```
It only happens on rpi3 32-bit build.

@ardera Have you ran into this one before?

ardera, 2022-08-22:
That field is there for 6 months now: https://github.com/flutter/engine/blame/cd119976dcc943dba362d5e99cdfe448521722ce/shell/platform/embedder/embedder.h#L1188-L1190

any chance it's running with an old `embedder.h` somehow?

jwinarske, 2022-08-22:
Yeah that's what I was thinking.  All the other jobs build flutter-pi, and reference the same header; no issue.  I'll dig a bit deeper.  Thanks.

jwinarske, 2022-08-23:
My bad.  https://github.com/meta-flutter/meta-flutter/pull/159

closed 2022-08-23 by jwinarske