meta-flutter/meta-flutter #164: Workspace Setup Fails

collinbrake, 2022-08-26:
Hi,

I am trying to run the Flutter workspace setup script as described here: https://github.com/meta-flutter/meta-flutter/blob/kirkstone/tools/README.md

Ubuntu 20.04 x86_64 desktop, x11.

The below error occurs when I run `./setup_flutter_workspace.py`:

```
~/meta-flutter/tools$ ./setup_flutter_workspace.py 
***********************************************************************
** Setting up Flutter Workspace in: /home/*******/meta-flutter/tools **
***********************************************************************

* Installing runtime package dependency: libcurl4-openssl-dev
[sudo] password for ventrac: 

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfwupdplugin1
Use 'sudo apt autoremove' to remove it.
Suggested packages:
  libcurl4-doc libidn11-dev libkrb5-dev libldap2-dev librtmp-dev libssh2-1-dev libssl-dev
The following NEW packages will be installed:
  libcurl4-openssl-dev
0 upgraded, 1 newly installed, 0 to remove and 10 not upgraded.
Need to get 322 kB of archives.
After this operation, 1,543 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 libcurl4-openssl-dev amd64 7.68.0-1ubuntu2.12 [322 kB]
Fetched 322 kB in 1s (462 kB/s)              
Selecting previously unselected package libcurl4-openssl-dev:amd64.
(Reading database ... 248254 files and directories currently installed.)
Preparing to unpack .../libcurl4-openssl-dev_7.68.0-1ubuntu2.12_amd64.deb ...
Unpacking libcurl4-openssl-dev:amd64 (7.68.0-1ubuntu2.12) ...
Setting up libcurl4-openssl-dev:amd64 (7.68.0-1ubuntu2.12) ...
Processing triggers for man-db (2.9.1-1) ...
Installing required Python packages: {'requests', 'pycurl'}
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-l6yz9f9g
       cwd: /tmp/pip-install-h7hf3l6u/pycurl/
  Complete output (20 lines):
  Using curl-config (libcurl 7.68.0)
  Using SSL library: OpenSSL/LibreSSL/BoringSSL
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.8
  creating build/lib.linux-x86_64-3.8/curl
  copying python/curl/__init__.py -> build/lib.linux-x86_64-3.8/curl
  running build_ext
  building 'pycurl' extension
  creating build/temp.linux-x86_64-3.8
  creating build/temp.linux-x86_64-3.8/src
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DPYCURL_VERSION="7.45.1" -DHAVE_CURL_SSL=1 -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python3.8 -c src/docstrings.c -o build/temp.linux-x86_64-3.8/src/docstrings.o
  In file included from src/docstrings.c:4:
  src/pycurl.h:178:13: fatal error: openssl/ssl.h: No such file or directory
    178 | #   include <openssl/ssl.h>
        |             ^~~~~~~~~~~~~~~
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for pycurl
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-phrjhyg1/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/ventrac/.local/include/python3.8/pycurl
         cwd: /tmp/pip-install-h7hf3l6u/pycurl/
    Complete output (20 lines):
    Using curl-config (libcurl 7.68.0)
    Using SSL library: OpenSSL/LibreSSL/BoringSSL
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.8
    creating build/lib.linux-x86_64-3.8/curl
    copying python/curl/__init__.py -> build/lib.linux-x86_64-3.8/curl
    running build_ext
    building 'pycurl' extension
    creating build/temp.linux-x86_64-3.8
    creating build/temp.linux-x86_64-3.8/src
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DPYCURL_VERSION="7.45.1" -DHAVE_CURL_SSL=1 -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python3.8 -c src/docstrings.c -o build/temp.linux-x86_64-3.8/src/docstrings.o
    In file included from src/docstrings.c:4:
    src/pycurl.h:178:13: fatal error: openssl/ssl.h: No such file or directory
      178 | #   include <openssl/ssl.h>
          |             ^~~~~~~~~~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h7hf3l6u/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-phrjhyg1/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/ventrac/.local/include/python3.8/pycurl Check the logs for full command output.
Traceback (most recent call last):
  File "./setup_flutter_workspace.py", line 1375, in <module>
    main()
  File "./setup_flutter_workspace.py", line 92, in main
    subprocess.check_call([python, '-m', 'pip', 'install', *missing], stdout=subprocess.DEVNULL)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', 'pycurl']' returned non-zero exit status 1.
```
The error states that `openssl/ssl.h` is missing, so I ran

`sudo apt-get install libssl-dev`

per [the instructions here](https://stackoverflow.com/questions/17915098/openssl-ssl-h-no-such-file-or-directory-during-installation-of-git).

This fixed the issue. Should the [setup script](https://github.com/meta-flutter/meta-flutter/blob/kirkstone/tools/setup_flutter_workspace.py) be modified to take care of this install?

jwinarske, 2022-08-26:
@collinbrake hey thanks for the report.

I'll add the dependency for `libssl-dev`.

jwinarske, 2022-08-26:
Checked in: https://github.com/meta-flutter/meta-flutter/commit/8520df7f2f177bd94c1db62cb6e77ded984b56cb

closed 2022-08-26 by jwinarske