meta-flutter/meta-flutter #167: profile mode sets dart.vm.product to true

ardera, 2022-08-28:
Hey @jwinarske ,

it seems like the flutter-app template will set the dart define `-Ddart.vm.product=true`, regardless of whether the app is being built in profile or release (product) mode.

However, when building apps with the flutter SDK, the flutter SDK will only set that define to true when actually building in release mode. This is the verbose build log in profile mode:

<details>

```
hannes@pop-os:~/devel/cruise-demo/demo$ flutter build linux --profile -v
[  +38 ms] executing: uname -m
[  +21 ms] Exit code 0 from: uname -m
[        ] x86_64
[   +3 ms] executing: [/home/hannes/devel/flutter_sdk/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +2 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] f1875d570e39de09040c8f79aa13cc56baab8db1
[        ] executing: [/home/hannes/devel/flutter_sdk/] git tag --points-at f1875d570e39de09040c8f79aa13cc56baab8db1
[   +8 ms] Exit code 0 from: git tag --points-at f1875d570e39de09040c8f79aa13cc56baab8db1
[        ] 3.0.5
[   +5 ms] executing: [/home/hannes/devel/flutter_sdk/] git rev-parse --abbrev-ref --symbolic @{u}
[   +4 ms] Exit code 128 from: git rev-parse --abbrev-ref --symbolic @{u}
[        ] fatal: HEAD zeigt auf keinen Branch
[  +33 ms] executing: [/home/hannes/devel/flutter_sdk/] git rev-parse --abbrev-ref HEAD
[   +4 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] HEAD
[  +26 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +1 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +22 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[        ] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[        ] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'PubDependencies' is not required, skipping update.
[  +27 ms] Skipping pub get: version match.
[  +15 ms] Found plugin desktop_window at /home/hannes/.pub-cache/hosted/pub.dartlang.org/desktop_window-0.4.0/
[  +11 ms] Found plugin path_provider_linux at /home/hannes/.pub-cache/hosted/pub.dartlang.org/path_provider_linux-2.1.7/
[   +1 ms] Found plugin path_provider_windows at /home/hannes/.pub-cache/hosted/pub.dartlang.org/path_provider_windows-2.1.0/
[   +2 ms] Found plugin shared_preferences at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences-2.0.15/
[   +2 ms] Found plugin shared_preferences_android at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_android-2.0.12/
[   +1 ms] Found plugin shared_preferences_ios at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_ios-2.1.1/
[   +1 ms] Found plugin shared_preferences_linux at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_linux-2.1.1/
[        ] Found plugin shared_preferences_macos at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_macos-2.0.4/
[   +1 ms] Found plugin shared_preferences_web at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_web-2.0.4/
[        ] Found plugin shared_preferences_windows at /home/hannes/.pub-cache/hosted/pub.dartlang.org/shared_preferences_windows-2.1.1/
[  +33 ms] Found plugin desktop_window at /home/hannes/.pub-cache/hosted/pub.dartlang.org/desktop_window-0.4.0/
[... 198 lines cut ...]
[        ] -- Installing: /home/hannes/devel/cruise-demo/demo/build/linux/x64/profile/bundle/data/flutter_assets/assets/images/logo.png
[        ] -- Installing: /home/hannes/devel/cruise-demo/demo/build/linux/x64/profile/bundle/data/flutter_assets/assets/images/music.png
[        ] -- Installing: /home/hannes/devel/cruise-demo/demo/build/linux/x64/profile/bundle/data/flutter_assets/assets/images/radio.icon.png
[        ] -- Installing: /home/hannes/devel/cruise-demo/demo/build/linux/x64/profile/bundle/data/flutter_assets/assets/images/video.png
[        ] -- Installing: /home/hannes/devel/cruise-demo/demo/build/linux/x64/profile/bundle/lib/libapp.so
[  +11 ms] Building Linux application... (completed in 17,4s)
[        ] "flutter linux" took 17.606ms.
[   +3 ms] Running shutdown hooks
[        ] Shutdown hooks complete
[        ] exiting with code 0
hannes@pop-os:~/devel/cruise-demo/demo$
```

</details>

Notice this line where the kernel snapshot generator is invoked:
```
[        ] [   +5 ms] /home/hannes/devel/flutter_sdk/bin/cache/dart-sdk/bin/dart --disable-dart-dev /home/hannes/devel/flutter_sdk/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root
/home/hannes/devel/flutter_sdk/bin/cache/artifacts/engine/common/flutter_patched_sdk/ --target=flutter --no-print-incremental-dependencies -Ddart.vm.profile=true -Ddart.vm.product=false --track-widget-creation --aot --tfa --packages
/home/hannes/devel/cruise-demo/demo/.dart_tool/package_config.json --output-dill /home/hannes/devel/cruise-demo/demo/.dart_tool/flutter_build/57cc96daf4423af8c28e63f10ce0b7f9/app.dill --depfile
/home/hannes/devel/cruise-demo/demo/.dart_tool/flutter_build/57cc96daf4423af8c28e63f10ce0b7f9/kernel_snapshot.d --source /home/hannes/devel/cruise-demo/demo/.dart_tool/flutter_build/dart_plugin_registrant.dart --source
package:flutter/src/dart_plugin_registrant.dart -Dflutter.dart_plugin_registrant=file:///home/hannes/devel/cruise-demo/demo/.dart_tool/flutter_build/dart_plugin_registrant.dart package:demo/main.dart
```

Is there any reason `dart.vm.product` is always set to true in meta-flutter? It seems like that stops it from launching the dart observatory in my case

jwinarske, 2022-08-28:
@ardera I'll spin a profile build for zero2w and take a look.  It's been a while.

jwinarske, 2022-08-28:
@ardera this job has the proposed change, including profile image.  Will smoke check later.
https://github.com/meta-flutter/meta-flutter/actions/runs/2943922014

ardera, 2022-08-28:
@jwinarske thanks for the fast fix :+1:

ardera, 2022-08-28:
Could be the recipe also needs to use `flutter_patched_sdk` instead of `flutter_patched_sdk_product` for profile mode. Still can't see the observatory with `-Ddart.vm.product=false`

jwinarske, 2022-08-28:
Just respun with change.  You might also need to pass `--start-paused` to the VM.

jwinarske, 2022-08-29:
I just tested `weston-profile` from this run:
https://github.com/meta-flutter/meta-flutter/actions/runs/2944815808

Setting up WiFi via `nmcli`, I can connect to flutter-auto and flutter-pi observatory.

flutter-auto with weston enabled:
```
flutter-auto --b=/usr/share/flutter/gallery --w=640 --h=480 --observatory-host=0.0.0.0 --observatory-port=1234
```

flutter-pi with weston disabled:
```
flutter-pi --release /usr/share/flutter/gallery --observatory-host=0.0.0.0 --observatory-port=1234
```

To disable weston compositor:
```
# systemctl disable weston
# reboot
```

ardera, 2022-08-29:
Okay I didn't test the `jw/ci_zero2w` branch but I applied the same changes to dunfell locally and it works, so I guess once this goes into dunfell and kirkstone it's fixed. Thanks 👍

ardera, 2022-08-29:
btw, just for reference, I used `flutter attach --device-url <observatory-url>` to get a devtools instance connected to the profile mode app. Just launching the devtools (using `dart devtools`) and entering the observatory url there is not enough (at least it wasn't in my case)

jwinarske, 2022-08-30:
https://github.com/meta-flutter/meta-flutter/pull/168

closed 2022-08-30 by jwinarske

jwinarske, 2022-08-30:
`honister` was cherry picked from `kirkstone`
`dunfell` gets https://github.com/meta-flutter/meta-flutter/pull/170