meta-flutter/meta-flutter #176: Failed to connect to Wayland display. No such file or directory

gencay3535, 2022-09-07:
Hello, I built the image for stm32mp15 and flashed it. I wanted to run flutter gallery, so from /usr/share/flutter I run 
`flutter-auto --w=800 --h=600 --b=/usr/share/flutter/gallery/`
it come out as below:
`[INFO:configuration.cc(248)] **********
[INFO:configuration.cc(249)] * Global *
[INFO:configuration.cc(250)] **********
[INFO:configuration.cc(251)] Application Id: .......... flutter-auto
[INFO:configuration.cc(255)] Cursor Theme: ............ 
[INFO:configuration.cc(256)] Disable Cursor: .......... false
[INFO:configuration.cc(258)] Debug Backend: ........... false
[INFO:configuration.cc(260)] ********
[INFO:configuration.cc(261)] * View *
[INFO:configuration.cc(262)] ********
[INFO:configuration.cc(269)] Bundle Path: .............. /usr/share/flutter/gallery/
[INFO:configuration.cc(270)] Window Type: .............. NORMAL
[INFO:configuration.cc(271)] Size: ..................... 800 x 600
[INFO:configuration.cc(273)] Fullscreen: ............... false
[INFO:configuration.cc(275)] Accessibility Features: ... 0
[ERROR:/usr/src/debug/flutter-auto-runtimerelease/git-r0/git/shell/wayland/display.cc(51)] Failed to connect to Wayland display. No such file or directory
`
so what I am missing?

jwinarske, 2022-09-07:
You're trying to run the app from root.

From root do
```
passwd -d weston
```
Then login as weston, then try again.

If you're using custom-device support via tools/setup_flutter_workspace
This happens automatically.

See the pattern here:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/tools/flutter_workspace_config.json#L88

gencay3535, 2022-09-07:
Okay I switched user, and when I do
`weston --use-pixman`
I get this:
`Date: 2022-04-28 UTC
[21:26:22.502] weston 10.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 10.0.0
[21:26:22.503] Command line: weston --use-pixman
[21:26:22.503] OS: Linux, 5.15.24, #1 SMP PREEMPT Wed Feb 16 11:56:41 UTC 2022, armv7l
[21:26:22.503] Flight recorder: enabled
[21:26:22.503] fatal: environment variable XDG_RUNTIME_DIR
is set to "/run/user/1000", which is not a directory.
Refer to your distribution on how to get it, or
http://www.freedesktop.org/wiki/Specifications/basedir-spec
on how to implement it.
`

jwinarske, 2022-09-07:
I'm confused, why are you running `weston --use-pixman`?

If the image is built correctly weston (the compositor) will be running as a session under the `weston` user, and you will see a desktop.

Just login as user `weston` remote or local and run `flutter-auto --w=800 --h=600 --b=/usr/share/flutter/gallery`

Alternatively open a terminal from the desktop and run `flutter-auto --w=800 --h=600 --b=/usr/share/flutter/gallery`

If you don't have a user `weston` then your OS configuration is different than the CI job:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-stm32mp15.yml#L15

gencay3535, 2022-09-07:
This is my local.conf I removed two recipes because they were giving fetcher error:
`ACCEPT_EULA_stm32mp15-disco = "1"
IMAGE_LINGUAS = "en-us"
TOOLCHAIN_HOST_TASK:append = " nativesdk-flutter-sdk"
CLANGSDK = "1"
DISTRO_FEATURES:remove = " x11 "
PACKAGECONFIG:pn-flutter-auto-runtimerelease = "backend-egl mouse-cursor package-info secure-storage text-input texture-test-egl"
IMAGE_INSTALL:remove = " packagegroup-st-demo "
IMAGE_INSTALL:remove = " gtk+3-demo "
IMAGE_ROOTFS_MAXSIZE = "3145728"
IMAGE_INSTALL:append = " flutter-auto-runtimerelease flutter-engine-runtimerelease-sdk-dev flutter-gallery-runtimerelease flutter-test-video-player-runtimerelease adwaita-icon-theme-cursors xdg-user-dirs avahi-daemon ntp ntp-tickadj gstreamer1.0-libav gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly"
LICENSE_FLAGS_ACCEPTED += " commercial"`
removed recipes are `flutter-test-texture-egl-runtimerelease \
                    flutter-test-localization-runtimerelease \`
There is a user weston, when I login with su - weston then run the above command I get the same error:
`[ERROR:/usr/src/debug/flutter-auto-runtimerelease/git-r0/git/shell/wayland/display.cc(51)] Failed to connect to Wayland display. No such file or directory
`

jwinarske, 2022-09-07:
If from serial console type `exit` to logout of terminal, then at login prompt enter `weston`.  Then run the command.

If in desktop open terminal, and run the command.

This may also be useful:
https://github.com/meta-flutter/meta-flutter/wiki/Weston-Compositor-Notes

closed 2022-09-08 by gencay3535

gencay3535, 2022-09-08:
Finally I can see the gallery app running but it runs a little bit slow. Here is the [video](https://1drv.ms/v/s!AtjdUEfjBNnoh0mLik-4aTNPIgOt?e=GeXoQA) if you want to see it. Is it running on cpu or gpu?

jwinarske, 2022-09-08:
Which build variant are you running?  Release/AOT?

If the Weston service is using `--use-pixman` then it's CPU.

Keep in mind this is an 800mHz (newer parts) 32-bit device, and the gallery app is pretty storage/memory heavy.  Perhaps try the flutter-app-igalia-homescreen recipe out; more of a simple design.

The other thing in question is the SD card bandwidth on the disco boards.  I think an eMMC would be a better choice, but I haven't looked into what the hardware SD controller can support (version/max bandwidth).

The touch controller may also introduce some latency; which could be dialed in on a custom design.  If USB mouse is identical latency then no.

jwinarske, 2022-09-08:
For testing on the STM32MP157F-DK2 I generally always use external HDMI monitor, and USB mouse.  It avoids touch controller, and display (platform specific) issues.

gencay3535, 2022-09-09:
Yes, igalia-homescreen works much smoother. I am running release version. Is there any way to choose to run from cpu or gpu and when I use `systemctl status weston` from root it says `Unit weston.service could not be found`. Is it normal for this build?

gencay3535, 2022-09-09:
> Yes, igalia-homescreen works much smoother. I am running release version. Is there any way to choose to run from cpu or gpu? And other thing is when I use `systemctl status weston` from root it says `Unit weston.service could not be found`. Is it normal for this build?

when I `cat /etc/default/weston` it returns `WESTON_USER=weston` so I think this means, gpu is being used?

jwinarske, 2022-09-09:
What about `systemctl status weston@weston` or `systemctl status weston@root`?

jwinarske, 2022-09-09:
`weston-info` may also be an option.

gencay3535, 2022-09-09:
> What about `systemctl status weston@weston` or `systemctl status weston@root`?

again `weston.service could not be found.`

jwinarske, 2022-09-09:
I'll boot an image later today and take a look

jwinarske, 2022-09-09:
It looks like the `weston-init` recipe is not being used.  The configuration is questionable.  Might be better to add on minimal Wayland to eglfs image.  I'll look into later.

`ps aux |grep weston`
```
root      1245  0.0  0.3   3152  1416 tty7     Ss+  21:47   0:00 /usr/bin/weston-launch -- --modules=systemd-notify.so --log=/home/weston/weston.log
weston    1308  0.2  1.3   7568  5384 ?        Ss   21:47   0:01 /lib/systemd/systemd --user
```

`less /home/weston/weston.log`
```
Date: 2022-04-28 UTC
[17:42:57.299] weston 10.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 10.0.0
[17:42:57.299] Command line: /usr/bin/weston --modules=systemd-notify.so --log=/home/weston/weston.log
[17:42:57.299] OS: Linux, 5.15.24, #1 SMP PREEMPT Wed Feb 16 11:56:41 UTC 2022, armv7l
[17:42:57.300] Flight recorder: enabled
[17:42:57.306] Using config file '/etc/xdg/weston/weston.ini'
[17:42:57.308] Output repaint window is 100 ms maximum.
[17:42:57.314] Loading module '/usr/lib/libweston-10/drm-backend.so'
[17:42:57.350] initializing drm backend
[17:42:57.350] Trying logind launcher...
[17:42:57.403] logind: session control granted
[17:42:57.434] using /dev/dri/card0
[17:42:57.434] DRM: supports atomic modesetting
[17:42:57.434] DRM: supports GBM modifiers
[17:42:57.435] DRM: supports picture aspect ratio
[17:42:57.449] Loading module '/usr/lib/libweston-10/gl-renderer.so'
[17:42:57.506] EGL client extensions: EGL_EXT_client_extensions
```

Looks like GPU.  I think if they added eMMC to the disco board, the cost would go way up.  Unclear if the SD interface can handle 50MHz or greater, or what version of eMMC is supported.

Disabling a bunch of things in weston.ini seems to help:
`vi /etc/xdg/weston/weston.ini`
disable:
[core]
#repaint window
#require-input

[shell]
#background-image
#background-color
#background-type
#panel-color
#panel-position
#animation
#startup-animation
#close-animation
#focus-animation
#binding-modifier

#[screen-share]
#command...
`:wq`
then `reboot`
That should give you a plain wayland desktop, and in my case the mouse and cursor started working.

Open a terminal and type `weston-info` to see what the platform supports.

jwinarske, 2022-09-09:
Running gallery I'm seeing this kind of output in the system log:

```
[ 1180.252090] [drm] ltdc fifo underrun: please verify display mode
[ 1185.028871] usb 1-1.1.3: reset low-speed USB device number 5 using ehci-platform
[ 1185.768944] usb 1-1.1.2: reset low-speed USB device number 4 using ehci-platform
[ 1186.849439] usb 1-1.1.3: reset low-speed USB device number 5 using ehci-platform
[ 1187.649294] usb 1-1.1.2: reset low-speed USB device number 4 using ehci-platform
[ 1188.649062] usb 1-1.1.3: reset low-speed USB device number 5 using ehci-platform
[ 1189.569011] usb 1-1.1.2: reset low-speed USB device number 4 using ehci-platform
[ 1190.458931] usb 1-1.1.3: reset low-speed USB device number 5 using ehci-platform
[ 1191.117992] [drm] ltdc fifo underrun: please verify display mode
```

I have no idea how well the STM32MP15 BSP has been optimized.  Based on the BSP, and some of the configs it's probably not that well optimized.

I would say it's fine for basic/simple stuff.  If you want an eye popping display, the STM32MP15 is not the right choice.  A NVidia or Qualcomm SoC would be a better choice for important UI/UX work.

jwinarske, 2022-09-09:
```
root@stm32mp15-disco:~# cat README-CHECK-GPU 
Check GPU
=========

GPU can be deactivated by fuse or not present in some SOC

So in case the GPU is not present, you must configure weston on pix-man by creating this file :
/etc/default/weston with this content :

OPTARGS=--use-pixman
root@stm32mp15-disco:~# cat /etc/default/weston
WESTON_USER=weston
```

Weston is definitely using the GPU.

gencay3535, 2022-09-12:
We will be using eMMC on the custom board, hopefully it will be an in vehicle infotainment with a couple of gauges at the end of the project and we are planing to use flutter for the UI. Igalia-homescreen worked smoother and there was not any errors on the system log so what do you think would it be acceptable performance using stm32mp15 with flutter for a low cost system?

jwinarske, 2022-09-12:
@gencay3535 
The STM32MP15 part is similar to a Qualcomm SoC from ~2008 (MSM720x).  The GPU is on par with lower end NXP iMX parts.  The point of flutter is around saving NRE and using commodity programmers.  If you use a low powered target, you don't really benefit from the NRE savings, more likely increase it as you have to train the platform specialists on Flutter/Dart.  Cost of ownership is around changes post launch.  If you lose the people involved in the initial NRE, then you have to do that all over.  If you go with a higher power SoC (higher BOM cost), then you will have less NRE, and lower cost of ownership as it becomes more general (commodity programmers require less ramp up).

That withstanding it all boils down to what the user expectations are in your market.

STMicro needs a STM32MP15 discovery board with eMMC, and a better display/touch controller for a proper evaluation.

Personally I would use something a bit more powerful: Qualcomm, Nvidia, or NXP iMX8.

gencay3535, 2022-09-30:
So have you tried flutter on NXP iMX8M-mini or iMX8M-plus? If you tried how was the performance and I don't see vulkan for them so would that create a problem?