meta-flutter/meta-flutter #182: Error building against the head of kirkstone branch

ajlennon, 2022-09-10:
I've had to move to a new system as the prior system failed. Have pulled the latest kirkstone but am getting an error

Any thoughts :)

[image]

jwinarske, 2022-09-10:
I'm about to merge 3.3.1 into kirkstone.  A few more CI jobs to go. 

What's the build environment, and is this a clean room build?

Did you ever cancel your build mid-build?  If yes be sure to clear the stored state.

ajlennon, 2022-09-10:
Yes clean room build in Docker container with Ubuntu 22.04.01 LTS

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "imx7ulpevk"
DISTRO               = "fsl-imx-wayland"
DISTRO_VERSION       = "5.15-kirkstone"
TUNE_FEATURES        = "arm vfp cortexa7 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-oe              
meta-multimedia      
meta-python          
meta-freescale       
meta-freescale-3rdparty 
meta-freescale-distro 
meta-bsp             
meta-sdk             
meta-ml              
meta-v2x             
meta-nxp-demo-experience 
meta-chromium        
meta-clang           
meta-gnome           
meta-networking      
meta-filesystems     
meta-qt6             
meta-virtualization  = "<unknown>:<unknown>"
meta-flutter         = "kirkstone:f025cfaea0f203903c8ff042323a1246f63c2881"
meta-phasora         = "main:aa0d39aeb24f0eeb5753813861db476f0ea11805"
```

jwinarske, 2022-09-10:
I haven't tried 22.04.  All the kirkstone CI is running 20.04 (this container):
https://github.com/meta-flutter/docker-containers/blob/main/ubuntu-20-dev/Dockerfile
The container is published as public, so you can pull it.

I periodically clean the server sstate and workspace folders, so regular clean room builds happen.  I've never hit this issue.

Generally any sign of corruption is related to killing an on going build.  Where the following should clean it up:
```
bitbake flutter-sdk-native -c do_cleansstate
bitbake <your app> -c do_cleansstate
bitbake <your app>
```

ajlennon, 2022-09-10:
That did it thanks. I think I was telling you porky pies. I am doing a clean build here but I must have built then done a pull on meta-flutter kirkstone.

Cleaning the build demo app here got it all sorted out

`bitbake -c cleansstate flutter-gallery-runtimerelease`

Now trying to work out how to set up the workspace so I can do a test app build to run with flutter-pi on the i.MX8M Mini

Thanks again!

closed 2022-09-10 by ajlennon

jwinarske, 2022-09-10:
Cool.  In workspace config the RPI zero config should work if you change the hostname to match yours.  Or just use it's ip address.

ajlennon, 2022-09-10:
Awesome! I'll be super pleased if I can get a little hello world app running on it :-D