meta-flutter/meta-flutter #183: kirkstone: setup_flutter_workspace is failing

ajlennon, 2022-09-10:
Apologies but I have run into another problem here

Running `setup_flutter_workspace.py` seems to do quite a lot then errors with

[image]

jwinarske, 2022-09-10:
I'm pretty sure you're running the script from a VS Code terminal, as it looks like the Dart tool watcher is kicking in.

Be sure to run the script from a clean non-VS Code terminal.  Otherwise the VS Code tooling tries to do things before the workspace is ready.

ajlennon, 2022-09-10:
No I'm running from a vanilla terminal?

jwinarske, 2022-09-10:
The other thing to check is the number of flutter occurrences in your path.  The script will remove the first instance, but not more.

jwinarske, 2022-09-10:
Also be sure to pull on kirkstone, as I just merged updates to config and script.

ajlennon, 2022-09-10:
```
ajlennon@ajlennon-Laptop-12th-Gen-Intel-Core:/data_drive/phasora/flutter/meta-flutter/tools$ which flutter
/snap/bin/flutter
```

Will pull again now

ajlennon, 2022-09-10:
Flutter doctor does have some awareness of the VS code installation. Dunno if that's relevant

```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.3.1, on Ubuntu 22.04.1 LTS 5.17.0-1016-oem, locale en_GB.UTF-8)
[✓] Linux toolchain - develop for Linux desktop
[✓] VS Code (version 1.71.0)
[✓] Connected device (1 available)
[✓] HTTP Host Availability
```

jwinarske, 2022-09-10:
Basically you have an instance of flutter running behind the scenes of the terminal which is causing the problem.  A full log would be helpful, opposed to a screen shot.  Just file drag into response window

ajlennon, 2022-09-10:
[ws.log](https://github.com/meta-flutter/meta-flutter/files/9540708/ws.log)

ajlennon, 2022-09-10:
ok this is interesting, so I

- removed the meta-flutter tree
- checked out kirkstone again
- ran  `../meta-flutter/tools/setup_flutter_workspace.py` from outside `meta-flutter`

And that seems to have completed!

ajlennon, 2022-09-10:
Hello World createed from the README and running locally

[image]

jwinarske, 2022-09-10:
Yeah one would want to use the Flutter Workspace as the origin of truth on a dev machine.  It's much more flexible, self contained, and the whole folder can be archived.  It has the same stuff as snap, serves a wider audience, with just a different delivery mechanism.

When you have multiple custom devices, and you don't specify the device using `--device-id=<custom device id>`, it will present a list.

ajlennon, 2022-09-10:
Sorry but I am super confused about my next step. I am trying `flutter build bundle` copying across and then running `flutter-pi` moving files around to fit with the `data` hierarchy but I get an error about debug versus release mode.

Ideally I guess I'd know how to set up a custom device in VS.code to automatically debug into `flutter-pi` with but I cant seem to find any docs for this?

jwinarske, 2022-09-10:
what variant are you running on the target?

ajlennon, 2022-09-10:
I followed the instructions in tools to create a hello world app. Then I did `flutter build bundle` which I thought would give me a debug bundle. Then I tried running that with flutter-pi on the target.

I've now just had a go at using the flutter command line to create a custom device and am trying

`flutter -d imx8mmini run` 

Which is sort of sitting there atm

```
Launching lib/main.dart on i.MX8M Mini in debug mode...
Pseudo-terminal will not be allocated because stdin is not a terminal.
```

ajlennon, 2022-09-10:
ooooh it's copied the files across. 

```
root@imx8mmevk:~/helloworld_flutter_assets# ls /tmp/hello_world/
AssetManifest.json  FontManifest.json  NOTICES.Z  fonts  kernel_blob.bin  packages  shaders  version.json  vm_snapshot_data
```

jwinarske, 2022-09-10:
This readme for `meta-agl-flutter` shows current usage of meta-flutter variants.
https://gerrit.automotivelinux.org/gerrit/c/AGL/meta-agl-devel/+/27976/5/meta-agl-flutter/README.md

If you have the configuration of your target correct prior to running the script, you would go into app folder and type:
`flutter run --device-id=<custom device id>`.

jwinarske, 2022-09-10:
To use `flutter run` the target needs to be running `runtimedebug` bits.

ajlennon, 2022-09-10:
With verbose enabled it's showing

```
[   +2 ms] Launching lib/main.dart on i.MX8M Mini in debug mode...
[   +2 ms] /data_drive/phasora/flutter/workspace/flutter/bin/cache/dart-sdk/bin/dart --disable-dart-dev
/data_drive/phasora/flutter/workspace/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root
/data_drive/phasora/flutter/workspace/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk/ --incremental --target=flutter
--debugger-module-names --experimental-emit-debug-metadata -DFLUTTER_WEB_AUTO_DETECT=true --output-dill
/tmp/flutter_tools.EGMDEP/flutter_tool.EZNTSU/app.dill --packages
/data_drive/phasora/flutter/workspace/app/hello_world/.dart_tool/package_config.json -Ddart.vm.profile=false -Ddart.vm.product=false
--enable-asserts --track-widget-creation --filesystem-scheme org-dartlang-root --initialize-from-dill
build/c075001b96339384a97db4862b8ab8db.cache.dill.track.dill --verbosity=error --enable-experiment=alternative-invalidation-strategy
[   +5 ms] Initializing file store
[   +2 ms] Done initializing file store
[   +7 ms] Skipping target: gen_localizations
[   +2 ms] Skipping target: gen_dart_plugin_registrant
[ +194 ms] Skipping target: kernel_snapshot
[ +149 ms] Skipping target: copy_flutter_bundle
[        ] Persisting file store
[   +3 ms] Done persisting file store
[   +4 ms] executing: ssh -o BatchMode=yes root@192.168.0.131 rm -rf "/tmp/hello_world"
[   +4 ms] <- compile package:hello_world/main.dart
[ +260 ms] executing: scp -r -o BatchMode=yes build/flutter_assets root@192.168.0.131:/tmp/hello_world
[+9602 ms] executing: ssh -o BatchMode=yes root@192.168.0.131 
[   +6 ms] Pseudo-terminal will not be allocated because stdin is not a terminal.
```

Sorry I am not following `runtimedebug`

ajlennon, 2022-09-10:
I wasn't sure what to put in the SDK bit. Was hoping it defaulted to something but maybe that's where I've gone wrong

```
{
  "custom-devices": [
    {
      "id": "imx8mmini",
      "label": "i.MX8M Mini",
      "sdkNameAndVersion": "",
      "platform": null,
      "enabled": true,
      "ping": [
        "ping",
        "-c",
        "1",
        "-w",
        "1",
        "192.168.0.131"
      ],
      "pingSuccessRegex": null,
      "postBuild": null,
      "install": [
        "scp",
        "-r",
        "-o",
        "BatchMode=yes",
        "${localPath}",
        "root@192.168.0.131:/tmp/${appName}"
      ],
```

jwinarske, 2022-09-10:
```
IMAGE_INSTALL += " \
    flutter-pi-runtimedebug \
"
```

If you have flutter-pi-runtimerelease on your target, then you can't use `flutter run`.  Dev image and demo/production/release image.

jwinarske, 2022-09-10:
The custom-device config just needs IP address and id changed.  If you keep the rest, it should just work - if you have `flutter-pi-runtimedebug` installed on your image.

ajlennon, 2022-09-10:
AHA! Brill thanks :) Will get on it. Really appreciate all your time and support Joel. I owe you a pint or non-alcoholic beverage of your choice!

jwinarske, 2022-09-10:
You may have done this already but adding for emphasis, after creating the workspace with `setup_flutter_workspace.py` you need to source the flutter workspace environment.

`source <my flutter workspace folder>/setup_env.sh`

setup_env.sh can be run from any folder.

jwinarske, 2022-09-11:
Then start VS Code from the terminal.
```
cd $FLUTTER_WORKSPACE
code .
```
Then you can use the built in debugger, select device+app, and select the play transport icon to begin debugging.

closed 2022-09-11 by jwinarske

ajlennon, 2022-09-12:
So I have the `flutter-pi-runtimedebug` recipe built in now (`flutter-pi` seems to look the same when I run it). Still getting the same behaviour when I execute `flutter -d $DEVICE run` from the workspace tree though.

ajlennon, 2022-09-12:
Getting there slowly. I had to take out the `meta-flutter` `flutter-pi` file location patch and move `icudtl.dat`

` cp /usr/share/flutter/icudtl.dat /usr/lib`

That leaves me seeing some errors that look related to the display driver setup on the imx8mm

[image]

jwinarske, 2022-09-12:
The moving of icudtl.dat doesn't quite make sense, as I patch that in meta-flutter, and it works on release image.  Unless you're cross compiling flutter-pi tip of tree using Yocto SDK; as that would skip the patch.  So by default you just use the system installed icudtl.dat as-is, no copy.

The error around DRM makes sense.  Until I have leased DRM implementation merged, Weston needs to be disabled for flutter-pi (DRM).

Given I haven't seen your config in full it's hard to comment further.

Also when you grab the setup_flutter_workspace.py log it should be:
```
./setup_flutter_workspace.py |& tee setup.log
```

Otherwise it leaves quite a bit out.

Also in the case of STM32MP15 (Vibrante) flutter-auto works much better than flutter-pi.  You might also compare flutter-auto to flutter-pi on your iMX7 target.

ajlennon, 2022-09-12:
>The moving of icudtl.dat doesn't quite make sense, as I patch that in meta-flutter, and it works on release image. 

I think that'll be as I had to take the patch out for flutter-pi to work

https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-pi/files/0001-path-updates.patch

jwinarske, 2022-09-12:
Was that before or after https://github.com/meta-flutter/meta-flutter/pull/160 ?

ajlennon, 2022-09-12:
OK so I've just done a pull for the latest `kirkstone` #1724903a

I rebuild, then went to my `hello world` app in the host workspace and `flutter -d imx8mm install`

Then on the target

```
cd /tmp/hello_world
root@imx8mmevk:/tmp/hello_world# flutter-pi .                                                                                                                  
[flutter-pi] Could not find "kernel_blob.bin" file inside "/tmp/hello_world/data/flutter_assets", which is required for debug mode.
```
I can fix this by removing your patch, but then I also remove the patched location of the `icudtl.dat`

jwinarske, 2022-09-12:
using meta-flutter flutter-pi on target the pattern needs to follow the bundle folder structure:
<bundle folder>
  data
    flutter_assets
  lib
    libapp.so <release/profile>
    
`flutter-pi <bundle folder>`

jwinarske, 2022-09-12:
The bundle folder layout comes from the Linux GTK embedder post build folder structure.

It allows a number of things like overriding the `libflutter_engine.so` and `icuctl.dat` with custom versions.  flutter-auto implements these override paths and allows a desktop workflow that makes switching SDK versions independent of the target.  It can also be implemented on the target in a case of LTS.  The mobile pattern is to have an engine library per app.  This is single engine per system is the default, where enabling multiple engine per app as needed.  flutter-pi is not yet conformant to this pattern.  I submitted an issue, and Hannes has it as an enhancement.

The implementation for the platform is 100% contained in your own version of `flutter_workspace_config.json`.  So the creation of the bundle folder is done in the install callback.  Take a look at desktop-auto and AGL-qemu custom device configs for implementation pattern.

The bonus here is once you have your own org `flutter_workspace_config.json`, it's easy to replicate a setup to support your custom platforms.

ajlennon, 2022-09-12:
Ok cool. I figured there must be a good reason for you doing it. Ok great I'll setup for this pattern.

Is the workspace config generated by the setup workspace script? Feels like it would ease things if the tool followed the pattern is so?

(Not asking others to do work for me! I could maybe have a go at a PR if the above is sensible? Or is there a reason the setup workspace script doesn't do this?)

jwinarske, 2022-09-12:
The workspace config is separate and intended to be unique per organization.  If there is no config present then it just pulls stable and installs nothing else.

For the meta-flutter workspace config I will add custom devices for each platform in CI as I have time.

The current priority is getting things ready for upcoming AGL F2F workshop in Tokyo.  It will be recorded and freely available.

ajlennon, 2022-09-12:
Awesome. Not to pry but I'm suspecting you're heavily involved with the flutter community then?

ajlennon, 2022-09-13:
OK so for anybody reading this, here's an example of a `flutter_workspace_config.json` that copies over files in the needed format

```
{
  "custom-devices": [
    {
      "id": "imx8mm",
      "label": "imx8mm",
      "sdkNameAndVersion": "",
      "platform": null,
      "enabled": true,
      "ping": [
        "ping",
        "-c",
        "1",
        "-w",
        "1",
        "10.0.31.109"
      ],
      "pingSuccessRegex": null,
      "postBuild": null,
      "install": [
	"bash", "-c", "ssh -t -oBatchMode=yes root@10.0.31.109 mkdir -p \"/tmp/${appName}/data/flutter_assets\" && scp -oBatchMode=yes -r \"${localPath}\" root@10.0.31.109:/tmp/${appName}/data"
      ],
      "uninstall": [
        "ssh",
        "-o",
        "BatchMode=yes",
        "root@10.0.31.109",
        "rm -rf \"/tmp/${appName}\""
      ],
      "runDebug": [
	"bash", "-c", "ssh -t -oBatchMode=yes root@10.0.31.109 flutter-pi -r 90 -i /dev/input/event2 /tmp/${appName}"
      ],
      "forwardPort": [
        "ssh",
        "-o",
        "BatchMode=yes",
        "-o",
        "ExitOnForwardFailure=yes",
        "-L",
        "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}",
        "root@10.0.31.109",
        "echo 'Port forwarding success'; read"
      ],
      "forwardPortSuccessRegex": "Port forwarding success",
      "screenshot": null
    }
  ]
}
```