meta-flutter/meta-flutter #187: How to include platform specific code in flutter application

bojidartonchev, 2022-09-12:
Hello,

Is there a way to include a platform specific code. Currently if I try to call the MethodChannel, I get MissingPluginException?

Best regards!

jwinarske, 2022-09-12:
@bojidartonchev Greetings

What embedder, generic use case scenario, and desired outcome?

jwinarske, 2022-09-12:
okay I think I understand.  You want to use 1P Linux plugins with one of the custom embedders?

bojidartonchev, 2022-09-12:
I have a MethodChannel with one function GetMacAddress, which uses platform specific c++ APIs. Is there a way to use that platform code with flutter-pi for example?

jwinarske, 2022-09-12:
The model to augment flutter-pi/flutter-auto plugins:

1. fork repo in question
2. create your static plugin adding CMake flag to enable it 
3. confirm build flag works as desired
4. create a new meta layer adding a minimum of this bbappend:
```
recipes-graphics/flutter-pi/flutter-pi_git.bbappend
```

Pattern of bbappend contents
```
SRC_REPO = "github.com/jwinarske/flutter-pi.git"
SRCREV = "my commit"
EXTRA_OECMAKE += " -DBUILD_MYCUSTOM_PLUGIN"
```

I just added support for this:
https://github.com/meta-flutter/meta-flutter/commit/1724903a9dc96bfbf7f1a1ddee4931d4e46869cb

jwinarske, 2022-09-13:
@bojidartonchev Does this make sense?

bojidartonchev, 2022-09-13:
@jwinarske Yes, I will give it a try today and will let you know the result. Thank you!

bojidartonchev, 2022-09-14:
It worked. Thank you!

closed 2022-09-14 by bojidartonchev