meta-flutter/meta-flutter #190: Error running  setup_env.sh Channel unknown, 3.3.1.

bernardcraddock, 2022-09-16:
Hello,
Yesterday i installed setup_flutter_workspace.py  within parallels Ubuntu  on my mac 12.6 and all went fine - very nice

Today I tried running setup_env.sh  which complained building flutter  with Channel unknown, 3.3.1.
and when running qemu_run a new window launches then immediately closes.
 
Today flutter bumped stable from 3.3.1 to 3.3.2 and i patched accordingly  as ped working fine 

I don't really want to mess about on my end, so was wondering can you bump meta-flutter to 3.3.2

please advise
best regards

parallels > . ./setup_env.sh
SCRIPT_PATH=/home/parallels/Documents/GitHub/meta-flutter/tools
********************************************
* Setting FLUTTER_WORKSPACE to:
* /home/parallels/Documents/GitHub/meta-flutter/tools
********************************************
Building flutter tool...
[!] Flutter (Channel unknown, 3.3.1, on Ubuntu 20.04.5 LTS 5.15.0-46-generic, locale en_US.UTF-8)
    ! Flutter version 3.3.1 on channel unknown at /home/parallels/Documents/GitHub/meta-flutter/tools/flutter
    ! Upstream repository unknown
    • Framework revision 4f9d92fbbd (9 days ago), 2022-09-06 17:54:53 -0700
    • Engine revision 3efdf03e73
    • Dart version 2.18.0
    • DevTools version 2.15.0

jwinarske, 2022-09-16:
I can bump meta-flutter.  Keep in mind AGL Needlefish is locked to 3.3.1.  Referencing something being on a channel such as `stable` represents a rolling tip of tree.  Having a rolling commit for a release is not desirable.

jwinarske, 2022-09-16:
Also the channel `error` is normal. It's just saying you're not on the latest (tip off tree) for that channel.

closed 2022-09-16 by jwinarske

jwinarske, 2022-09-16:
https://github.com/meta-flutter/meta-flutter/pull/192

bernardcraddock, 2022-09-17:
Rolling tip not desirable I can appreciate that, I should have  emphases my real issue is that qemu_run would launch a new window then immediately close it, I just assumed issue was flutter release related, but now I'm not so sure, perhaps its qemu related. I will try again today and advise regards

jwinarske, 2022-09-17:
You can set the bash debug flag in setup_env.sh and examine the output.

Are you running from a container?  It may require custom tailored qemu flags.  I've not tried that.

bernardcraddock, 2022-09-18:
Working thank you i had 2 issues

1. Multiple flutter installs 
The original snap flutter install was conflicting  with new meta-flutter/tools/flutter install once i separated those setup_flutter_workspace.py ran correctly

2. invalid kvm args  launching qemu-system-x86_64
whilst debugging setup_env I realised i hadn't yet enabled kvm on my mac  so removed kvm  options 
 "-enable-kvm -cpu kvm64"  and qemu-system-x86_64 started fine. 
I'll look at enabling Intel® VT) if necessary

Once up i could navigate to meta-flutter/tools/app and launch gallery, flutter_homescreen, flutter_animated_background ( failed missing lib/main.dart), flutter_secure_storage (failed missing some things) and ped

Couple of points/ questions ? 
a) Qemu is grabbing and lockng my mouse any idea how to stop that ? 
b) How do i run using profile mode so as i can check Flutter DevTools as debug mode is kill it ?
c) Why only 5 apps where are the rest from /recipes-graphics/flutter-apps - i'd like to play around with all of them?
d) how often do i need to re-run  setup_flutter_workspace and rebuild flutter-auto ?
e) When launching PED it is spinning  on "Detecting Locations" which is a bit of a worry, but it's also giving  different feedback depending upon which run option i use Toyota or AGL. As follows Toyota throws an exception whilst AGL does not, and running on Ubuntu desktop-linux  just works fine, obviously i'm not asking you to debug this for me,  but I do need your help what can i enable within qemu system logging for Toyota and AGL to display more information on what's going on. 

Could it just be that I need location services enabled in qemu ?

 launching ped using  Toyota
[1]: Toyota flutter-auto (desktop-auto) 1 
display bunch of stuff .../... then
[INFO:engine.cc(94)] (0) flutter: [2022-09-18 19:09:35.435717]: DEBUG: LocationDataSource : Checking Location Service Enabled
 then throws  unhandled Exception ERROR:flutter/runtime/dart_vm_initializer.cc(41)] 
Unhandled Exception: FileSystemException: Cannot open file, path =
'/home/parallels/Documents/GitHub/meta-flutter/tools/.config/flutter/dconf/user' 
(OS Error: No such file or directory, errno = 2)

it never returns  from location service is enable checking and just sits there until i kill it. 

Launching  ped using AGL 
[2]: AGL x86_64 QEMU Image (AGL-qemu)
display bunch of stuff .../... then
[INFO:engine.cc(94)] (0) flutter: [2022-09-18 09:26:58.795786]: DEBUG: LocationDataSource : Checking Location Service Enabled
[INFO:engine.cc(94)] (0) flutter: [2022-09-18 09:26:58.820782]: DEBUG: LocationDataSource : Location Service is Enabled ? true
[INFO:engine.cc(94)] (0) flutter: [2022-09-18 09:27:02.195128]: DEBUG: SplashScreen : Backend is not under maintenance.

then just sits here until i kill it

launch ped using Ubuntu linux-desktop (it just works)
display bunch of stuff .../... then
[2022-09-18 18:56:25.241]: DEBUG: LocationDataSource : Checking Location Service Enabled
[2022-09-18 18:56:25.243]: DEBUG: LocationDataSource : Location Service is Enabled ? true
[2022-09-18 18:56:25.380]: DEBUG: SplashScreen : Backend is not under maintenance.
[2022-09-18 18:56:33.621]: DEBUG: LocationDataSource : Location found as : Latitude: -33.865913, Longitude: 151.2052124
[2022-09-18 18:56:33.621]: DEBUG: LocationDataSource : Overriding the actual location with static one, as the Browser / Linux does not allow mocking the location
[2022-09-18 18:56:33.622]: DEBUG: SplashScreen : latitude : -33.865107, longitude : 151.205282

thank you very much for writing this, it's really quite amazing, 
I see it as a 1/2 point resolve most if not everything before your app hits AGL proper

please advise best regards
Bern

jwinarske, 2022-09-20:
>     1. Multiple flutter installs
>        The original snap flutter install was conflicting  with new meta-flutter/tools/flutter install once i separated those setup_flutter_workspace.py ran correctly
> 

There are cases of conflict:
1.  Running script from terminal in VS Code which has Flutter tools enabled.  The Flutter tools has a file watcher running, and post repo clone will attempt to runs hooks using a mis-matched SDK version.  The workaround here is to close any IDE with Flutter tools, and run `setup_flutter_workspace.py` from a fresh terminal window.
2.  Multiple instances of `flutter` in the path.  The script will attempt to remove one instance from `PATH` if it can be resolved.  The new workspace SDK path is prefixed during runtime of script.  Running `source ./setup_env.sh` is the expected method for normal use.  In general you should limit the number of folders in your system path regardless.

If you are moving to using a Flutter Workspace, there is no reason to have the `flutter` snap installed.  The workspace construct replaces the need for other instances of the Flutter SDK.  So your best bet is to remove it.

>     2. invalid kvm args  launching qemu-system-x86_64
>        whilst debugging setup_env I realised i hadn't yet enabled kvm on my mac  so removed kvm  options
>        "-enable-kvm -cpu kvm64"  and qemu-system-x86_64 started fine.
>        I'll look at enabling Intel® VT) if necessary
> 

I would only recommend running a container with HW Hypervisor support.  While it may work with only software, there are performance penalties.

> 
> Once up i could navigate to meta-flutter/tools/app and launch gallery, flutter_homescreen, flutter_animated_background ( failed missing lib/main.dart), flutter_secure_storage (failed missing some things) and ped
> 

The example projects are not in the root of some of the projects.  The launch.json generated for VS Code have the correct pubspec path.  Switch to debugger view and select the drop down for desired target, then select the `play` icon.

> a) Qemu is grabbing and lockng my mouse any idea how to stop that ?
>

`Ctrl+Alt+G` should release the mouse grab.

>  b) How do i run using profile mode so as i can check Flutter DevTools as debug mode is kill it ?
>

Steps for running profile mode are here:
https://gerrit.automotivelinux.org/gerrit/gitweb?p=AGL/meta-agl-devel.git;a=blob;f=meta-agl-flutter/README.md;h=86b4f75b71fb070963ae2bbda52a1a3500003e87;hb=HEAD

>  c) Why only 5 apps where are the rest from /recipes-graphics/flutter-apps - i'd like to play around with all of them?
>

At the time the AGL flutter apps had not yet landed, and for some reason they ended in another layer.  Unclear why Scott did that.  Feel free to create your own workspace config JSON and add as needed.

> d) how often do i need to re-run setup_flutter_workspace and rebuild flutter-auto ?
>

Anytime you need/want to jump Flutter SDK versions, or pickup latest flutter-auto or AGL-QEMU image.  e.g. A few days ago I moved the meta-flutter AGL QEMU images to AGL master.  So official images with the addition of kuksa support.
 
> e) When launching PED it is spinning on "Detecting Locations" which is a bit of a worry, but it's also giving different feedback depending upon which run option i use Toyota or AGL. As follows [...]

If you are running a debugging QEMU image you get all the output.  Running a release image the debug spew is limited.
 
> Could it just be that I need location services enabled in qemu ?
> 

So if you see it work, two things:
1. Geoclue is included in the OS image
2. You're running a debug build.

When you don't see it work on a debug image it's missing Geoclue from the OS image.

When you don't see it work on a release image, it's because you aren't building for the Flutter 1P Linux embedding, which is what the plugin you linked to is intended for. I.e., it's the intended result. The fact that it "works" in debug is a bug tracked at https://github.com/flutter/flutter/issues/103660

So the underlying issue is that you're using a Flutter 1P Linux plugin.  The longer term solution here is to implement cross compiling in the Flutter SDK for Flutter Desktop.  At which point the AOT will have the correct registration code.

> launching ped using Toyota [1]: Toyota flutter-auto (desktop-auto) 1 display bunch of stuff .../... then [INFO:engine.cc(94)] (0) flutter: [2022-09-18 19:09:35.435717]: DEBUG: LocationDataSource : [...]

Yep, I suspect geoclue requires some additional config beyond the base recipe.  If that was present you will still be presented with it not working on AOT image.

> launch ped using Ubuntu linux-desktop (it just works) display bunch of stuff .../... then [2022-09-18 18:56:25.241]: DEBUG: LocationDataSource : Checking Location Service Enabled [2022-09-18 [...]

`linux-desktop` is what Flutter 1P "Linux" plugins are made for.  The naming is mis-leading.  They use the term "Linux" very loose.  It would not run on Android (Linux), or with a Custom Embedder (Linux).  It's just a logical name used in pubspec.yaml to select the Linux GTK platform (desktop).

Normally in a vehicle there is an API exposed to query the location.  Geoclue is just a "good-enough" stand-in for AGL.

> thank you very much for writing this, it's really quite amazing, I see it as a 1/2 point resolve most if not everything before your app hits AGL proper
>

You're welcome.