meta-flutter/meta-flutter #191: flutter-auto - backend-egl vs backend-vulkan

munoz0raul, 2022-09-16:
Hello, I'm building flutter-auto with gallery example and it is just running if I disable `DISTRO_FEATURES:remove = "vulkan "`.

Could someone comment what is the difference between run flutter with vulkan or egl, which one is better?

jwinarske, 2022-09-16:
If your SoC supports Vulkan you can save up to 48% on your CPU load.  Vulkan is much more power efficient.  Which also gives you more CPU bandwidth to do other things.  Think about it in relation to BOM cost.  Your software no longer consumes 48% of the SoC.  So you get more for less.

So unless there is an EGL specific feature you need, i you use Vulkan.

munoz0raul, 2022-09-16:
Well, I'm using the iMX8MM-evk, it should be fine with vulkan. Let me get the error and past it here.

jwinarske, 2022-09-16:
Run `vulkaninfo` from vulkan-tools recipe.  Shared that output.

jwinarske, 2022-09-16:
As a minimum for Vulkan support you want these two included to your image:
```
vulkan-loader vulkan-tools
```

munoz0raul, 2022-09-16:
On My distro, vulkan is not installed, I'm trying to understand what is better fo iMX8.

Looking to your build, it is also removed:

https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-imx8mmevk.yml#L77

jwinarske, 2022-09-16:
I have no idea what state the Vulkan stack on imx8mmevk is, given I don't have hardware to test on.  If someone wants to donate a board I'm happy to look into it.

jwinarske, 2022-09-16:
It may just be the two missing recipes I referenced above.  If not `vulkantools` will highlight the issue.

jwinarske, 2022-09-16:
While you're at it you might want to add `strace`, as that comes in handy troubleshooting vulkan loader path resolution.

munoz0raul, 2022-09-16:
I have done strace already:

```
root@imx8mm-lpddr4-evk:/var/rootdirs/home/fio# strace flutter-auto --b=/usr/share/flutter/gallery --w=1920 --h=1024
execve("/bin/flutter-auto", ["flutter-auto", "--b=/usr/share/flutter/gallery", "--w=1920", "--h=1024"], 0x7fda884538 /* 18 vars */) = 0
brk(NULL)                               = 0x55ac651000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=12606, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 12606, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fb1a13000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc++.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=964648, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb1a11000
mmap(NULL, 1122232, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb18cf000
mmap(0x7fb18d0000, 1056696, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb18d0000
munmap(0x7fb18cf000, 4096)              = 0
munmap(0x7fb19d2000, 61368)             = 0
mprotect(0x7fb19b6000, 61440, PROT_NONE) = 0
mmap(0x7fb19c5000, 28672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe5000) = 0x7fb19c5000
mmap(0x7fb19cc000, 24504, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb19cc000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc++abi.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=255928, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 385784, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb1871000
mmap(0x7fb1880000, 320248, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb1880000
munmap(0x7fb1871000, 61440)             = 0
munmap(0x7fb18cf000, 760)               = 0
mprotect(0x7fb18ba000, 65536, PROT_NONE) = 0
mmap(0x7fb18ca000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3a000) = 0x7fb18ca000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0@\264\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\263RJ\0\301\210{\343f\263\32\flZI$"..., 68, 768) = 68
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1630088, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 1805320, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb16c7000
mmap(0x7fb16d0000, 1739784, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb16d0000
munmap(0x7fb16c7000, 36864)             = 0
munmap(0x7fb1879000, 27656)             = 0
mprotect(0x7fb1858000, 61440, PROT_NONE) = 0
mmap(0x7fb1867000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x187000) = 0x7fb1867000
mmap(0x7fb186d000, 48136, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb186d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libwayland-client.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=68264, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 197392, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb169f000
mmap(0x7fb16a0000, 131856, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb16a0000
munmap(0x7fb169f000, 4096)              = 0
munmap(0x7fb16c1000, 58128)             = 0
mprotect(0x7fb16ae000, 65536, PROT_NONE) = 0
mmap(0x7fb16be000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe000) = 0x7fb16be000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libwayland-cursor.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=34808, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 164016, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb1677000
mmap(0x7fb1680000, 98480, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb1680000
munmap(0x7fb1677000, 36864)             = 0
munmap(0x7fb1699000, 24752)             = 0
mprotect(0x7fb1684000, 65536, PROT_NONE) = 0
mmap(0x7fb1694000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x7fb1694000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libxkbcommon.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=280560, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 409616, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb161b000
mmap(0x7fb1620000, 344080, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0) = 0x7fb1620000
[... 334 lines cut ...]
close(6)                                = 0
openat(AT_FDCWD, "/usr/lib/tls/aarch64/libvulkan.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/tls/aarch64", 0x7fdf9193c0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/libvulkan.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
newfstatat(AT_FDCWD, "/usr/lib", {st_mode=S_IFDIR|0755, st_size=20480, ...}, 0) = 0
munmap(0x7fb19d7000, 12606)             = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xffffffffffffffe8} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

And I'm sure vulkan are not well installed:

```
 vulkaninfo
ERROR at /usr/src/debug/vulkan-tools/1.3.204.1-r0/git/vulkaninfo/vulkaninfo.h:612: Failed to initialize: Vulkan loader is not installed, not found, or failed to load.
```

I will check with my team as well, if they know something about it.

Thanks.

jwinarske, 2022-09-16:
In that run from above log was `vulkan-loader` present in the image?

munoz0raul, 2022-09-21:
There is no Vulkan support for 8M Mini.
https://github.com/Freescale/meta-freescale/issues/1216

closed 2022-09-21 by munoz0raul