meta-flutter/meta-flutter #204: imx6sx runtime error - Illegal instruction

camaronut, 2022-10-10:
Hi there,

First off, thank you for creating this framework!  It has helped us to get flutter up on our imx6sx target very quickly.  We are currently doing an R&D evaluation comparing QT and Flutter.  We are using the dunfell branch of meta-flutter.  We need to do this evaluation in dunfell for "apples-to-apples" comparisons to our already established QT environment.

Our target board is an imx6sx chip (cortex a9 32 bit).  We created an image with flutter-wayland-client and flutter-gallery on the SD card.  We start up the board, bring up weston, and execute:
```
flutter-client --bundle=/usr/share/flutter/gallery/lib -f
Illegal instruction
```
We are able to work around the problem with the following diff:
```
diff --git a/classes/flutter-app.bbclass b/classes/flutter-app.bbclass
index da7fc7e..757de1b 100644
--- a/classes/flutter-app.bbclass
+++ b/classes/flutter-app.bbclass
@@ -228,6 +228,7 @@ do_compile() {
         #
         ${S}/engine_sdk/sdk/clang_x64/gen_snapshot \
             --deterministic \
+            --no-use-integer-division \
             --snapshot_kind=app-aot-elf \
             --elf=libapp.so \
             --strip \
```

However, this seems like a sledgehammer approach.  Is there a better way to fix this in a way that bitbake can use ARCH or some other variable to intelligently select this for 32 bit ARM parts?  That said, I'm not sure it even applies to all 32 bit ARM parts.

Thanks!
Chris

jwinarske, 2022-10-10:
@camaronut You're welcome.  Yeah that's rough.

What floating point hardware support does the imx6sx part have?

Are you using default package config options?  Which would be Dart SDK prebuilt.  I wonder if trying no pre-built dart SDK has an effect.

jwinarske, 2022-10-11:
For building an AOT you want `flutter-engine-runtimerelease-sdk-dev` installed in your SDK.  It installs to TARGET ${datadir}/flutter/engine_sdk.zip which has the required host side tool to build an AOT.

camaronut, 2022-10-12:
Hi Joel,

Adding ``flutter-engine-runtimerelease-sdk-dev`` to the SDK works!  Thanks for that nugget!

I still need to experiment with the non-pre-built dart SDK.
Thanks,
Chris

camaronut, 2022-10-12:
I created a quick .bbappend with this line:
PACKAGECONFIG_remove = " prebuilt-dart-sdk"

I then reverted my local change to flutter-app.bbclass that provided the "--no-use-integer-division" flag.  I then re-built the flutter apps using bitbake.  Same result when running: Illegal instruction.

Let me know if you'd like me to try something else, thanks!
Chris

jwinarske, 2022-10-12:
So ultimately this resolves around passing the proper compiler tuning for the Dart SDK.  Two things you would want to know:
1.  Yocto machine tune parameters passed to engine recipe.  Assignment happens here with python function to derive:
  https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-engine/flutter-engine_git.bb#L113-L116
2. what does the Dart SDK support for tuning parameters specific to your floating point hardware
If you sort that out, then you could make a bbappend to override the `arm-tune` parameter.  Even then it may require a patch to the engine to get it passed correctly to the Dart SDK build.  Definitely you want to keep your recent change which would pass `--no-prebuilt-dart-sdk`.

martinburkot, 2022-11-22:
Hello,
I would like to ask if there is any development on this issue? I've run int the same issue and I would like to use the dunfell branch from the repo without having to manually add --no-use-integer-division into the flutter-app.bbclass. 
I am new to yocto but if I understand it correctly it is not a good practice to override the .bbclass files.
Thank you, Martin.

camaronut, 2022-11-22:
Hi Martin, Joel,

The good news is that our flutter evaluation went flawlessly, so much so
that the next project will be ditching QT and using Flutter.  The bad news
is that after the evaluation I was put on a project revolving around parts
sourcing issues.

I'd expect I'll be back on the Flutter in Yocto side of things at the end
of December or early January.

Thanks!
Chris

jwinarske, 2023-03-04:
Revisiting this.  Did you resolve this?  Have you tried building/running jit runtime image?  I ran across something about this being required for 32-bit, only I've seen 32-bit AOT work fine on STM.

camaronut, 2023-03-06:
Hi Joel,

I haven't had bandwidth to dig into this.  Our workaround for now has been something like this:

`diff --git a/classes/flutter-app.bbclass b/classes/flutter-app.bbclass
index da7fc7e..0dd8061 100644
--- a/classes/flutter-app.bbclass
+++ b/classes/flutter-app.bbclass
@@ -172,6 +172,7 @@ do_compile() {
     export PATH=${FLUTTER_SDK}/bin:$PATH
     export PUB_CACHE=${PUB_CACHE}
     export PKG_CONFIG_PATH=${STAGING_DIR_TARGET}/usr/lib/pkgconfig:${STAGING_DIR_TARGET}/usr/share/pkgconfig:${PKG_CONFIG_PATH}
+    local INTEGER_DIV
 
     bbnote `env`
 
@@ -226,8 +227,10 @@ do_compile() {
         #
         # Create libapp.so
         #
+        INTEGER_DIV=${@'--no-use-integer-division' if 'mx6' in d.getVar('MACHINEOVERRIDES').split(":") else ''}
         ${S}/engine_sdk/sdk/clang_x64/gen_snapshot \
             --deterministic \
+            $INTEGER_DIV \
             --snapshot_kind=app-aot-elf \
             --elf=libapp.so \
             --strip \
`
My time is currently digging into a host of build issues when releasing, one of which relates to the gn tarball.  I won't cloud up the waters of this issue with that issue.  :)

jwinarske, 2023-03-07:
@camaronut we would need to detect soft floating support.  Something in the machine config could be used.

What's the value of `TARGET_FPU`?  It's not being set to `softfp` is it?

Also at one point dunfell poky pulled in support for the colon override specifier.  Which enables the use of the kirkstone branch.  That might be an option.

At some point I need to update dunfell to match kirkstone updates.  Lots of time to test.

camaronut, 2023-03-07:
@jwinarske, here's a bit from a bitbake -e command:
```
# $TARGET_FPU [2 operations]
#   set /hd/dunfell/sources/poky/meta/conf/machine/include/arm/arch-arm.inc:16
#     "${@d.getVar('TUNE_CCARGS_MFLOAT') or 'soft'}"
#   set /hd/dunfell/sources/poky/meta/conf/documentation.conf:417
#     [doc] "Specifies the method for handling FPU code. For FPU-less targets, which include most ARM CPUs, the variable must be set to 'soft'. If not, the kernel emulation gets used, which results in a performance penalty."
# pre-expansion value:
#   "${@d.getVar('TUNE_CCARGS_MFLOAT') or 'soft'}"
TARGET_FPU="hard"
```
Some other variables bitbake dumped out that might be interesting:

- ARMPKGARCH="cortexa9"
- TUNE_CCARGS_MFLOAT="hard"
- TUNE_CCARGS_MFPU=" vfp neon"
- TUNE_CCARGS=" -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9"
- TUNE_FEATURES="arm vfp cortexa9 neon thumb callconvention-hard"

I hope I'm down to the last onion layer on the gn tarball reproducibility, when I get a matching tar file I'll submit an issue along with my workaround patch.

camaronut, 2023-03-16:
We had an observation this week regarding running gen_snapshot.
Our application build script runs this to create the package (outside of Yocto) for our target:
```
# extract sdk
rm -rf ./engine_sdk
unzip ${SDKTARGETSYSROOT}/usr/share/flutter/engine_sdk.zip -d ./engine_sdk

# Create libapp.so
${OECORE_NATIVE_SYSROOT}/lib/ld-linux-x86-64.so.2 ./engine_sdk/sdk/clang_x64/gen_snapshot \
    --deterministic \
    --no-use-integer-division \
    --snapshot_kind=app-aot-elf \
    --elf=libapp.so \
    --strip \
    .dart_tool/flutter_build/*/app.dill
```
The bit with "${OECORE_NATIVE_SYSROOT}/lib/ld-linux-x86-64.so.2" is new... The explanation: we found when we installed the toolchain on an Ubuntu 18.04 system/container and performed a build there it complained about glibc versions:
```
./engine_sdk/sdk/clang_x64/gen_snapshot: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ./engine_sdk/sdk/clang_x64/gen_snapshot)
./engine_sdk/sdk/clang_x64/gen_snapshot: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by ./engine_sdk/sdk/clang_x64/gen_snapshot)
```
It appears that gen_snapshot is reaching out to the host and using host libraries, rather than staying in the "bounds" of the toolchain.
Maybe that's a piece of the puzzle?

As an aside, it might be nice to "build in" the contents of the zip file direct in the toolchain, so we don't need to unzip the engine sdk in our build scripts.  My list of items to do is getting long (we're learning a lot with flutter), so I won't be able to submit a patch soon for this.

jwinarske, 2023-03-16:
@camaronut that makes sense, and I think it's a valid concern.   Ideally the SDK binary loads at runtime the same libraries it was linked against.  I'll take a look with ldd and look into packaging the dependent libraries.

jwinarske, 2023-03-17:
@camaronut I've added a variable for you in kirkstone.  APP_GEN_SNAPSHOT_FLAGS.
To use this you would override in your flutter app recipe:

    APP_GEN_SNAPSHOT_FLAGS="--no-use-integer-division"

https://github.com/meta-flutter/meta-flutter/commit/949a5637c554f0637649b25872819f4de89b1c92

closed 2023-03-17 by jwinarske