meta-flutter/meta-flutter #205: gallery: issue with font loading

camaronut, 2022-10-10:
Hi there,

This issue follows "on the heels" of https://github.com/meta-flutter/meta-flutter/issues/204.

When attempting to run the gallery app, we get errors regarding font loading:
```
flutter-client --bundle=/usr/share/flutter/gallery/lib -f
flutter: Error loading AssetManifest.json, e: Unable to load asset: AssetManifest.json
flutter: Error: google_fonts was unable to load font Montserrat-Bold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Bold was not found in the application assets. Ensure Montserrat-Bold.otf exists in a folder that is included in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Oswald-SemiBold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Oswald-SemiBold was not found in the application assets. Ensure Oswald-SemiBold.otf exists in a folder that is included in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Oswald-Medium because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Oswald-Medium was not found in the application assets. Ensure Oswald-Medium.otf exists in a folder that is included in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-Medium because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Medium was not found in the application assets. Ensure Montserrat-Medium.otf exists in a folder that is included in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-Regular because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-Regular was not found in the application assets. Ensure Montserrat-Regular.otf exists in a folder that is included in your pubspec's assets.
flutter: Error: google_fonts was unable to load font Montserrat-SemiBold because the following exception occured:
Exception: GoogleFonts.config.allowRuntimeFetching is false but font Montserrat-SemiBold was not found in the application assets. Ensure Montserrat-SemiBold.otf exists in a folder that is included in your pubspec's assets.
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Unable to load asset: packages/flutter_localized_locales/data/en.json
#0      PlatformAssetBundle.load (package:flutter/src/services/asset_bundle.dart:258)
<asynchronous suspension>
#1      AssetBundle.loadString (package:flutter/src/services/asset_bundle.dart:83)
<asynchronous suspension>
```
The app appears to continue to run, however nothing changes on the screen (still shows the weston background).

I checked /usr/share/flutter/gallery on the target and there's plenty of "*.ttf" files, including montserrat, oswald, etc.  There are no "*.otf" files though:
```
root@dsm-devel:/usr/share/flutter/gallery# find . -name "Oswald-SemiBold*"
./data/flutter_assets/packages/flutter_gallery_assets/fonts/google_fonts/Oswald-SemiBold.ttf
root@dsm-devel:/usr/share/flutter/gallery# find . -name "Oswald-Medium*"
./data/flutter_assets/packages/flutter_gallery_assets/fonts/google_fonts/Oswald-Medium.ttf
root@dsm-devel:/usr/share/flutter/gallery# find . -name "Montserrat-Medium*"
./data/flutter_assets/packages/flutter_gallery_assets/fonts/google_fonts/Montserrat-Medium.ttf
root@dsm-devel:/usr/share/flutter/gallery# find . -name "Montserrat-SemiBold*"
./data/flutter_assets/packages/flutter_gallery_assets/fonts/google_fonts/Montserrat-SemiBold.ttf
```
I'm unfamiliar with the flutter build process, so apologies in advance if this issue is in the wrong bucket.  Let me know if there's anything I need to change on my end to make this work right.

We are able to get the flutter-test-animated-background application to work on target.  It displays the animated bubbles very smoothly, we're pretty happy about that.  :)

Thanks!
Chris

jwinarske, 2022-10-10:
What does the bundle folder layout look like?  As-is generated by meta-flutter `dunfell` or manually created?

camaronut, 2022-10-12:
Hi Joel,

Here's a look at the usr/share/flutter/gallery folder in the rootfs.  It is as-is generated by meta-flutter/yocto, no modifications to the gallery recipe.
```
cfester@buildhost:/hd/dunfell/build/tmp/work/dsm_devel-poky-linux-gnueabi/core-image-foo-devel/1.0-r0/rootfs/usr/share/flutter/gallery$ tree .
.
├── data
│   └── flutter_assets
│       ├── AssetManifest.json
│       ├── FontManifest.json
│       ├── fonts
│       │   └── MaterialIcons-Regular.otf
│       ├── isolate_snapshot_data
│       ├── kernel_blob.bin
│       ├── NOTICES.Z
│       ├── packages
│       │   ├── cupertino_icons
│       │   │   └── assets
│       │   │       └── CupertinoIcons.ttf
│       │   ├── flutter_gallery_assets
│       │   │   ├── assets
│       │   │   │   ├── demos
│       │   │   │   │   ├── 1.5x
│       │   │   │   │   │   └── bottom_navigation_background.png
│       │   │   │   │   ├── 2.0x
│       │   │   │   │   │   └── bottom_navigation_background.png
│       │   │   │   │   ├── 3.0x
│       │   │   │   │   │   └── bottom_navigation_background.png
│       │   │   │   │   ├── 4.0x
│       │   │   │   │   │   └── bottom_navigation_background.png
│       │   │   │   │   └── bottom_navigation_background.png
│       │   │   │   ├── icons
│       │   │   │   │   ├── cupertino
│       │   │   │   │   │   ├── 1.5x
│       │   │   │   │   │   │   └── cupertino.png
│       │   │   │   │   │   ├── 2.0x
│       │   │   │   │   │   │   └── cupertino.png
│       │   │   │   │   │   ├── 3.0x
│       │   │   │   │   │   │   └── cupertino.png
│       │   │   │   │   │   ├── 4.0x
│       │   │   │   │   │   │   └── cupertino.png
│       │   │   │   │   │   └── cupertino.png
│       │   │   │   │   ├── material
│       │   │   │   │   │   ├── 1.5x
│       │   │   │   │   │   │   └── material.png
│       │   │   │   │   │   ├── 2.0x
│       │   │   │   │   │   │   └── material.png
│       │   │   │   │   │   ├── 3.0x
│       │   │   │   │   │   │   └── material.png
│       │   │   │   │   │   ├── 4.0x
│       │   │   │   │   │   │   └── material.png
│       │   │   │   │   │   └── material.png
│       │   │   │   │   └── reference
│       │   │   │   │       ├── 1.5x
│       │   │   │   │       │   └── reference.png
│       │   │   │   │       ├── 2.0x
│       │   │   │   │       │   └── reference.png
│       │   │   │   │       ├── 3.0x
│       │   │   │   │       │   └── reference.png
│       │   │   │   │       ├── 4.0x
│       │   │   │   │       │   └── reference.png
│       │   │   │   │       └── reference.png
│       │   │   │   ├── logo
│       │   │   │   │   ├── 1.5x
│       │   │   │   │   │   ├── flutter_logo_color.png
│       │   │   │   │   │   └── flutter_logo.png
│       │   │   │   │   ├── 2.0x
│       │   │   │   │   │   ├── flutter_logo_color.png
│       │   │   │   │   │   └── flutter_logo.png
│       │   │   │   │   ├── 3.0x
[... 1026 lines cut ...]
│       │       ├── 31-0.jpg
│       │       ├── 32-0.jpg
│       │       ├── 33-0.jpg
│       │       ├── 34-0.jpg
│       │       ├── 35-0.jpg
│       │       ├── 36-0.jpg
│       │       ├── 37-0.jpg
│       │       ├── 4-0.jpg
│       │       ├── 5-0.jpg
│       │       ├── 6-0.jpg
│       │       ├── 7-0.jpg
│       │       ├── 8-0.jpg
│       │       ├── 9-0.jpg
│       │       ├── diamond.png
│       │       └── slanted_menu.png
│       ├── shaders
│       │   └── ink_sparkle.frag
│       └── vm_snapshot_data
└── lib
    └── libapp.so

74 directories, 1036 files
```
Thanks!
Chris

jwinarske, 2022-10-12:
What branch are you on?

camaronut, 2022-10-12:
dunfell

jwinarske, 2022-10-12:
The bundle format is correct.  Does your build have a default linguas/locale setup?  I haven't smoke tested a dunfell build in a while, and will look into doing this in the next day or so.

camaronut, 2022-10-13:
Hi Joel,

We set IMAGE_LINGUAS = " " in our image recipe.  I don't see any mention of GLIBC_GENERATE_LOCALES in our layer either.  Would it help if we did set it to something meaningful?
Thanks!
Chris

jwinarske, 2022-10-13:
Hi Chris,

Of you're trying to run the gallery app as-is, it's know to work in all the CI images.  So I would rule out the image has an installed language, timezone, and locale set.  

See this CI for a minimum example:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-imx8mmevk.yml#L74

Also running `localectl` on target should return values that match what are available on the target image.

martinburkot, 2022-11-28:
Hello, 
please see the issue #222 as it seems to have the same root cause as this one.
BR Martin

camaronut, 2023-01-31:
@martinburkot , thank you for the pointer on this.  @jwinarske , thank you for providing flutter-auto!

I removed flutter-wayland-client from our image, and added flutter-auto.  I then launched weston and invoked the app with: 
`flutter-auto --b=/usr/share/flutter/animated_background_example --f`
That ran without any exceptions and successfully launched the gallery app.  I also tested with the animated background demo and that also works well.
@jwinarske , maybe the Sony recipes should be removed from the layer?

closed 2023-01-31 by camaronut

jwinarske, 2023-01-31:
@camaronut I'm trying to be neutral in the layer.  But at some point I may remove them   If you want to provide me some warning words to include in the readme, I will consider adding them.