meta-flutter/meta-flutter #206: Unable to fetch URL from any source.', 'gn://github.com/flutter/engine.git;name=src/flutter'

gencay3535, 2022-10-11:
Hello,

I am trying to build a imx-image-core with i.MX8MP but I got this error for the flutter-engine-runtimerelease. 
What kind of fetcher is **gn** at the recipe flutter-engine_git.bb? 
Am I missing something other than configuration files from the workflow for i.MX8MM?

Log file is attached. Thank you in advance.

[mylog.LOG](https://github.com/meta-flutter/meta-flutter/files/9756119/mylog.LOG)

jwinarske, 2022-10-11:
fetcher `gn` caches a gclient enlistment to the DL_DIR using `depot-tools`.

What branch/commit?
Are you overriding `FLUTTER_SDK_TAG`?

I am not hitting this error in `kirkstone` branch CI.

gencay3535, 2022-10-13:
Hello, Thank you for your response.

My reference was workflow at Kirkstone Branch.
I am not overriding `FLUTTER_SDK_TAG`
I also have this kind of warning I dont know if it matters.

`WARNING: depot-tools-native-git-r0 do_populate_sysroot: File '/home/gencay/NXP_workspace/build/tmp/work/x86_64-linux/depot-tools-native/git-r0/recipe-sysroot-native/usr/share/depot_tools/ninja-linux64' from depot-tools-native was already stripped, this will prevent future debugging!
WARNING: depot-tools-native-git-r0 do_populate_sysroot: File '/home/gencay/NXP_workspace/build/tmp/work/x86_64-linux/depot-tools-native/git-r0/recipe-sysroot-native/usr/share/depot_tools/ninja-linux32' from depot-tools-native was already stripped, this will prevent future debugging!`

jwinarske, 2022-10-13:
You can ignore the warning.

To get unblocked try setting `FLUTTER_SDK_TAG` to 3.3.3 in your local.conf.  There's a upstream issue, but it's unclear why my CI isn't hitting it.  It will go away with roll to 3.3.5.

gencay3535, 2022-10-19:
Hello, thank you for your response.

I was able to built an i.mx6 image, the error for the flutter-engine resolved with only building recipe: `bitbake flutter-engine_runtimerelease` However some other recipes, I couldn't built due to do_unpack error so I removed some of them. Here is my current local.conf:

`MACHINE ??= 'imx6sxsabresd'
DISTRO ?= 'fsl-imx-wayland'
PACKAGE_CLASSES ?= 'package_rpm'
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    HALT,${TMPDIR},100M,1K \
    HALT,${DL_DIR},100M,1K \
    HALT,${SSTATE_DIR},100M,1K \
    HALT,/tmp,10M,1K"
PACKAGECONFIG:append:pn-qemu-system-native = " sdl"
CONF_VERSION = "2"
SSTATE_DIR="/hdd/imx6/sstate-cache/"
DL_DIR ?= "/hdd/downloads/"
ACCEPT_FSL_EULA = "1"
# Switch to Debian packaging and include package-management in the image
PACKAGE_CLASSES = "package_deb"
EXTRA_IMAGE_FEATURES += "package-management"
INHERIT += "rm_work"
GLIBC_GENERATE_LOCALES:append = " en_US.UTF-8 es_US.UTF-8 en_GB.UTF-8"
IMAGE_LINGUAS:append = " en-us en-gb es-us"
DEFAULT_TIMEZONE = "America/Los_Angeles"
DISTRO_FEATURES:remove = "vulkan"
IMAGE_INSTALL:append = "\
    tzdata-core \
    tzdata-americas \
    flutter-auto-runtimerelease \
    flutter-gallery-runtimerelease \
    flutter-test-video-player-runtimerelease \
    flutter-test-frb-runtimerelease \
    "
LICENSE_FLAGS_ACCEPTED += " commercial"
`
 When I try to test gallery on the board at user weston, I get:

`imx6sxsabresd:~$ flutter-auto --w=800 --h=480 --b=/usr/share/flutter/gallery
[INFO:configuration.cc(248)] **********
[INFO:configuration.cc(249)] * Global *
[INFO:configuration.cc(250)] **********
[INFO:configuration.cc(251)] Application Id: .......... flutter-auto
[INFO:configuration.cc(255)] Cursor Theme: ............ 
[INFO:configuration.cc(256)] Disable Cursor: .......... false
[INFO:configuration.cc(258)] Debug Backend: ........... false
[INFO:configuration.cc(260)] ********
[INFO:configuration.cc(261)] * View *
[INFO:configuration.cc(262)] ********
[INFO:configuration.cc(269)] Bundle Path: .............. /usr/share/flutter/gallery
[INFO:configuration.cc(270)] Window Type: .............. NORMAL
[INFO:configuration.cc(271)] Size: ..................... 800 x 480
[INFO:configuration.cc(273)] Fullscreen: ............... false
[INFO:configuration.cc(275)] Accessibility Features: ... 0
[ERROR:/usr/src/debug/flutter-auto-runtimerelease/git-r0/git/shell/wayland/display.cc(51)] Failed to connect to Wayland display. No such file or directory
'
When I try it at root:
`root@imx6sxsabresd:~# flutter-auto --w=800 --h=480 --b=/usr/share/flutter/gallery
[INFO:configuration.cc(248)] **********
[INFO:configuration.cc(249)] * Global *
[INFO:configuration.cc(250)] **********
[INFO:configuration.cc(251)] Application Id: .......... flutter-auto
[INFO:configuration.cc(255)] Cursor Theme: ............ 
[INFO:configuration.cc(256)] Disable Cursor: .......... false
[INFO:configuration.cc(258)] Debug Backend: ........... false
[INFO:configuration.cc(260)] ********
[INFO:configuration.cc(261)] * View *
[INFO:configuration.cc(262)] ********
[INFO:configuration.cc(269)] Bundle Path: .............. /usr/share/flutter/gallery
[INFO:configuration.cc(270)] Window Type: .............. NORMAL
[INFO:configuration.cc(271)] Size: ..................... 800 x 480
[INFO:configuration.cc(273)] Fullscreen: ............... false
[INFO:configuration.cc(275)] Accessibility Features: ... 0
[INFO:display.cc(275)] Keyboard Present
[INFO:display.cc(284)] Touch Present
[INFO:engine.cc(567)] (0) Loading AOT: /usr/share/flutter/gallery/lib/libapp.so
Illegal instruction
`
What can be the cause of this issue?

jwinarske, 2022-10-20:
@gencay3535 You're now hitting https://github.com/meta-flutter/meta-flutter/issues/204

He offers a short term work around.

jwinarske, 2022-10-20:
Ultimately to solve the illegal instruction with vfp+neon enabled requires some digging.  I'm suspecting Dart SDK build either with parameter passing or patching source tree.  A complete stack trace would be helpful, which requires a hand built engine debug build; non-stripped with symbols.

jwinarske, 2022-10-20:
@gencay3535 Weston should be setup to run under a user session.  Given the issue no longer matches the title of the bug I am closing this issue.  Feel free to open an issue if there is an issue with meta-flutter.

closed 2022-10-20 by jwinarske