meta-flutter/meta-flutter #208: gallery app build error after migration from dunfell to kirkstone

ratara, 2022-10-24:
Hi,
I migrated my build from dunfell to kirkstone and everything is back working now besides the gallery app.

I am using the latest commit of the meta-flutter repository for the kirkstone branch and got this error message. Since today I got no trouble with building the gallery-app.

Do you have any ideas what I am missing here?

[image]

jwinarske, 2022-10-24:
Did you try this?
```
bitbake flutter-gallery-runtimerelease -c do_cleansstate
bitbake flutter-gallery-runtimerelease
```

ratara, 2022-10-25:
@jwinarske 
I'm feeling a little goofy now but this worked.
Apparently, clean and cleanall was not enough. Learned something again. Thanks very much!

jwinarske, 2022-10-25:
@ratara NP