meta-flutter/meta-flutter #220: More Apps with Workspace Automation

bernardcraddock, 2022-11-24:
Hi @jwinarske ,

Just a couple of things if i may

1. workspace automation's flutter_workspace_config.json  we seem to be very limited on what apps are being included
Therefore would it be possible to extend this and add a few more, and i was specifically thinking the flutter-test apps in 
recipes-graphics.

ped-vm /home/parallels/Documents/GitHub/meta-flutter /recipes-graphics/flutter-apps/
  flutter-test-secure-storage_git.bb
  flutter-test-frb_git.bb
  flutter-test-plugins_git.bb
  flutter-test-localization_git.bb
  flutter-test-membrane_git.bb
  flutter-test-video-player_git.bb
  flutter-test-texture-egl_git.bb

2. 2023 Q1, I'd like to start, exploring the feasibility of integrating PED with Kuksa, and was wondering is there any possibility you would consider extending meta-flutter and workspace automation  to include "kuksa val client and server"

please advise, and thanks for you time 
regards

jwinarske, 2022-11-24:
Sure I'll add all the test apps.

I have kuksa present in the QEMU x86_64 and arm64 builds.  They run the AGL kuksa enabled apps.

bernardcraddock, 2022-11-25:
@jwinarske  thank you  appreciated

Concerning kuksa in QEMU x86_64 and arm64 builds, that sounds really interesting and i  think i can benefit from it but not sure how! . Can we bring the GSoc flutter kuksa apps across to workspace automation so as i can run and test as that will be perfect provide me a big boost to my kusa learning curve.

If true and you're happy to do so, do you need me to create an AGL JIRA ticket ot something for all this work ?  
 i'll reach out and notify each of  GSoc students explain what/why.

If not true :- Then how can i benefit

Ok i'll leave it there , best regards have a great weekend

jwinarske, 2022-11-25:
If you look at the change lists on AGL you may find the required config/setup.  The config for each app gets written to /etc/xdg/AGL IIRC.  Sensitive data should really only be accessed through the secret plugin...

jwinarske, 2022-11-30:
@bernardcraddock I'm waiting for the CES dust to settle, then I will have a few consolidations.

bernardcraddock, 2022-12-01:
@jwinarske - Yep no worries got heaps to keep me busy until then.

Also not pushing, just politely requesting in advance as your time permits in addition to the above anything related to maps/navigation, hands-free phone connections, vehicle voice commands, and V2C. 

In regards to V2C i've joined AGL's V2C & Containers n Mesh EGs, to get a better understanding of where these EG's are coming from as i'm pretty sure Pumped backend (Merchant) could be a very good use-case for Amazon. Mentioned this to Hayden he agreed worth investigating but they're not interested if it's  just me they need a tier 1 automotive to adopt ped.

Anyway for all this new functionality we need to drop out of dart via FFI  and integrate with c++ , rust etc which is not our expertise and need a lot of help with. so maybe this is something we can collaborate on and perhaps do a joint ped / joel showing @CES 2024

Don't decide now  just have a think and any questions let me know here or bernard@pumpedfuel.com

regards

bernardcraddock, 2023-01-21:
Hey @jwinarske, 
i spoke with the AGL boys last Thursday and they mentioned you're going to include GTK in the embedder, this is excellent news thank you very much this now provides me a boost in confidence to continue on, as i'm sure many others will follow  

I also imagine this will entail some work on your behalf, including upstream  input, so if there's anything i can do for you downstream to assist just let me know.  Lastly of course fully appreciate your point refer pub dev plugin code quality. if we encounter any issues with those we use, we'll raise an issue with the Author.

again thank you
and regards

jwinarske, 2023-01-29:
@bernardcraddock it's not an official plan of record just yet.  It also depends on the willingness of upstream to accommodate changes.  Personally I am**not** a fan of glib.

bernardcraddock, 2023-01-30:
@jwinarske I understand it's not yet official, plenty of t's to cross and i's to dot , but it sounds like a very interesting  project that potentially could really open thing up for flutter devs. 

I also heard a rumour that the Flutter team is already working to make the custom GTK more modular and less tightly coupled from the rest of the engine, so as it can be used more easily as a standalone library by other projects.

Other AGL members are pushing for pipewire and gstreamer to be added this year, which down the line could also be complimentary for GTK apps.

 i know there are several glib alternatives , but  i wouldn't know the subtle differences can you tell me why you're not a fan, and what do you like and i'll do some research just for my own benefit

jwinarske, 2023-03-04:
@bernardcraddock I would emphasize the workspace configuration on meta-flutter is specific to meta-flutter.  The AGL configuration on AGL git is specific to AGL.  The point is not create a one size fits all, but provide the tool to enable diverse work environments that follow a well worn pattern.  The script is going through some changes, and will be relocated to the workspace automation repo soon.  It will no longer be flutter specific.  Becoming a genetic script to setup a dev environment.  At some point I will package and publish it.

So closing for now since one can create and are encouraged to create their own workspace environment (repos), etc

closed 2023-03-04 by jwinarske

bernardcraddock, 2023-03-05:
@jwinarske - ok thanks for the update, I'll transition to the upcoming workspace automation repo when available, and is GTK still on the roadmap for the flutter embedded ?

jwinarske, 2023-03-05:
@bernardcraddock it's complicated.  There will be some effort in that direction, but no guarantees.

bernardcraddock, 2023-03-05:
@jwinarske I appreciate that, 
If I can muster up other much larger and more prominent parties interested in getting this done will that raise the priority for you and if not what can/will?