meta-flutter/meta-flutter #221: AttributeError: 'BBLogger' object has no attribute 'debug2'

ardera, 2022-11-26:
I'm running into this error with latest dunfell meta-flutter:
```
- snip -
File: '/home/hannes/devel/tisdk/sources/meta-flutter/lib/gn.py', lineno: 82, function: _rungnclient
     0078:    def _rungnclient(self, ud, d, quiet):
     0079:        bb.utils.mkdirhier(ud.syncpath)
     0080:        os.chdir(ud.syncpath)
     0081:
 *** 0082:        logger.debug2("Fetching %s using command '%s'" % (ud.url, ud.basecmd))
     0083:        bb.fetch2.check_network_access(d, ud.basecmd, ud.url)
     0084:        runfetchcmd(ud.basecmd, d, quiet, workdir=None)
     0085:        logger.debug2("Packing %s using command '%s'" % (ud.url, ud.packcmd))
     0086:        runfetchcmd(ud.packcmd, d, quiet, workdir=None)
Exception: AttributeError: 'BBLogger' object has no attribute 'debug2'

DEBUG: Python function base_do_fetch finished
DEBUG: Python function do_fetch finished
```

Seems to be caused by commit 18581e6f392bf8de3a605c4a8bb3891aee5df8db. (It changed `logger.debug(2, ...)` to `logger.debug2` in the gn.py)

jwinarske, 2022-11-26:
I believe there's a fix for that in kirkstone.  Allows running with debug flag.

jwinarske, 2022-11-29:
@ardera See if reverting this on `dunfell` helps:
https://github.com/meta-flutter/meta-flutter/commit/80728ca8206ae8f68728880257581e43a1ff4582

WAKazuyoshiAkiyama, 2022-11-30:
@jwinarske We faced on the same error.
Replacing `debug2` with `debug`, we could solve the error. 
So, reverting https://github.com/meta-flutter/meta-flutter/commit/80728ca8206ae8f68728880257581e43a1ff4582 is required for `dunfell` branch.

WAKazuyoshiAkiyama, 2022-11-30:
https://git.yoctoproject.org/poky/tree/bitbake/lib/bb/__init__.py?h=dunfell#n84
In dunfell, `debug2` is not implemented.

https://git.yoctoproject.org/poky/tree/bitbake/lib/bb/__init__.py?h=kirkstone#n50
In kirkstone, `debug2` is implemented.

jwinarske, 2022-11-30:
@WAKazuyoshiAkiyama thanks for the confirmation.

jwinarske, 2022-11-30:
0ca204fd23e4c15fe9fed189d21cbb6af0895241 fixes this.

closed 2022-11-30 by jwinarske