meta-flutter/meta-flutter #222: Unable to load asset: AssetManifest.json

martinburkot, 2022-11-28:
Hello,
I am using a dunfell branch and it seems that the flutter apps are unable to load its assets. I've created simple flutter app containing just one picture and I try to run it, but I always get the "Unable to load asset" exception. I think that the flutter is unable to see its assets, but I don't know why. When I try to run the "flutter_animated_background" app from the "flutter-apps" recipes and use the "Image" instead of shape is also not working. This issue may have the same root cause as the issue #205 that is also occurring on my build. Please let me know if there's anything else I can do to help you solve the issue.

Please see the rest of the issue for more details. 
```
/usr/bin/flutter-client -f -b /usr/share/flutter/img_example/lib
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Unable to load asset: AssetManifest.json
#0      PlatformAssetBundle.load (package:flutter/src/services/asset_bundle.dart:258)
<asynchronous suspension>
#1      AssetBundle.loadString (package:flutter/src/services/asset_bundle.dart:83)
<asynchronous suspension>
```
Here is the code of my app:
```
import 'package:flutter/material.dart';

void main() {
  runApp(const MyApp());
}

class MyApp extends StatelessWidget {
  const MyApp({super.key});
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
        title: 'Img example', home: Image.asset('assets/frbul.jpg'));
  }
}
```
And the pubspec.yaml
```
name: img_example
description: A new Flutter project.

publish_to: 'none' # Remove this line if you wish to publish to pub.dev

version: 1.0.0+1

environment:
  sdk: '>=2.18.2 <3.0.0'

dependencies:
  flutter:
    sdk: flutter

  cupertino_icons: ^1.0.2

dev_dependencies:
  flutter_test:
    sdk: flutter

  flutter_lints: ^2.0.0

flutter:

  # To add assets to your application, add an assets section, like this:
  assets:
     - assets/frbul.jpg
```
The bundle folder layout looks like this:
```
.
└── img_example
    ├── data
    │   └── flutter_assets
    │       ├── AssetManifest.json
    │       ├── assets
    │       │   └── frbul.jpg
    │       ├── FontManifest.json
    │       ├── isolate_snapshot_data
    │       ├── kernel_blob.bin
    │       ├── NOTICES.Z
    │       ├── packages
    │       │   └── cupertino_icons
    │       │       └── assets
    │       │           └── CupertinoIcons.ttf
    │       ├── shaders
    │       │   └── ink_sparkle.frag
    │       └── vm_snapshot_data
    └── lib
        └── libapp.so

9 directories, 10 files
```

Thank you, Martin

martinburkot, 2022-11-29:
Hello,
I tried
`/usr/bin/flutter-client --b=/usr/share/flutter/img_example --f`
and i got:
`Not found option: b`
So i tried:
`/usr/bin/flutter-client --bundle=/usr/share/flutter/img_example -f`
And I got:
```
embedder.cc (960): 'FlutterEngineCreateAOTData' returned 'kInvalidArguments'. Invalid ELF path specified.
[ERROR][flutter_project_bundle.cc(82)] Failed to load AOT data from: /usr/share/flutter/img_example/libapp.so
[ERROR][flutter_elinux_engine.cc(164)] Unable to start engine without AOT data.
Failed to create view controller.
```
So I tried:
`/usr/bin/flutter-client --bundle=/usr/share/flutter/img_example/lib -f`
And I got the same exception:
```
root@colibri-imx6:~# /usr/bin/flutter-client --bundle=/usr/share/flutter/img_example/lib 
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Unable to load asset: AssetManifest.json
#0      PlatformAssetBundle.load (package:flutter/src/services/asset_bundle.dart:258)
<asynchronous suspension>
#1      AssetBundle.loadString (package:flutter/src/services/asset_bundle.dart:83)
<asynchronous suspension>
```
I also tried it without the fullscreen argument and the exception is still there. Is there something more I can try?

BR Martin

jwinarske, 2022-11-29:
`flutter-client` ?  Which embedder are you using?

martinburkot, 2022-11-29:
Hello,
`flutter-client` is added to /usr/bin when I add the recipe `flutter-wayland-client` to the build. I am new to yocto so I hope this is the answer for your question.

BR, Martin

jwinarske, 2022-11-29:
`--b=` is only an option with `homescreen` or `flutter-auto`.  You might consider using `flutter-auto`, as I don't regularly test the Sony embedders.  `flutter-auto` is part of Automotive Grade Linux` and I am publishing a fair amount of info of how to work with it 

https://ossjapan2022.sched.com/event/1D11L

martinburkot, 2022-11-30:
Hello,
I tried to use the `ivi-homescreen` and the `flutter-auto`, but I am unable to compile it on the dunfell branch. 

```
/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot-native/usr/bin/arm-tdx-linux-gnueabi/arm-tdx-linux-gnueabi-clang++ --sysroot=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr  -DBUILD_BACKEND_WAYLAND_VULKAN -DEGL_NO_X11 -DENABLE_PLUGIN_MOUSE_CURSOR -DENABLE_PLUGIN_PACKAGE_INFO -DENABLE_PLUGIN_SECURE_STORAGE -DENABLE_PLUGIN_TEXT_INPUT -DENABLE_PLUGIN_URL_LAUNCHER -DENV32BIT -DFILAMENT_SUPPORTS_WAYLAND -DHAVE_MEMFD_CREATE -DHAVE_STRCHRNUL -DLINUX -DMESA_EGL_NO_X11_HEADERS -DPATH_PREFIX=\"/usr\" -DWL_EGL_PLATFORM -Ishell -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/include/libsecret-1 -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/include/libmount -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/include/blkid -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/include/glib-2.0 -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/lib/glib-2.0/include -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/.. -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/../third_party -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/../third_party/flutter -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/../third_party/flutter/shell/platform/common/public -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/../third_party/flutter/shell/platform/common/client_wrapper/include -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/../third_party/rapidjson/include -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/third_party/filament/libs/bluevk/include -target arm-tdx-linux-gnueabi  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc --stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0=/usr/src/debug/flutter-auto/git-r0                      -fdebug-prefix-map=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0=/usr/src/debug/flutter-auto/git-r0                      -fdebug-prefix-map=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot-native=  --stdlib=libc++ -fvisibility-inlines-hidden -target arm-tdx-linux-gnueabi  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc --stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot -fno-rtti -DNDEBUG -fPIE   -stdlib=libc++ -I/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/recipe-sysroot/usr/include/c++/v1 -std=c++17 -MD -MT shell/CMakeFiles/flutter-auto.dir/backend/wayland_vulkan.cc.o -MF shell/CMakeFiles/flutter-auto.dir/backend/wayland_vulkan.cc.o.d -o shell/CMakeFiles/flutter-auto.dir/backend/wayland_vulkan.cc.o -c /home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/backend/wayland_vulkan.cc
/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/backend/wayland_vulkan.cc:82:39: error: invalid operands to binary expression ('VkCommandPool' (aka 'unsigned long long') and 'nullptr_t')
    if (state_.swapchain_command_pool != nullptr) {
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~
/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/backend/wayland_vulkan.cc:86:45: error: invalid operands to binary expression ('VkSemaphore' (aka 'unsigned long long') and 'nul [... cut ...]
                                                         ^~~~~~~
/home/martinburkot/workspace/oe-_new/build/tmp/work/armv7at2hf-neon-imx-tdx-linux-gnueabi/flutter-auto/git-r0/git/shell/backend/wayland_vulkan.cc:798:59: error: cannot initialize a parameter of type 'VkFence' (aka 'unsigned long long') with an rvalue of type 'nullptr_t'
  bluevk::vkQueueSubmit(b->state_.queue, 1, &submit_info, nullptr);
                                                          ^~~~~~~
6 errors generated.
```

BR Martin

jwinarske, 2022-11-30:
@martinburkot it looks like `vulkan` is present in your `DISTRO_FEATURES`, as it defaults to the Vulkan backend if this is the case.  Is this intentional?  I know the NXP BSP incorrectly adds `vulkan` even when the machine doesn't actually support Vulkan.  What's the SoC?

martinburkot, 2022-12-01:
Hello,
Vulkan was not there on purpose, so removing it from `DISTRO_FEATURES` solved the compiling issues so now I am able to use the `flutter-auto` and the `homescreen` and assets are working correctly in both of them. 
Thank you for your help.

closed 2022-12-01 by martinburkot