meta-flutter/meta-flutter #232: Running workspace on Mac M1 - 'Dart Error: Can't load Kernel binary: Invalid SDK hash.'

shoaibmerchant, 2022-12-18:
Hi, I am trying to setup the flutter-workspace with its default configuration on my M1 Mac since past 7 days but I am constantly hitting a dead-end with the error below when I try to run the flutter gallery app or a new empty app.

Note - I did have flutter installed earlier on my mac on a different folder, but I removed the folder; cleaned my $PATH and removed any .dart / or .flutter in my home directory.

```bash
> flutter run
INFO:configuration.cc(248)] **********
[INFO:configuration.cc(249)] * Global *
[INFO:configuration.cc(250)] **********
[INFO:configuration.cc(251)] Application Id: .......... flutter-auto
[INFO:configuration.cc(253)] JSON Configuration: ...... /tmp/gallery/default_config.json
[INFO:configuration.cc(255)] Cursor Theme: ............
[INFO:configuration.cc(256)] Disable Cursor: .......... false
[INFO:configuration.cc(258)] Debug Backend: ........... false
[INFO:configuration.cc(260)] ********
[INFO:configuration.cc(261)] * View *
[INFO:configuration.cc(262)] ********
[INFO:configuration.cc(264)] VM Args:
[INFO:configuration.cc(266)] --observatory-host=0.0.0.0
[INFO:configuration.cc(266)] --observatory-port=1234
[INFO:configuration.cc(269)] Bundle Path: .............. /tmp/gallery
[INFO:configuration.cc(270)] Window Type: .............. BG
[INFO:configuration.cc(271)] Size: ..................... 1920 x 1080
[INFO:configuration.cc(273)] Fullscreen: ............... false
[INFO:configuration.cc(275)] Accessibility Features: ... 0
[INFO:display.cc(275)] Keyboard Present
[INFO:engine.cc(173)] (0) Runtime=debug
[ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(385)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(604)] Could not launch engine with configuration.
```

I could see random reasons on the internet on why this error would come but nothing has worked so far. All fingers point towards some mismatch between the flutter and dart versions.

I tried updating the channel to stable in my pursuit to fix this, which upgraded to flutter 3.3.10. Below are some additional information, maybe it helps - 

1. flutter doctor -v
```bash
[✓] Flutter (Channel stable, 3.3.10, on macOS 12.4 21F79 darwin-arm, locale en-IN)
    • Flutter version 3.3.10 on channel stable at /Users/shoaibmerchant/Projects/Mecha/flutter-workspace/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 135454af32 (3 days ago), 2022-12-15 07:36:55 -0800
    • Engine revision 3316dd8728
    • Dart version 2.18.6
    • DevTools version 2.15.0

[!] Xcode - develop for iOS and macOS
    ✗ Xcode installation is incomplete; a full installation is necessary for iOS development.
      Download at: https://developer.apple.com/xcode/download/
      Or install Xcode via the App Store.
      Once installed, run:
        sudo xcode-select --switch /Applications/Xcode.app/Contents/Developer
        sudo xcodebuild -runFirstLaunch
    • CocoaPods version 1.11.3

[✓] VS Code (version 1.74.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.54.0

[✓] Connected device (2 available)
    • macOS (desktop)               • macos    • darwin-arm64 • macOS 12.4 21F79 darwin-arm
    • AGL arm64 QEMU Image (mobile) • AGL-qemu • linux-arm64  • agl-image-flutter-runtimedebug-qemuarm64

[✓] HTTP Host Availability
    • All required HTTP hosts are available

! Doctor found issues in 1 category.
```

2. flutter --version
```bash
Flutter 3.3.10 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 135454af32 (3 days ago) • 2022-12-15 07:36:55 -0800
Engine • revision 3316dd8728
Tools • Dart 2.18.6 • DevTools 2.15.0
```

3. dart --version
```bash
Dart SDK version: 2.18.6 (stable) (Tue Dec 13 21:15:14 2022 +0000) on "macos_arm64"
```

Any help would be appreciated!

jwinarske, 2022-12-18:
@shoaibmerchant This is a mis-match between Dart VM on Flutter side verses Embedder.  What custom-device are you trying to run on?

If AGL, and you are not using the [AGL workspace config](https://gerrit.automotivelinux.org/gerrit/gitweb?p=AGL/meta-agl-devel.git;a=blob;f=meta-agl-flutter/tools/flutter_workspace_config.json;h=bb7f23cc807ef32e98d3b15dad9aa71bc62fc2d3;hb=HEAD) found in the AGL repo, and want to run meta-flutter AGL image:  `./setup_flutter_workspace.py --flutter-version 3.3.1`

shoaibmerchant, 2022-12-18:
Ohh this must be it, I am using the AGL image on QEMU. So you mean that the AGL image has 3.3.1 flutter-engine and I am trying to run 3.3.9 / 3.3.10? 

I will confirm this and get back by tomorrow. 

Thanks again for the help!

shoaibmerchant, 2022-12-19:
@jwinarske Thanks for all help, everything is working fine now. Switching to 3.3.1 worked fine. Let me know if I can send a PR to update the tools/README.md to clarify this a bit. Because by default the script fetches the latest flutter version from stable channel.

jwinarske, 2022-12-19:
@shoaibmerchant AGL image artifacts will soon move downstream.  For AGL you should be using the downstream config.  The AGL platform inclusion in the meta-flutter workspace config is solely for regression testing; staging downstream changes.  The expectation is you are following one of my AGL OSS talks, which points to the AGL config.  I'll add a note in the README.

shoaibmerchant, 2022-12-19:
Noted @jwinarske thanks for all the help. I am not really working on AGL per se but I am deeply interested in flutter on embedded. I will definitely check your AGL talk.

We are actually trying to build a flutter based launcher for our IMX8 based device. I am fairly new on the flutter side but my intention was to setup the workspace to get a QEMU based arm64 environment running so that I can replace it with our custom image for a portable dev env & running tests.

closed 2022-12-19 by shoaibmerchant

jwinarske, 2022-12-19:
@shoaibmerchant Makes sense.

We are mid feature development for support Docker in Flutter Workspace Automation.  This support is around running a slim container on Ubuntu Linux
The active areas for this are:
```
https://github.com/meta-flutter/meta-flutter/blob/jw/fwa_docker/tools/setup_flutter_workspace.py
https://github.com/meta-flutter/meta-flutter/pull/233
```

This support is around targeting container on Ubuntu Linux ("desktop simulation"), and container in target OS.

We also have an upcoming work ticket around improving QEMU perf on M1 Mac.  Ventura enables native virtualization support that is better than modified QEMU.  It will be in form of QEMU image launcher, and will be integrated with Flutter Workspace Automation.  Initial target is AGL, but will be configurable to use for arm64 QEMU + Darwin.

shoaibmerchant, 2022-12-19:
Thanks for sharing this @jwinarske, docker will definitely simplify things to a large extent. 

On M1 the apps on AGL QEMU are quite sluggish right now, the native virtualization should help.

Is there a public chat / zulip around flutter for embedded just to stay updated on these developments?

jwinarske, 2022-12-20:
@shoaibmerchant the custom embedders channel on the flutter discord server is where some hang out.  I sometimes announce things there.

shoaibmerchant, 2022-12-20:
Great thanks will check!