meta-flutter/meta-flutter #234: ERROR: flutter-engine-git-r0 do_fetch: Bitbake Fetcher

WolsYang, 2022-12-21:
Is the mirror not work at this moment?
```
$ bitbake flutter-engine

ERROR: flutter-engine-git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'gn://github.com/flutter/engine.git;name=src/flutter')
```

jwinarske, 2022-12-21:
It's working fine with the CI jobs.  Perhaps a network issue on your end.

What overrides are you using?

WolsYang, 2022-12-21:
Sorry, what's means about overrides ?

I am work on Yocto Project 4.0 (Kirkstone)
```
$ git clone https://github.com/meta-flutter/meta-flutter.git
# Then
$ bitbake flutter-engine
```

jwinarske, 2022-12-21:
You are missing quite a few steps.  Are you able to build a core-image-minimal?

WolsYang, 2022-12-21:
Yes

I already build a 'imx-image-full' which is imx full linux image.

Before include 'flutter-engine', I wish I can make sure this recipe work, that why I only $ bitbake flutter-engine.

Should I just add it to my linux recipe and try again ?

jwinarske, 2022-12-21:
Try `bitbake flutter-engine-runtimerelease`.  Unless you're using the `jw/remove_bbclass` branch.

jwinarske, 2022-12-21:
Take a look at the CI jobs under the .github folder for reference usage.

closed 2022-12-21 by jwinarske