meta-flutter/meta-flutter #237: Automation tool flutter version when running with QEMU

bernardcraddock, 2022-12-29:
When using automation tool to run flutter_homescreen and others on AGL x86_64 QEMU Image (AGL-qemu) I'm getting the following SDK hash error

ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(385)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(604)] Could not launch engine with configuration.

if i recall it's probably because i'm running the wrong QEMU flutter version i need to change it
./setup_flutter_workspace.py --clean   --flutter-version xxx

Ques: How do i know what the correct QEMU flutter version is ?

please advise
regards

shoaibmerchant, 2022-12-29:
Please check this issue, it maybe relevant to you - 
https://github.com/meta-flutter/meta-flutter/issues/232

bernardcraddock, 2022-12-29:
@shoaibmerchant  thanks for that and looks like i need to set version 3.3.1  as I'm using AGL 
./setup_flutter_workspace.py --flutter-version 3.3.1

@jwinarske - what's the protocol for bumping AGL's flutter version

regards

jwinarske, 2022-12-29:
@bernardcraddock AGL is a downstream consumer of meta-flutter, so largely the roll is gated by Jan-Simon.  We have discussed planning around AGL changes in the last couple of production ready meetings.  AGL changes will happen post CES.

bernardcraddock, 2022-12-30:
@jwinarske - ok post CES is fine with me

Can you bump the SRCREV in flutter-app-pumped-fuel-ped_git.bb (agl branch) to the latest i've just pushed my final change for the year. 

One last thing the IVI production ready meetings are at your 9am my 4am, so i can never attend them and missing out so is there any chance you can change it to anytime your afternoon, so as i can attend?

regards

jwinarske, 2022-12-30:
@bernardcraddock Jan-Simon is in Germany, Walt is in Nashville, Scott is in San Jose, and I'm in Seattle.  The time window was what Jan-Simon was willing to accept, any later and it would be the middle of the night for him.

Notice most of their meetings are earlier.  I occasionally attend the weekly 6AM dev call.

bernardcraddock, 2022-12-31:
@jwinarske  weekly dev calls ok I'll start attending  if you're on it great if not no worries i'm sure Jan-Simon can provide enough to get me started

closed 2022-12-31 by bernardcraddock