meta-flutter/meta-flutter #239: Wrong syntax for PREFERRED_PROVIDER

witekio-kwe, 2023-01-13:
https://github.com/meta-flutter/meta-flutter/blob/81b5994786fe51307d092c8e5082bc97ff7181f0/recipes-graphics/toyota/ivi-homescreen_git.bb#L34 ans others set `PREFERRED_PROVIDER` as `PREFERRED_PROVIDER:<name>` but the valid syntax is still `PREFERRED_PROVIDER_<name>` (even in upstream poky)

jwinarske, 2023-01-15:
@witekio-kwe Thanks for raising the issue.  I'll take a look when I'm back from travels on Tuesday.

jwinarske, 2023-01-17:
@witekio-kwe I see the use of `PREFERRED_PROVIDER:` in container build which is correctly picking up `linux dummy`.  I reverted all cases of `PREFERRED_PROVIDER:` to `PREFERRED_PROVIDER_` in https://github.com/meta-flutter/meta-flutter/pull/241

Thanks again for the second set of eyes!

closed 2023-01-17 by jwinarske