meta-flutter/meta-flutter #240: Test on flutter beta channel

ardera, 2023-01-15:
meta-flutter does not currently compile with the beta channel, and beta as a bit tedious to use, as you can't just set `FLUTTER_SDK_TAG = "beta"` in local.conf, you need to specify (for example) `3.7.0-1.4.pre`.

My fork currently has some work done to enable building with beta channel (though that means it _exclusively_ builds with flutter beta right now):
https://github.com/meta-flutter/meta-flutter/compare/dunfell...ardera:meta-flutter:dunfell

CI testing with flutter beta would ensure that meta-flutter always builds with flutter beta

ardera, 2023-03-17:
Maybe the best way to go forward is to upstream the engine patches instead, so that there's no patches to be applied (or only some) and flutter beta has a higher chance of building right out of the box.

jwinarske, 2023-03-17:
Potentially.  I think their primary push back would be another platform on their end to test.  Already embedder support is the lowest priority for them.

jwinarske, 2023-03-21:
https://github.com/flutter/flutter/issues/123073

closed 2023-11-02 by jwinarske