meta-flutter/meta-flutter #242: When i am typing qemu_run a screen pop-up and closed in miliseconds, how to fix it

rishuriya, 2023-01-26:


jwinarske, 2023-01-26:
Questions:
1. Desktop OS?
2. meta-flutter branch/commit?

rishuriya, 2023-01-27:
OS is manjaro 
and using kirkstone branch /tools/readme.md instructions

jwinarske, 2023-02-01:
The only supported OS variants at the moment are:
* Ubuntu 20, 22
* Fedora 35, 36
* Mac M1

I may consider adding Redhat, Suse, and CentOS.  Only if they're capable of running a Wayland session.

What I suspect is the runtime packages required to run the qemu session are not met.

To trouble shoot is take the run qemu command from setup_environment.sh and run in a terminal.  The other is to translate the package names installed for Ubuntu or Fedora.  This is the process I use when bringing up script on new OS.

Personally I wouldn't use mint as it defaults to X11.  Another option is to run a docker image.

jwinarske, 2023-03-04:
Closing since I'm not getting a response.  Feel free to reopen if needed.

closed 2023-03-04 by jwinarske