meta-flutter/meta-flutter #247: Hardcoded paths prevent running any flutter app

witekio-kwe, 2023-02-09:
https://github.com/meta-flutter/meta-flutter/blob/0b4a6bb9c59125cfe47456500614f875c91d20d8/recipes-graphics/sony/sony-flutter.inc#L24

references a patch that requires icudtl.dat to be in /usr/share/flutter/icudtl.dat.
After the recent changes I’m facing the following issue:
My app build with flutter-app.bbclass creates

- /usr/share/<appname>/<version>/<profile>/data/icudtl.dat

nothing in /usr/share/flutter gets installed at all.
I would need to copy/symlink that file to run any of our apps.

If the patch is removed I can easily run any of the app I got with

`flutter-client -b /usr/share/<appname>/<version>/<profile>/`

Is there a need for this patch?
To me it doesn’t feel right to keep that one

jwinarske, 2023-02-10:
@witekio-kwe I don't use or actively test the Sony embedders.  I'm happy to make adjustments if you do.

The performance with the Sony embedders is inferior to both flutter-auto and flutter-pi.

Is there a feature you like that only Sony does?

jwinarske, 2023-02-10:
Removed in kirkstone branch

closed 2023-02-10 by jwinarske

witekio-kwe, 2023-02-10:
> @witekio-kwe I don't use or actively test the Sony embedders. I'm happy to make adjustments if you do.
> 
> The performance with the Sony embedders is inferior to both flutter-auto and flutter-pi.
> 
> Is there a feature you like that only Sony does?
I just noticed that one an older demo we have that is based on the sony client.

Flutter-auto doesn’t work for us, because of the need for egl backend.
Haven’t had a look at flutter-pi, but it could be a replacement.

Tbh I was so far quite happy with the flutter-client as it works find and performant enough in out setup - but good to know that is it isn’t actively tested here, I will have an eye on future regressions and maybe post some patches when something breaks

jwinarske, 2023-02-10:
@witekio-kwe what backend do you need?  It supports EGL and Vulkan.  I will be adding DRM lease in the next couple of quarters.