meta-flutter/meta-flutter #248: Tune FLUTTER_RUNTIME_MODES when installing

witekio-kwe, 2023-02-09:
https://github.com/meta-flutter/meta-flutter/blob/0b4a6bb9c59125cfe47456500614f875c91d20d8/classes/flutter-app.bbclass#L286

takes everything into account that was created by the PACKAGECONFIG of flutter-engine.
In my setup I want to be able to control which of the profiles will get installed for my app - this will save a ton of space and maybe makes the need to this `FLUTTER_APP_SKIP_DEBUG_INSTALL` switch obsolete.

In my opinion it should install everything found in terms of profiles to the package-image-dir, unless the user defines a subset of things to install

Something like `FLUTTER_APP_INSTALL ?= "release"` as a default, which then will be tuneable in a recipe scope.
Profile not being present but requested should result in an error in my opition

jwinarske, 2023-02-10:
@witekio-kwe 
* If you include the profile engine then you want profile enabled apps installed.  Dev build.
* the jit runtime only applies to 32-bit, so really this should be mutually exclusive with release.  Which is why the package config is setup to be overriden.  CI includes this simply to test the build.
* In case of the SDK you wouldn't include apps in image for obvious reasons.

So I'm not clear on your concern.

witekio-kwe, 2023-02-10:
we have enabled all 3 variants (debug, release and profile) via package config for the flutter-engine, as some of our recipes need one variant of that.
If now a recipe is build using flutter-app and with the mentioned flutter-engine with all 3 variants, I basically end up with an app in 3 variants in my image.
Not sure if that is intended behavior, it’s just what I saw.

After digging through the code, for me it would be the easiest to control which app-variants actually makes into the image/is actually compiled, by tweaking above mentioned line.

Maybe I’m missing out on something, but I see no way to control the behavior of a flutter-app based recipe on a recipe level, only way out would be to create 3 variants of flutter-engine, which just would create an even bigger mess with all the dependency to be set

jwinarske, 2023-02-10:
@witekio-kwe

It will not install debug app unless you ask it to.  Debug builds are something that should only originate from the host...  I only added the override flag for debug install in case someone wanted to do things backwards.

The current approach uses engine config as the origin of truth.  Setting app to install selectively removes the origin of truth.  Which directly makes it more confusing/difficult when someone is trying to figure out how it works.

Given there is no reason to install a debug app variant; ever, the only item to selectively install should be perf.  I'll change the debug install flag to selectively install perf.

witekio-kwe, 2023-02-10:
yeah, that would be already an improvement - in our setup we have very few profile apps (mostly demos) and a lot of release apps - so getting profile off the images when release is wanted would be already great

witekio-kwe, 2023-02-10:
https://github.com/meta-flutter/meta-flutter/pull/249 kinda moves in that direction too

jwinarske, 2023-03-17:
https://github.com/meta-flutter/meta-flutter/commit/5b03d0af7cc2f01b730dd27aa75d981e8b9bd9ab

closed 2023-03-17 by jwinarske