meta-flutter/meta-flutter #255: MissingPluginsException in flutter-test-plugins [kirkstone]

TWeidi, 2023-03-02:
Hi,

I wanted to try out your demo-application from your meta-flutter layer on an i.MX 8M Plus on a [[Toradex Verdin Board](https://www.toradex.com/de/computer-on-modules/verdin-arm-family/nxp-imx-8m-plus)]. The image builds fine and works as expected, but when it comes to running the linux_plugins demo application using:
`flutter-pi --release linux_plugins/3.3.10/release`
from the boards terminal, selecting for instance 'Misc' from the main menu, and switching to any one tab either displays a MissingPluginException or prints an error message to the terminal. It seems that the Flutter plugins have not been installed to the image.
Moreover, I cannot use the mouse cursor, since moving the mouse immediately leads flutter-pi to printing an error message to the terminal, as well.
I was trying to increase the engine version to a more recent one, to see if this has any influence on the demo applications, setting `FLUTTER_SDK_TAG = "AUTOINC"`, but this led to the yocto build to fail.

Can you please give some advice on how to get the demo app running correctly?

jwinarske, 2023-03-02:
@ardera Hannes can comment on cursor behavior.

@TWeidi One important thing to note is around 1P Linux plugins.  These are not implicitly supported, nor will work out of the box with a Linux embedder.  They will work for a host build of the Linux GTK embedder.  When Flutter indicates it supports Linux with their plugin, it really means it only supports the Flutter Linux GTK embedder.  That said adding missing platform channel support in flutter-pi or homescreen/flutter-auto is not a difficult task.  Are there specific plugins which you want/need support for?  Currently 3.3.10 is the latest build available.  3.7+ is in the works.  The problems you are seeing are most likely 100% 1P plugin issues.

jwinarske, 2023-03-02:
In addition if you try to run your 1P Linux plugins in a debug build, it will appear to work.  This is a Flutter bug.  They are not supposed to work according to Stuart Morgan.

TWeidi, 2023-03-03:
Hello Joel,

thank you very much for your quick reply. Sorry to say, but to me it is not clear what you mean with "1P Linux plugins" as I could not find a source on the web that explains that term in this context. I am wondering why plugins like the "shared_preferences", which seems to be written purely in dart language, cannot be directly integrated in the image, but admittedly I am fairly new to the flutter ecosystem. 
Actually, I would really need the flutter_riverpod and the ffi plugins. Since you mentioned that adding platform channel support in flutter-pi or homescreen/flutter-auto is not a difficult task, can you show me some example implementation or hint me to some resources that demonstrate how to achieve this?

Thank you and kind regards

Thomas

ardera, 2023-03-03:
> Moreover, I cannot use the mouse cursor, since moving the mouse immediately leads flutter-pi to printing an error message to the terminal, as well.

Flutter-pi only supports hardware mouse cursors right now. Some devices don't have hardware mouse cursor support, in which case an error message will be printed. Though I have some updates to mouse cursor support in `feature/gstreamer-player-from-pipeline` that might make it work on iMX as well.

In the meantime you can use https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626 to get a software mouse cursor :)

jwinarske, 2023-03-04:
@TWeidi 
In the case of flutter-pi the plugins are static and implemented in C.

In the case of home screen/flutter-auto the plugins are currently static.  I just released the quarters release, it had a few more plugins.

https://github.com/toyota-connected/ivi-homescreen/tree/main/shell/static_plugins

The process is run in an embedder, observe the channel failures.  Cross check the dart code, implement following known pattern.  This is if your flutter plugins require/have a native component.

Ask if you get stuck.

The pattern with meta-flutter is to use a bbappend recipe overriding repo url.  Point it to your fork.

jwinarske, 2023-03-10:
Feel free to re-open if needed be

closed 2023-03-10 by jwinarske