meta-flutter/meta-flutter #261: Question: Error running flutter-client

omarhogni, 2023-03-08:
After building an image running the image in release mode segfaults and debug mode produces this error
```
root@intel-corei7-64:~# FLUTTER_LOG_LEVELS=TRACE LD_LIBRARY_PATH=/usr/share/flutter/iqf/3.3.10/debug/lib/ XDG_RUNTIME_DIR=/run/  WAYLAND_DISPLAY=wayland-0 DISPLAY=0.0 /usr/bin/flutter-client --bundle=/usr/share/flutter/iqf/3.3.10/debug/ --fullscreen --no-cursor
[TRACE][elinux_window_wayland.cc(171)] presentation info: clk_id = 1
[INFO][elinux_window_wayland.cc(606)] Display scale for output(8): 1
[INFO][elinux_window_wayland.cc(573)] Display output info: width = 640, height = 480, refresh = 74998
[TRACE][elinux_window_wayland.cc(1148)] Created the Wayland surface: 640x480
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(385)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(604)] Could not launch engine with configuration.
flutter: The Dart VM service is listening on http://127.0.0.1:44599/IeMjEzuuO5M=/
```

Segfaulting output

```
FLUTTER_LOG_LEVELS=TRACE LD_LIBRARY_PATH=/usr/share/flutter/iqf/3.3.10/release/lib/ XDG_RUNTIME_DIR=/run/ WAYLAND_DISPLAY=wayland-0 DISPLAY=0.0 /usr/bin/flutter-client --bundle=/usr/share/flutter/iqf/3.3.10/release/ --fullscreen --no-cursor
[TRACE][elinux_window_wayland.cc(171)] presentation info: clk_id = 1
[INFO][elinux_window_wayland.cc(606)] Display scale for output(8): 1
[INFO][elinux_window_wayland.cc(573)] Display output info: width = 640, height = 480, refresh = 74998
[TRACE][elinux_window_wayland.cc(1148)] Created the Wayland surface: 640x480
flutter: Initalizing sqlite3
flutter: Registering linux for pathprovider and sharedpreferences
flutter: Loading persistent defaults
flutter: Starting application
flutter-client[379]: segfault at 21c01100 ip 0000559f36ecbe87 sp 00007f43060bf4b0 error 4 in flutter-client[559f36e97000+5a000]
Code: 08 48 89 e2 b9 01 00 00 00 41 ff 54 24 28 83 f8 01 0f 85 e2 00 00 00 49 8d 5c 24 38 49 8b 6e 20 bf 30 00 00 00 e8 39 b4 fc ff <0f> 10 45 00 0f 10 4d 10 0f 11 40 10 0f 11 48 20 48 89 c7 48 89 de
Segmentation fault
```

Any pointers to how to solve this would be greatly appriciated.

omarhogni, 2023-03-08:
After tracking head for kirkstone branch and adding X11 as distro feature it runs in release mode with sdk version 3.7.6.

Debug mode still does not start. Closing the issue as it is solved for me.

closed 2023-03-08 by omarhogni

jwinarske, 2023-03-08:
@omarhogni Hi.  I personally don't use the Sony embedders.  ivi-home screen/flutter-auto is my go-to.  Unless you need X11, then Sony is the only option.  For DRM flutter-pi.