meta-flutter/meta-flutter #263: [Docs/Help] How to create a new app and build for RaspberryPi4

michele-ruocco, 2023-03-14:
Hi everyone, 
thank you very much for the work that is being done in this repository with Flutter. 

I am new to the AGL world and still have many aspects to understand. I have built an image for RaspberryPi4 with the latest available version of AGL. (agl-ivi-demo-platform-flutter) 

Everything seems to be working. I wanted to better understand the workflow and how I can port and start a simple hello world app. 

Could anyone help me with this?

Once the workspace is set up via the python automation script. How can I proceed to:

- Create a new Flutter Hello World app.
- Build the app and insert it into a new image.
- Start the image on Raspberry Pi4 and launch the new app.

This is not an issue but I hope someone can provide me with some minimal support for these first steps. 

Thank you all.

closed 2023-03-15 by michele-ruocco

jwinarske, 2023-03-16:
Have you seen this slide deck?

https://ossjapan2022.sched.com/event/1D11L/update-on-flutter-in-agl-and-ivi-expert-group-joel-winarske-toyota-connected-north-america

michele-ruocco, 2023-03-16:
Yes, and I tested the latest green build for Raspberry Pi 4 Vulkan. A tutorial or documentation on compiling a new image with app flutter on the local machine would be helpful. Is the only way to insert new apps is to create an image from scratch or is there an alternate way to "replace/install" flutter app on a flash sd card image for Raspberry Pi 4?

jwinarske, 2023-03-16:
A number of workflows are supported.
1. Develop against embedder running on desktop using `flutter run -d desktop-auto`.  This requires running a `wayland-session` (Ubunu 20, 22, Fedora 35-37).
2. Build a x86_64 container image (docker-compose or just docker) and debug flutter code running in container on host.  This does not require a `wayland-session`, and there is a Ventura Mac launcher (virt-qemu on github.com/meta-flutter).  Meaning if you build an arm64 QEMU variant of your OS image, you can run on your M1/M2 Mac.
3. Build OS image including runtime variant runtime=debug.  Debug from host on target with your favorite tools over Ethernet.  Could also install adbtools and setup port forwarding.
4. Build final OS image with only runtime=release for distribution.

flutter-auto/ivi-homescreen was designed with the idea that the same code that runs on the host runs on the target device.  The primary delta is the environments GPU+backend (EGL or Vulkan).  If you intend to use Texture Widgets you need EGL.  If you intend to use Wayland Compositor Surfaces, then either.

In case of Yocto SDK, this gives you the option to install the Flutter SDK.  This supports Long Term Support (LTS).  Where you ship something into the wild, and you need to make changes.  For this it would make sense to install the Yocto SDK in a Docker image based on your releases.  Then you will always be able to support your app regardless of when in the future changes are needed.

So in general what you need is an OS image with runtime=debug.  Then you can debug using host tools.

mikeleruo, 2023-03-17:
Thank you, minimal flutter image with runtimedebug does not contain weston ? How can I start the apps I included in the .bb?

michele-ruocco, 2023-03-17:
How can I add a test app to the agl-ivi-demo-platform-flutter image? 

I tried the build with bitbake of the minimal debug image by adding in the .bb file some sample apps. Flashing to Raspberry Pi4 I don't have Weston terminal, only blank screen. How can I view and launch the apps?